adafruit/circuitpython #3341: Support the ESP32-S2's Digital Signature Peripheral

askpatrickw, 2020-08-28:
The ESP32-S2 includes a limited hardware security module which Espressif calls the Digital Signature Peripheral. 

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/ds.html
 
>  The Digital Signature (DS) module provides hardware acceleration of signing messages based on RSA.
> It uses pre-encrypted parameters to calculate a signature. The parameters are encrypted using HMAC as a
> key-derivation function. In turn, the HMAC uses eFuses as input key. The whole process happens in hardware
> so that neither the decryption key for the RSA parameters nor the input key for the HMAC key derivation function
> can be seen by the software while calculating the signature.
> 
> Look into the [ESP32-S2 Technical Reference Manual](https://www.espressif.com/sites/default/files/documentation/esp32-s2_technical_reference_manual_en.pdf)for more detailed information about the involved
> hardware during the signature calculation process and the used registers.
>

tannewt, 2020-08-28:
What CPython API would you implement with this peripheral? Or what would the CircuitPython API be?

askpatrickw, 2020-09-11:
I opened this after a brief discussion with @brentru about the Azure Iot Library supporting X509 based identification. It was Brent's suggestion that this would more likely be a CP library specific to the esp32s2 port than in the Azure IoT Library itself.

For additional context Azure IoT Device Provisioning Services(DPS) allows you to ship devices with Keys which are then used as Device IDs. Based on these Device IDs, DPS gives the device its name and its IoT Hub to which it connects and an additional Key.

I'm not a crypto expert, but I believe the necessary operations are storing the keys from Azure IoT Device Provisioning Service and using the peripheral to do message encryption\decryption as well. 

As I re-read all these docs, it also appears though the HMAC will need to be supported as well as it stores the keys.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-reference/peripherals/hmac.html

All of my explanations are Azure IoT specific, but I could also see, in a very straight-forward manner Adafruit.io also supporting this style of device ID and shipping boards with pre-provisioned DSP\HMAC (You can order them this way from Espressif). While security may be hard and frequently leads to an opaque hard to understand system, with an approach such as this it could be made secure by default for beginners and hide the complexity for them. </sales pitch> :-)

tannewt, 2020-09-14:
I agree with you about the motivation. It is a good thing to have.

I'm wondering what it means in practice. AKA, what is the API? Finding a matching API in CPython is the easiest approach but isn't always possible due to CircuitPython's constraints.

askpatrickw, 2020-12-03:
Just a link for now... 
https://github.com/espressif/esp-idf/tree/master/examples/protocols/mqtt/ssl_ds#configure_dspy
Espressif has some helper scripts for the DSP. I'm working on something also with this chip for work so when I figure it out there... I'll share here as well.

askpatrickw, 2021-01-19:
Another update, the DS Peripheral was supported as of the Dec 7, 2020 release tagged 4.2
https://github.com/espressif/esp-idf/releases/tag/v4.2

Today the port is using https://github.com/adafruit/esp-idf/tree/ebe7784258d8c10e9cc334ccc00c3fd270746c8b which is Dec 11th... so should be good, but I didn't look too closely.

Just mentioning it because the DSP support is relatively new...

torntrousers, 2022-01-16:
Hi, a year on, whats the status of DSP support, any chance of using it yet?

tannewt, 2022-01-18:
Nothing has been added and I haven't seen a proposed API. We're open to merging an API in when someone creates one.

closed 2026-11-02 by tannewt