adafruit/circuitpython #3345: Website Github release action broken: can't find circuitpython-stubs

dhalbert, 2020-08-28:
I made a CircuitPython release and one aspect of the the build failed:

```
Website1
##[error]Process completed with exit code 1.
Run python3 build_board_info.py
Logged in as adafruit-blinka
Traceback (most recent call last):
  File "build_board_info.py", line 310, in <module>
    generate_download_info()
  File "build_board_info.py", line 243, in generate_download_info
    support_matrix = shared_bindings_matrix.support_matrix_by_board(
  File "../docs/shared_bindings_matrix.py", line 148, in support_matrix_by_board
    base = build_module_map()
  File "../docs/shared_bindings_matrix.py", line 69, in build_module_map
    modules = get_shared_bindings()
  File "../docs/shared_bindings_matrix.py", line 47, in get_shared_bindings
    return [item.name for item in shared_bindings_dir.iterdir()]
  File "../docs/shared_bindings_matrix.py", line 47, in <listcomp>
    return [item.name for item in shared_bindings_dir.iterdir()]
  File "/opt/hostedtoolcache/Python/3.8.5/x64/lib/python3.8/pathlib.py", line 1118, in iterdir
    for name in self._accessor.listdir(self):
FileNotFoundError: [Errno 2] No such file or directory: '/home/runner/work/circuitpython/circuitpython/circuitpython-stubs'
##[error]Process completed with exit code 1.
```

jepler, 2020-08-28:
Since #3332 the stubs are used instead of shared-bindings to get the list of modules shown in the matrix.  I _THINK_ the fix for this is to move the stub-generating rules into create_website_pr.yml.  However, I could use some guidance as to how to test a PR, since it only runs on a release-publish.

dhalbert, 2020-08-28:
Maybe this or this?
https://github.community/t/github-actions-manual-trigger-approvals/16233/3
https://github.blog/changelog/2020-07-06-github-actions-manual-triggers-with-workflow_dispatch/

PR #3352 (2020-09-07): shared_bindings_matrix revert to using shared-bindings
.. hard-coding ulab for now.

It also fixes a problem where board_name was unassigned when use_branded_name was False, which only happened at release-building time.

Trying to change this caused multiple problems in the release process.

Closes: #3345
closed 2020-09-07 by jepler