adafruit/circuitpython #3351: USB VID/PID Request for MuseLab nanoESP32-S2

maditnerd, 2020-08-29:
I'd like to give CPY support to nanoESP32-S2 boards on behalf of MuseLab. 
Would it be possible to receive a VID/PID for this purpose?

I didn't find any VID/PID in MuseLab documentation : https://github.com/wuxx/nanoESP32-S2

I'm an individual with no affiliation with Muselab.

Thanks!

dhalbert, 2020-08-31:
If you plug one of the boards in now, what VID/PID does it show?

tannewt, 2020-08-31:
@wuxx Please provide a USB VID and PID for this board.

wuxx, 2020-09-01:
Hi，nanoESP32-S2 has two USB port，one connect to CH340 which is usb to serial IC from WCH，
the another USB is from ESP32-S2，the USB VID & PID is define in your code，We  hasn't register USB VID & PID currently.

maditnerd, 2020-09-01:
I reupload the default firmware (from https://github.com/wuxx/nanoESP32-S2), and I got this VID/PID used for tiny USB Mass storage
VID: 0xCAFE
PID: 0x4003

Should I use this one ? I don't know how to check if this is a valid VID/PID.

Here is USBDeview screenshot
[image]

wuxx, 2020-09-01:
this vid/pid is the demo code from expressif, if just for test yourself, you can use it freely

maditnerd, 2020-09-01:
I want to add this board to circuitpython, I follow @tannewt's instructions but I need a unique VID/PID to finish my pull request.
So I guess as cool as having CAFE ☕ as VID is, it won't be OK for this purpose ?

wuxx, 2020-09-01:
ok.... if you or ciruitpython don't care about the USB vid/pid register things... just use it.
and the vid CAFE currently not belong to any company...

here is some reference:
How to obtain an USB VID/PID for your project
https://www.voti.nl/docs/usb-pid.html

The USB ID Repository
http://www.linux-usb.org/usb-ids.html

How to get a PID code
http://pid.codes/howto/

tannewt, 2020-09-01:
@ladyada How do you want to handle this?

ladyada, 2020-09-14:
@tannewt getting to some old emails, @dhalbert can allocate a pid

dhalbert, 2020-09-14:
okee, will do tomorrow

tannewt, 2020-09-15:
@maditnerd Please use:

```
MuseLab nanoESP32-S2       VID 0x239A    PID 0x00DE  # bootloader
                                         PID 0x80DE  # circuitpython
```

maditnerd, 2020-09-16:
Thanks ❤️ , i'll check what i've done (probably rewatch @tannewt video), and do a pull request on both **adafruit/circuitpython** and **adafruit/circuitpython-org**

I asked my sister (who is a photographer) to take a picture of the board.

[image]

tannewt, 2020-09-16:
Pic looks awesome! I've merged both PRs so I'll close this. Thanks for the addition!

closed 2020-09-16 by tannewt