adafruit/circuitpython #3353: Reading the value of an output pin behaves differently in ESP32S2 than in SAMD

anecdata, 2020-08-30:
Is this a bug, or is it explicitly undefined behavior to read the value of an output?

PyPortal (alpha.3):
```
>>> led.value = True
>>> led.value
True
```
(the LED is ON)

Saola (alpha.3):
```
>>> led.value = True
>>> led.value
False
```
(the LED is ON)

jepler, 2020-08-30:
If it's not intended to be able to read back the value previously set on an output pin, accessing the value property to read it should give an exception.

I think the problem arises because the esp-idf defines the return value of this routine in a surprising way, rather than returning the value currently driven onto the pin:
```
 * @brief  GPIO get input level
 *
 * @warning If the pad is not configured for input (or input and output) the returned value is always 0.
 *
 * @param  hw Peripheral GPIO hardware instance address.
 * @param  gpio_num GPIO number. If you want to get the logic level of e.g. pin GPIO16, gpio_num should be GPIO_NUM_16 (16);
 *
 * @return
 *     - 0 the GPIO input level is 0
 *     - 1 the GPIO input level is 1
```

anecdata, 2020-08-30:
FWIW, input behaves the same on both. It's conceptually an easier case to make that it's an error since the value is controlled by an external source:
```
>>> pin = DigitalInOut(any_gpio)
>>> pin.direction = Direction.INPUT
>>> pin.value
False
>>> pin.value = True
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: Cannot set value when direction is input.
```

PR #3495 (2020-10-01): Add Metro ESP32S2
Also fix two bugs:
* Crash when resetting board.SPI when it hasn't been inited.
* Reading back the output value is always false. Fixes #3353
commit a739d314 (2020-10-01): Add Metro ESP32S2

Also fix two bugs:
* Crash when resetting board.SPI when it hasn't been inited.
* Reading back the output value is always false. Fixes #3353
closed 2020-10-01 by tannewt