adafruit/circuitpython #3360: Current unchanged when trying to turn off QSPI during sleep

catequalsgood, 2020-08-31:
After reading about the ability to turn off QSPI during sleep for the nRF52840 in 6.0.0 alpha 3 I tried emulating what @xiongyihui did in issue #3244 on my itsybitsy BLE (also nRF52840). However after building CircuitPython with the CIRCUITPY_ENABLE_MPY_NATIVE=1 flag to be able to use the @micropython.asm_thumb decorator and adding a few lines to xiongyihui's code to disable the DotStar LED the current drawn from battery (no USB connected) is still identical to what it was when using 6.0.0 alpha 2 (about 2.5 mA). The quiescent current of the DotStar is around 1 mA and I will physically remove it once the project is nearly finished.

I am not sure how I could tell if QSPI is actually off other than looking at the current and since xiongyihui measured far less (on a different board however) I assume I am either doing something wrong or the difference in hardware is to blame (maybe turning off the DotStar in this way doesn't allow QSPI to turn off?).

this is all I added to [xiongyihui's code](https://github.com/adafruit/circuitpython/pull/3244#issuecomment-670410940):
`import board`
`import adafruit_dotstar as dotstar`
`dot = dotstar.DotStar(board.APA102_SCK, board.APA102_MOSI, 1)`
`dot[0] = (0, 0, 0)`

xiongyihui, 2020-08-31:
Does `dotstar.DotStar` use SPI bus?

We need turn off other peripherals(USB, PWM and etc) using high frequency clock (HFCLK), and then go to sleep.

dhalbert, 2020-08-31:
Yes, the dotstar library uses SPI if it can. Below is from https://github.com/adafruit/Adafruit_CircuitPython_DotStar/blob/master/adafruit_dotstar.py. The way around this would be to modify the code to force it to use bit-bang method it does when SPI is not avialable on the requested pins.

```python
        self._spi = None
        try:
            self._spi = busio.SPI(clock, MOSI=data)
            while not self._spi.try_lock():
                pass
            self._spi.configure(baudrate=baudrate)

        except (NotImplementedError, ValueError):
            self.dpin = digitalio.DigitalInOut(data)
            self.cpin = digitalio.DigitalInOut(clock)
            self.dpin.direction = digitalio.Direction.OUTPUT
            self.cpin.direction = digitalio.Direction.OUTPUT
            self.cpin.value = False
```

catequalsgood, 2020-08-31:
It didn't even occur to me to test current with the LED still on but I did and got this: 
Doing nothing but importing time and going to sleep (LED still on) results in about 1.95 mA. Lower than before even though the LED is now on. 
When using the code below setting the brightness for the first time using the DotStar object causes the current to jump from 1.95 to 2.85 mA even though the brightness of the LED doesn't change. I assume that increase is caused by the HFCLK turning on again (and then not turning off). But even with the LED physically removed, current would still be around 1 mA which still seems high in comparison to the measurements from #3244. 

```
import time
import adafruit_dotstar as dotstar
import board
dot = dotstar.DotStar(board.APA102_SCK, board.APA102_MOSI, 1)

while True:
    time.sleep(10)
    dot[0] = (0, 9, 0) #current jumps from 1.95 mA to 2.85 mA on first execution
    time.sleep(10)
```

tannewt, 2020-08-31:
@catequalsgood try disabling the Dotstar in the board definition you are using. It will configure SPI if it's marked as a status LED for the board.

xiongyihui, 2020-09-01:
@catequalsgood When LFCLK is off, the current of nRF52840 is a few uA. While, other parts of the board may have higher power consumption.  As you see, a LED takes about 0.95 mA.

The hardware I used is the M60 keyboard. It is optimized for low power consumption. Other parts of it still needs around 50 uA.

catequalsgood, 2020-09-01:
Thanks for everyone's explanations and tips!
I removed the DotStar in the board definition (I think) by removing the following lines from `mpconfigboard.h`
```
#define MICROPY_HW_APA102_MOSI   (&pin_P0_08)
#define MICROPY_HW_APA102_SCK    (&pin_P1_09)
```
That dropped the current during sleep to 0.95 mA which is obviously way better than what I got when turning off the LED using `adafruit_dotstar.Dotstar`. Just out of curiosity I did also try physically removing the LED from one of my itsybitsy boards and the current dropped further to just above 0.5 mA (even with `mpconfigboard.h` untouched). That is pretty close to power consumption I can live with. For my application I would probably do the sleep part of the code first and then everything that might cause any peripherals to turn on. After that the board could reset itself by triggering the reset pin. But that would obviously not work for every application.

Is 0.5 mA going to be the limit for the itsybitsy simply because it was not designed with low-power applications in mind?

PS: Defining all digital pins as outputs might result in a further reduction of about 30 uA or so but I is hard to tell with my cheap multimeter.

tannewt, 2020-09-01:
@catequalsgood Yup, that's what I was thinking to disable the dotstar status neopixel. I'm not sure what the lowest possible value is. @xiongyihui's work is the best indication of it. When I added the low power support, it was mainly to get the code base ready for it.