adafruit/circuitpython #3361: cannot make hex file for BOARD=makerdiary_nrf52840_mdk_usb_dongle

T2brozz, 2020-08-31:
i have tried to create the hex file with  
>  make BOARD=makerdiary_nrf52840_mdk_usb_dongle SD=s140 V=1 -j4 bin

like in the Readme.md described.
But I've got every time the error  
`make: *** No rule to make target 'hex'.  Stop.`
Can someone help me

jerryneedell, 2020-08-31:
Try without the "bin"
```
 make BOARD=makerdiary_nrf52840_mdk_usb_dongle SD=s140 V=1 -j4
```
worked for me
```
Create build-makerdiary_nrf52840_mdk_usb_dongle-s140/firmware.bin
Create build-makerdiary_nrf52840_mdk_usb_dongle-s140/firmware.hex
Create build-makerdiary_nrf52840_mdk_usb_dongle-s140/firmware.uf2
python3 ../../tools/uf2/utils/uf2conv.py -f 0xADA52840 -c -o "build-makerdiary_nrf52840_mdk_usb_dongle-s140/firmware.uf2" build-makerdiary_nrf52840_mdk_usb_dongle-s140/firmware.hex
Converting to uf2, output size: 836608, start address: 0x26000
Wrote 836608 bytes to build-makerdiary_nrf52840_mdk_usb_dongle-s140/firmware.uf2
[Jerry-desktop-mini:circuitpython/ports/nrf] jerryneedell% ls -lrt build-makerdiary_nrf52840_mdk_usb_dongle-s140/
total 66776
drwxr-xr-x   3 jerryneedell  staff       102 Aug 31 13:37 peripherals
drwxr-xr-x   4 jerryneedell  staff       136 Aug 31 13:37 nrfx
drwxr-xr-x  12 jerryneedell  staff       408 Aug 31 13:37 lib
drwxr-xr-x   3 jerryneedell  staff       102 Aug 31 13:37 device
drwxr-xr-x  21 jerryneedell  staff       714 Aug 31 13:37 common-hal
drwxr-xr-x   3 jerryneedell  staff       102 Aug 31 13:37 boards
drwxr-xr-x  29 jerryneedell  staff       986 Aug 31 13:37 shared-module
-rw-r--r--   1 jerryneedell  staff     11176 Aug 31 13:37 autogen_usb_descriptor.c
-rw-r--r--   1 jerryneedell  staff   1886650 Aug 31 13:37 ld_defines.pp
-rw-r--r--   1 jerryneedell  staff      7648 Aug 31 13:37 common.ld
drwxr-xr-x   3 jerryneedell  staff       510 Aug 31 13:38 genhdr
drwxr-xr-x   2 jerryneedell  staff      7956 Aug 31 13:38 py
drwxr-xr-x   3 jerryneedell  staff      1802 Aug 31 13:38 extmod
-rw-r--r--   1 jerryneedell  staff    981696 Aug 31 13:38 main.o
-rw-r--r--   1 jerryneedell  staff     20942 Aug 31 13:38 main.P
drwxr-xr-x   2 jerryneedell  staff       136 Aug 31 13:38 build-makerdiary_nrf52840_mdk_usb_dongle-s140
-rw-r--r--   1 jerryneedell  staff    825208 Aug 31 13:38 fatfs_port.o
-rw-r--r--   1 jerryneedell  staff     10840 Aug 31 13:38 fatfs_port.P
-rw-r--r--   1 jerryneedell  staff    874900 Aug 31 13:38 background.o
-rw-r--r--   1 jerryneedell  staff     18508 Aug 31 13:38 background.P
-rw-r--r--   1 jerryneedell  staff      5064 Aug 31 13:38 autogen_display_resources.c
drwxr-xr-x   2 jerryneedell  staff       136 Aug 31 13:38 bluetooth
-rw-r--r--   1 jerryneedell  staff    827780 Aug 31 13:38 sd_mutex.o
-rw-r--r--   1 jerryneedell  staff     10656 Aug 31 13:38 sd_mutex.P
drwxr-xr-x  42 jerryneedell  staff      1496 Aug 31 13:39 shared-bindings
drwxr-xr-x   3 jerryneedell  staff       408 Aug 31 13:39 supervisor
-rw-r--r--   1 jerryneedell  staff    807556 Aug 31 13:39 autogen_display_resources.o
-rw-r--r--   1 jerryneedell  staff  11345754 Aug 31 13:40 firmware.elf.map
-rwxr-xr-x   1 jerryneedell  staff  14078984 Aug 31 13:40 firmware.elf
-rw-r--r--   1 jerryneedell  staff   1176045 Aug 31 13:40 firmware.hex
-rwxr-xr-x   1 jerryneedell  staff    421936 Aug 31 13:40 firmware.bin
-rw-r--r--   1 jerryneedell  staff    836608 Aug 31 13:40 firmware.uf2
```

T2brozz, 2020-08-31:
`
GEN build-makerdiary_nrf52840_mdk_usb_dongle-s140/genhdr/mpversion.h
PreProcess ld_defines.c
make: arm-none-eabi-gcc: No such file or directory
make: *** [../../py/mkrules.mk:63: build-makerdiary_nrf52840_mdk_usb_dongle-s140/ld_defines.pp] Error 127
make: *** Waiting for unfinished jobs....
`
for me not, 
and its 
> make BOARD=makerdiary_nrf52840_mdk_usb_dongle SD=s140 V=1 -j4 hex

i have tried it with bin but both dosen't work.

I use manjaro and gcc 10.1.0

jerryneedell, 2020-08-31:
Just to be sure, did you try the command I gave above -- no "hex"and no "bin"

T2brozz, 2020-08-31:
yeah i have tried your command im running it in ciruitpython/ports/nrf

jerryneedell, 2020-08-31:
OK -- I don't have any better suggestions - sorry.

dhalbert, 2020-08-31:
`make: arm-none-eabi-gcc: No such file or directory` is pretty suspicious. Can you run `arm-none-eabi-gcc` from the command line? Did you download the toolchain file and unpack it? See https://learn.adafruit.com/building-circuitpython

T2brozz, 2020-08-31:
nop im installing it right now.
But is it possible to include that information in that Readme ?

closed 2020-08-31 by T2brozz