adafruit/circuitpython #3364: Bug in time.localtime(t) CircuitPython?

powersoft, 2020-09-01:
I wont to convert a unixtimestamp to localtime in circuit python.
This is the simple code:

unix_correction = 946684800 # correct to 01-01-2000
timezone = 7200
t=1598887049

def convertUnixTime(t,timezone):
dummy=time.localtime(t-unix_correction+timezone)
return "%4d-%02d-%02d %02d:%02d" % (dummy[0],dummy[1],dummy[2],dummy[3],dummy[4])

Running it I get the error message:

"timestamp out of range for platform time_t"

could not found a solution for this problem (it is working well in micropython!)

thank for any help.
cheers,
Jan

nk521, 2020-09-03:
What is the expected output of this? @powersoft 

I ran it on desktop python3.8 and it returned `1990-09-01 22:47`. Is this the expected output?

nk521, 2020-09-04:
This code works in micropython because there is no inbuilt "unix epoch correction" there. 
[time_localtime](https://github.com/micropython/micropython/blob/073b9a5eb888b26be7ce4c65e9649b6c3857ba39/ports/esp32/modutime.c#L37) calls `timeutils_seconds_since_2000_to_struct_time` compare this with circuitpython's timeutils [timeutils_seconds_since_epoch_to_struct_time](https://github.com/adafruit/circuitpython/blob/main/lib/timeutils/timeutils.c#L161) function which gets called during `localtime`

```
void timeutils_seconds_since_epoch_to_struct_time(mp_uint_t t, timeutils_struct_time_t *tm) {
    t -= EPOCH1970_EPOCH2000_DIFF_SECS;
    timeutils_seconds_since_2000_to_struct_time(t, tm);
}
```

second line does what you're trying to do. So you can remove `unix_correction` from your code and it will work fine.

garethhcoleman, 2021-02-27:
@powersoft are you happy that the explanation / solution given closes this issue?

nk521, 2022-03-30:
I just noticed that this issue is still open. Can we close this issue as there's no activity going on here.

DJDevon3, 2022-08-17:
This is no longer an issue. You can easily convert Unix epoch timestamps and timezone offset manually with something like this:

`      def _format_datetime(datetime):
    return "{:02}/{:02}/{} {:02}:{:02}:{:02}".format(
        datetime.tm_mon,
        datetime.tm_mday,
        datetime.tm_year,
        datetime.tm_hour,
        datetime.tm_min,
        datetime.tm_sec,
    )

unix_time = 1660761378
tz_offset_seconds = -14400  # NY Timezone

get_timestamp = int(unix_time + tz_offset_seconds)
        current_unix_time = time.localtime(get_timestamp)
        current_struct_time = time.struct_time(current_unix_time)
        current_date = "{}".format(_format_datetime(current_struct_time))

        print("Timestamp:", current_date) `

Just because the syntax doesn’t work as it does in micropython doesn’t mean it’s necessarily a bug. Trying to help plug holes with stuff tagged as good first issue.

skerr92, 2022-10-16:
Given @DJDevon3 's solution above, this could probably be closed.

dhalbert, 2022-10-16:
Thanks! Closing.

closed 2022-10-16 by dhalbert