adafruit/circuitpython #3365: hid_report_descriptors

CatalinPopovici, 2020-09-01:
I want to add usb_hid for a braille display. 
In the hid_report_descriptors I have added my own report descriptor and I noticed that for the output reports I can't have a report count larger than 8 (0x08). 
Is there a limitation for the report count? 
Here is the output report I am talking about, there are also other output reports but this is the only one that exceeds 0x08.

            # SET DOTS
            0x09, 0x41,		    # 08|1   , Usage      (vendor defined)
            0x85, 0x01, 		    # 0x09, 0x05, 08|1   , Usage      (vendor defined)
            0x15, 0x00,		    # 14|1   , Logical Minimum(0 for signed byte?)
            0x26, 0xFF, 0x00,	    # 24|1   , Logical Maximum(255 for signed byte?)
            0x75, 0x08,		    # 74|1   , Report Size(8) = field size in bits = 1 byte
            0x95, 0x10, 		    # no of Braille cells
            0x91, 0x02,		    # 90|1   , OUT report (Data,Variable, Absolute)

CatalinPopovici, 2020-09-01:
To test the output report I am using a tool called SimpleHIDWrite. The error I get is from this tool but it's not a specific error, rather a generic one: "A device attached to the system is not functioning"..  but if I get the report count down to 0x08 in the report descriptor and then test it with SimpleHIDWrite it works.
My thoughts were that maybe there is a maximum limitation for report counts which you can input in the hid_report_descriptors.py

CatalinPopovici, 2020-09-02:
I realized that I can use USBlyzer together with SimpleHIDWrite (the application for which we do not have the source code).
So I get the following response in the USBlyzer when sending the set dots output report with report count larger than 8:
URB Control Transfer failed
Device Object	0000005b
Driver Object	usbccgp

URB Function	URB_FUNCTION_CONTROL_TRANSFER
URB Status	USBD_STATUS_STALL_PID

Endpoint 0	Default Control

Request	Set Report
Report Type	Output
Report ID	1
Report Length	41

dhalbert, 2020-09-02:
@hathach Could this be a tinyusb issue?

CatalinPopovici, 2020-09-02:
It could be.. I mean that's the first place I have searched after I discovered that it worked with a smaller or equal report count than 0x08 but I couldn't find anything related to any limitation of the report count or something that could go wrong if you input a bigger report count size.

CatalinPopovici, 2020-09-03:
Seems like the problem was indeed caused by tinyusb. 
In "lib/tinyusb/src/class/hid/hid_device.h" there is a: 

#define CFG_TUD_HID_BUFSIZE     16

If you replace 16 with 64 it fixes the problem meaning I can set a report count over 0x08 in my descriptor.

dhalbert, 2020-09-03:
@CatalinPopovici Could you file an issue on tinyusb? Thanks.

hathach, 2020-09-04:
ah this is not a bug, but rather a usage/configuration. By default the hid bufsize is 16, which allow 1 byte for report id for composite hid, and 1 byte for report itself. TinyUSB is default to run in most common scenario with smallest footprint, so yeah you need to tweak its configuration to have more advanced features.

closed 2020-09-11 by dhalbert