adafruit/circuitpython #3367: CLUE display issues with 6.0.0-alpha3

jerryneedell, 2020-09-01:
With 6.0.0-Alpha3 and later, on a CLUE I am not seeing the REPL displayed on the screen.
I also tried running the  'spirit level' demo does not display anything -- no errors, just no display.
works ok with CP 5.3.1
and
6.0.0.alpha2

The REPL is woking OK in the terminal session, just not echoing to the display.

jerryneedell, 2020-09-01:
PyPortal works normally with 6.0.0.alpha3

jerryneedell, 2020-09-01:
It broke between these 2 builds from S3

last working build
```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-alpha.2-432-gfe73cfb92 on 2020-08-27; Adafruit CLUE nRF52840 Express with nRF52840
>>> 
>```

not working
```

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-alpha.2-434-g610002724 on 2020-08-27; Adafruit CLUE nRF52840 Express with nRF52840
>>> 
>>>
```

jerryneedell, 2020-09-01:
this commit made some changes to code common to all ports - no idea why to would impact CLUE and not PyPortal though  ... https://github.com/adafruit/circuitpython/commit/a15f948a406b498691bbc41645c95d99aa1cd458

Nothing in the other commits jumps out as a likely cause.

jerryneedell, 2020-09-01:
hmm weird -- now ```Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-alpha.2-432-gfe73cfb92 on 2020-08-27; Adafruit CLUE nRF52840 Express with nRF52840
>>> 
>```  is not working any more ....

jerryneedell, 2020-09-01:
Repeated the search and now find that the last working version was

Adafruit CircuitPython 6.0.0-alpha.2-26-gb7e6bf959 on 2020-07-25; Adafruit CLUE nRF52840 Express with nRF52840

and first broken version was

Adafruit CircuitPython 6.0.0-alpha.2-43-g7ab5c520e on 2020-07-25; Adafruit CLUE nRF52840 Express with nRF52840

HOWEVER when I reloaded 6.0.0-alpha.2-43-g7ab5c520e worked for REPL, but not when I ran a program thatt used the display it did display but with odd behavior -- after that it no longer worked for REPL either.
so  6.0.0-alpha.2-43-g7ab5c520e appears to be causing problems that get worse...

6.0.0-alpha.2-26-gb7e6bf959  appears to be stable

jerryneedell, 2020-09-01:
Reverting the changed in #3190 seems to fix it.

```
[Jerry-desktop-mini:circuitpython/ports/nrf] jerryneedell% git diff
diff --git a/ports/nrf/Makefile b/ports/nrf/Makefile
index 3fef68e88..1fdf79958 100755
--- a/ports/nrf/Makefile
+++ b/ports/nrf/Makefile
@@ -89,13 +89,11 @@ ifeq ($(DEBUG), 1)
   CFLAGS += -ggdb3
   OPTIMIZATION_FLAGS = -Og
 else
-  OPTIMIZATION_FLAGS ?= -O2
-  CFLAGS += -DNDEBUG -ggdb3
+  CFLAGS += -Os -DNDEBUG -ggdb3
   CFLAGS += -flto -flto-partition=none
 endif
 
 # option to override compiler optimization level, set in boards/$(BOARD)/mpconfigboard.mk
-CFLAGS += $(OPTIMIZATION_FLAGS)
 
 CFLAGS += $(INC) -Wall -Werror -std=gnu11 -nostdlib -fshort-enums $(BASE_CFLAGS) $(CFLAGS_MOD) $(COPT)
 
```
now works on the tip of main 6.0.0-alpha.3-62-gf0e60da51-dirty

jepler, 2020-09-01:
A similar fix was offered at  #3249 but closed without merge because @dhalbert filed #3281 and @DavePutz reported that it fixed #3247.

Were any other nrf-with-display boards tested?  This reminds me, I had trouble with the Sharp Memory Display corrupting random lines of the display with nrf52840 and not on other micros.  I chalked it up to the particular device being a Particle Xenon rather than an Adafruit board and moved on with a different board (that happened to be non-nRF)

DavePutz, 2020-09-01:
Issue #3296 is another case where compiling with -O2 optimization on the Clue broke something (in that case, BLE). @dhalbert has mentioned that these sorts of issues are most often due to a bug such as an uninitialized variable or  something that should be volatile is not. I have spent a lot of time looking at #3296, and have traced it down to compiling ports/nrf/common-hal/busio/SPI.c with -O2 causing the BLE issue. But, I have not found anything in that source file that would seem to fit Dan's categories. Given the number of issues compiling with -O2 on the Clue seems to be causing, perhaps we should re-visit #3249?

jerryneedell, 2020-09-02:
this keeps getting stranger -- With the build i crated yesterday - reverting the optimization settings,  the CLUE works normally as long as it is connected to the computer USB port. If I power it via a battery alone, the Display does not work. The program in code.py runs, but nothing is displayed. If I go back to CP 5.3.1  it works normally via battery....  It seems like something in the display is requiring the USB connection to be active...
I'm confused!

DavePutz, 2020-09-02:
I can reproduce this as well. But, I don't think we can blame this one on
the -O2 optimization. I tried the
first build after that change went in (7ab5c52
<https://adafruit-circuit-python.s3.amazonaws.com/bin/clue_nrf52840_express/en_US/adafruit-circuitpython-clue_nrf52840_express-en_US-20200725-7ab5c52.uf2>)
and the display still works with the battery. I'll see if I can
narrow down which commit broke this functionality.

DavePutz, 2020-09-03:
Seems that the cause of this is PR #3244. The intent was to turn off QSPI
when USB was not connected.
Looks like this is interfering with the display on the Clue ??

DavePutz, 2020-09-11:
@xiongyihui - what was your intent with the PR #3244 patch in regard to displays? Right now I see that qspi_enable() is being called during startup after  qspi_disable(); but the Clue display does not come back on. I see that you did not use the NRF provided nrf_qspi_disable() to turn off QSPI, but you did use nrf_qspi_enable() to turn it back on. Was there a reason for this?  I did find out the commenting out the  
` NRF_QSPI->TASKS_DEACTIVATE = 1;`
line in qspi_disable() keeps the display on, but I don't know what impact that has when you want QSPI off. I' just trying to understand what's going on so that we can let the Clue display work on battery power.

jerryneedell, 2020-09-11:
This issue appears to have been resolved -- at least, it works on my CLUE now ... should this be closed?

DavePutz, 2020-09-11:
Hmm, interesting that yours is now working. I just pulled and built the
latest main (commit 9256e6b37)
and I still have nothing on the display when booting the Clue off a
battery, and running some
displayio code still produces nothing.

jerryneedell, 2020-09-11:
Sorry -- it is not working on battery -- I updated my comment above....

DavePutz, 2020-09-15:
@jepler - do you know what the intent was for PR #3244? I asked @xiongyihui but have not received any response. Since qspi_disable() is called in  port_sleep_until_interrupt(), any routine that calls mp_hal_delay_ms() will turn off QSPI. In effect, this disables QSPI by default at boot time when USB is not connected, as several startup routines have short delays in them. The effect on the Clue is to disable the display (which uses SPI) and there does not seem to be any way to turn it back on. Calling qspi_enable() does not turn the display back on. So, do we really want to have the Clue unable to use the display when on battery power?

xiongyihui, 2020-09-16:
QSPI is not SPI. Normally, they don't affect each other. Does SPI share pins with QSPI?

DavePutz, 2020-09-16:
From what I can see SPI and QSPI do not share pins on the Clue.
From boards/clue_nrf52840_express/board.c:
void board_init(void) {
...
        &pin_P0_13, // TFT_DC Command or data
        &pin_P0_12, // TFT_CS Chip select
        &pin_P1_03, // TFT_RST Reset
...
        &pin_P1_05,  // backlight pin

and in supervisor/qspi_flash.c:
    nrfx_qspi_config_t qspi_cfg = {
        .xip_offset = 0,
        .pins = {
            .sck_pin = MICROPY_QSPI_SCK,  // defined as NRF_GPIO_PIN_MAP(0,
19)
            .csn_pin = MICROPY_QSPI_CS,   // defined as NRF_GPIO_PIN_MAP(0,
20)
            .io0_pin = MICROPY_QSPI_DATA0, // defined
as NRF_GPIO_PIN_MAP(0, 17)
...
    qspi_cfg.pins.io1_pin = MICROPY_QSPI_DATA1; // defined
as NRF_GPIO_PIN_MAP(0, 22)
    qspi_cfg.pins.io2_pin = MICROPY_QSPI_DATA2; // defined
as NRF_GPIO_PIN_MAP(0, 23)
    qspi_cfg.pins.io3_pin = MICROPY_QSPI_DATA3; // defined
as NRF_GPIO_PIN_MAP(0, 21)

One thing I have seen in my testing is that in qspi_disable() taking out
the line
        NRF_QSPI->TASKS_DEACTIVATE = 1;
allows the display to work. I have not found in the nrf52840 documentation
a mention about
doing this. Is it required?

tannewt, 2020-09-16:
The intent of #3244 was to turn off the QSPI peripheral while going to sleep so that the high speed clock would be turned off and save a bunch of power. It should start back up at the next file access.

The SPI to the display should be separate from QSPI. Perhaps the SPI isn't keeping the high speed clock active?

DavePutz, 2020-09-16:
I did some further testing, and it isn't only the display that's affected. An externally connected SPI device works with USB power but
does nothing with battery power. Logic analyzer shows no activity on the pins to the device at all. So, it looks like all SPI is affected when qspi is disabled.

xiongyihui, 2020-09-17:
@DavePutz I tested it on nrf52840 m.2 devkit which has a SPI display. Turning off QSPI doesn't affect the SPI display.

[image]

xiongyihui, 2020-09-17:
My test code is:

```

import time
import board
import digitalio

@micropython.asm_thumb
def mem(r0):
    ldr(r0, [r0, 0])

@micropython.asm_thumb
def mem_write(r0, r1):
    str(r1, [r0, 0])

peripherals = {
'RADIO': 0x40001000,
'UART0': 0x40002000,
'UARTE0': 0x40002000,
'SPI0': 0x40003000,
'SPIM0': 0x40003000,
'SPIS0': 0x40003000,
'TWI0': 0x40003000,
'TWIM0': 0x40003000,
'TWIS0': 0x40003000,
'SPI1': 0x40004000,
'SPIM1': 0x40004000,
'SPIS1': 0x40004000,
'TWI1': 0x40004000,
'TWIM1': 0x40004000,
'TWIS1': 0x40004000,
'NFCT': 0x40005000,
'GPIOTE': 0x40006000,
'SAADC': 0x40007000,
'TIMER0': 0x40008000,
'TIMER1': 0x40009000,
'TIMER2': 0x4000A000,
'RTC0': 0x4000B000,
'TEMP': 0x4000C000,
'RNG': 0x4000D000,
'ECB': 0x4000E000,
'AAR': 0x4000F000,
'CCM': 0x4000F000,
'WDT': 0x40010000,
'RTC1': 0x40011000,
'QDEC': 0x40012000,
'COMP': 0x40013000,
'LPCOMP': 0x40013000,
'EGU0': 0x40014000,
'SWI0': 0x40014000,
'EGU1': 0x40015000,
'SWI1': 0x40015000,
'EGU2': 0x40016000,
'SWI2': 0x40016000,
'EGU3': 0x40017000,
'SWI3': 0x40017000,
'EGU4': 0x40018000,
'SWI4': 0x40018000,
'EGU5': 0x40019000,
'SWI5': 0x40019000,
'TIMER3': 0x4001A000,
'TIMER4': 0x4001B000,
'PWM0': 0x4001C000,
'PDM': 0x4001D000,
'ACL': 0x4001E000,
'NVMC': 0x4001E000,
'PPI': 0x4001F000,
'MWU': 0x40020000,
'PWM1': 0x40021000,
'PWM2': 0x40022000,
'SPI2': 0x40023000,
'SPIM2': 0x40023000,
'SPIS2': 0x40023000,
'RTC2': 0x40024000,
'I2S': 0x40025000,
'FPU': 0x40026000,
'USBD': 0x40027000,
'UARTE1': 0x40028000,
'QSPI': 0x40029000,
'CC_HOST_RGF': 0x5002A000,
'CRYPTOCELL': 0x5002A000,
'PWM3': 0x4002D000,
'SPIM3': 0x4002F000
}

def show_peripherals():
    print('\nPeripherals:')
    for name, addr in peripherals.items():
        value = mem(addr + 0x500)
        if value:
            print('{}: {}'.format(name, value))

show_peripherals()

button = digitalio.DigitalInOut(board.USR_BTN)
button.direction = digitalio.Direction.INPUT
button.pull = digitalio.Pull.UP

last_value = button.value

while True:
    if last_value != button.value:
        last_value = button.value
        if last_value == 0:
            # print(last_value)
            show_peripherals()
    time.sleep(0.01)
```

xiongyihui, 2020-09-17:
On issue #3360, it seems turning off QSPI doesn't affect the SPI display too.

tannewt, 2020-09-17:
This seems very weird that USB vs battery would be different. Does the Clue wake up OK in this case? It seems like it must be related to the regulator: https://cdn-learn.adafruit.com/assets/assets/000/087/877/original/adafruit_products__CLUE_sch.png?1580423083 The only difference it will see is 5v from USB vs <4.2V from the battery.

@xiongyihui Is your schematic open source? I wonder how the power circuitry differs.

xiongyihui, 2020-09-18:
The schematic is at https://github.com/makerdiary/nrf52840-m2-devkit

DavePutz, 2020-09-18:
Just as a point of information, We tried a  Feather nRF52840 Express with a
TFT Featherwing and see the
same issue - the display lights up (backlight?) but does not show any
outputs from the code.py script. So, this
does not look like just a Clue issue.

tannewt, 2020-09-18:
Thanks @xiongyihui! The power does look different on the feather as well.

https://cdn-learn.adafruit.com/assets/assets/000/068/545/original/circuitpython_nRF52840_Schematic_REV-D.png?1546364754

Looks like vbat goes through the diode of the transistor before the regulator.

@DavePutz do other things work ok? This is so weird that it works on USB but not battery power.

DavePutz, 2020-09-18:
I determined that the calls to  common_hal_time_delay_ms() in common_hal_displayio_fourwire_reset are the reason for this issue.  Since common_hal_time_delay_ms() ends up calling port_sleep_until_interrupt(), which is where qspi_disable() is called; I suspect there must be some timing issue with disabling QSPI while the fourwire is being initialized. @tannewt , I see those delays were added in a commit for e-paper. If these delays are needed for e-paper to work, is there some other way to accomplish that? Maybe use mp_hal_delay_us()?

tannewt, 2020-09-18:
@DavePutz why would that only break when on battery power?

DavePutz, 2020-09-18:
@tannewt - qspi_disable() only does the disabling when USB is not connected.

cjsieh, 2020-09-18:
There are 2 test cases here.

1.  Boot on battery power only.

2.  Boot on USB Power  with battery plugged in and then remove USB power
without reseting.

Both the CLUE and Feather with TFT work ok for test case 2 .
Both fail for test case 1.

xiongyihui, 2020-09-19:
@DavePutz
You can try to disable QSPI when USB is connected. Only need to change the line at https://github.com/adafruit/circuitpython/pull/3244/files#diff-46ca91d54ad3cf6fcc4906eecc666ebaR45 to

```
    if (NRF_QSPI->ENABLE) {
```

I tried it. The SPI display still works when QSPI is off and USB is connected.

DavePutz, 2020-09-20:
OK, here is the issue the way I see it. If the two calls to common_hal_time_delay_ms(1) in common_hal_displayio_fourwire_reset() are replaced with mp_hal_delay_us(1000) then the SPI TFT displays on Clue and Feather work fine. However, an e-paper display on the Feather still does not work, due to the call to common_hal_time_delay_ms() in send_command_sequence(). The delay there looks like it might be for quite a long time (up to 500ms). Assuming that the e-paper displays need that delay, I can think of two alternatives. Either send_command_sequence() gets modified in some way to not call common_hal_time_delay_ms(); or we need to take the qspi_disable() call out of port_sleep_until_interrupt() (possibly by making it a call of its own, so it is a deliberate choice and not the default?) @tannewt , @xiongyihui  - any thoughts?

tannewt, 2020-09-21:
Is the sleep too short when qspi is disabled? I still don't understand the connection between disabling QSPI and SPI display. Maybe there is a clock glitch on the SPI lines when sleeping?

DavePutz, 2020-09-21:
I do not understand why disabling QSPI affects SPI at all. And yet, it
clearly does.
There are no pin conflicts that I have found, it just looks like some sort
of timing
issue or conflict on the nrf chip. The only issue with sleeping while
initializing an
SPI device is that it will end up calling qspi_disable(); it doesn't matter
how long or short
the sleep is.

xiongyihui, 2020-09-22:
`common_hal_time_delay_ms`  will delay uncertain time because of [background tasks](https://github.com/adafruit/circuitpython/blob/main/supervisor/shared/tick.c#L151).
If initializing these display is time-critical, `mp_hal_delay_us` is a better option.

tannewt, 2020-09-22:
@xiongyihui I doubt those delays are broken if it takes longer than expected. Is there some way for the delay to shorten with the qspi disable?

DavePutz, 2020-09-22:
I have not determined the root cause of the qspi_disable interfering with
regular SPI
peripherals. We can't use common_hal_delay_ms() with a value less than 1 in
any
case. I'm not sure if it is the length of the delay causing the issue, or
something in the
SPI/QSPI subsystems that have some sort of dependency on each other. I do
know that
once the SPI display is initialized properly, subsequent calls to
qspi_disable() do not have any
effect on it.

xiongyihui, 2020-09-23:
I think `qspi_disable() `  takes less than 10 us. The uncertain delay is caused by [background tasks](https://github.com/adafruit/circuitpython/blob/main/supervisor/shared/tick.c#L151), not by `qspi_disable() `.

DavePutz, 2020-09-24:
It seems as though the fix for  nRF52840 Revision 1 Errata V1.4 - 3.8 in qspi_disable() is doing something to the SPI. If I comment out the two lines from that errata,  SPI works correctly.  Then running on battery  I do not see QSPI in the peripheral list generated by the script above from @xiongyihui . However, I don't have a way of measuring any differences in the current draw from the battery with or without that code.  I have not been able to find any reference saying what the line
`*(volatile uint32_t *)0x40029054 = 1;`
actually does.

tannewt, 2020-09-25:
@DavePutz Thank you for finding that! That's enough rood cause for me.

closed 2020-09-28 by tannewt