adafruit/circuitpython #3381: SPI bus issue on STM32F405RG.

water5, 2020-09-08:
I run two SPI relate library on Feather STM32F405 Express, the SPI bus look like incorrect.

more detail:
[https://forums.adafruit.com/viewtopic.php?f=60&t=169223](url)
[https://github.com/2bndy5/CircuitPython_nRF24L01/issues/13](url)

tannewt, 2020-09-08:
@water5 Please describe the issue here.

water5, 2020-09-10:
On CircuitPython Release 5.3.1 , when run 
adafruit_wiznet5k examples (from: [https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases](https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases))
and
nRF24L01 examples (from: [https://github.com/adafruit/CircuitPython_Community_Bundle/releases](https://github.com/adafruit/CircuitPython_Community_Bundle/releases))

the NeoPixel LED goes off (in REPL it should be steady white), REPL stop respone, but the CIRCUITPY drive work still.

--------------------------------
On CircuitPython 6.0.0 Alpha-3,

> the NeoPixel LED goes off (in REPL it should be steady white), REPL stop respone, but the CIRCUITPY drive work still.

This issue fixed.
but nRF24L01 library (from: [https://github.com/adafruit/CircuitPython_Community_Bundle/releases](https://github.com/adafruit/CircuitPython_Community_Bundle/releases))
run incorrect still, figuring out.

tannewt, 2020-09-10:
Ok thanks for the update! Please let us know when you've figured out the issue on 6.0.0-alpha.3. Sounds like the fixed one issue already.

jepler, 2020-09-10:
The library uses spi `readfrom` with the `write_value` parameter.  This does not presently work on stm32's implementation of spi in circuitpython (arbitrary/uncontrolled data is sent instead).  #3176 is an existing issue about this problem, and it is possible that this explains why the library does not work on stm32f405.

hierophect, 2020-09-10:
@jepler I thought that issue was exclusively specific to SD cards?

2bndy5, 2020-09-11:
@hierophect good eye! I didn't see that issue.

@water5 I will look into updating the library with a fix that doesn't involve using the `write_value` parameter in any `spi.readfrom()` calls.

hierophect, 2020-09-11:
@2bndy5 I'm also working on a fix for the write_value issue itself, but please let me know if removing those calls helps at all - my original research indicated that issue was exclusively for SD cards as per an oversight by ST regarding SD specific spec requirements, but if it affects other devices that makes the fix much higher priority.

water5, 2020-09-11:
@hierophect Since CircuitPython 6.0.0 Alpha-3, added `sdioio` module to access SD Card, I think the `sdioio` is major interface to access SD Card since 6.0.0 (sdioio faster than SPI), unless want to save IO or other reason to specify use SPI to access SD Card, or port to none SDIO interface STM32 chip?

hierophect, 2020-09-11:
@water5 SDIO is generally preferable, but it is not available for some breakout boards like the Adalogger, which assume SD card access over SPI. I brought it up because the "write_value" issue Jepler mentioned (a problem with dictating what value the SPI write line is idling at while a read is occurring) is most commonly found on SD cards, but it also appears to affect some sensors like this one which require specific data to be transferred while reading.

To be honest, I see some redundancy here between using `write_value` in `readinto()`, as opposed to simply using the write_readinto() call for cases like this. I assume it is performance? It's probably worth putting up some kind of warning for the authors of sensor libraries, since the NRF and CXD ports also don't implement this feature and it's easily avoided by just using the full transfer function instead.

2bndy5, 2020-09-12:
I've been slowly optimizing the nrf24l01 library, and I inadvertently already replaced the `spi.readinto()` calls with `spi.write_readinto()` calls (a while ago). Now I have a reason to expedite debugging my [lite-beta branch](https://github.com/2bndy5/CircuitPython_nRF24L01/tree/lite-beta) for a new release. 

@hierophect I don't own a STM32F405 Feather, so I'm depending on @water5 to test my changes on that board. Initial tests of my changes work well with the nRF24L01 transceiver, although I'm using an Rpi2 and ItsyBitsy M4.

More development helping @water5 is downstream on the original [nrf24 lib issue](https://github.com/2bndy5/CircuitPython_nRF24L01/issues/13).

2bndy5, 2020-09-12:
@water5 has notified me that my changes (replacing `readinto()` with `write_readinto()`) are working on the STM32F405 Feather. This issue can be closed. THANK YOU everyone for your help. Go Team!

water5, 2020-09-12:
This issue come from when I test [nRF24L01 library](https://github.com/2bndy5/CircuitPython_nRF24L01) , that library improved now, reason is: [#3176 ](https://github.com/adafruit/circuitpython/issues/3176)

Thanks jepler remind : "This does not presently work on stm32's implementation of spi in circuitpython (arbitrary/uncontrolled data is sent instead)."
Thanks all participants.

closed 2020-09-12 by water5