adafruit/circuitpython #3383: SAM E54 Xplained: microcontroller.cpu.voltage hangs

jepler, 2020-09-08:
Except possibly with very long delays, `microcontroller.cpu.voltage` hangs on the SAM E54 Xplained board.  Sometimes it will read after previously hanging and then being reset, but this testing procedure would reliably hang for me:
```
>>> import microcontroller; microcontroller.cpu.voltage; microcontroller.cpu.temperature; microcontroller.cpu.voltage; microcontroller.cpu.temperature; 
3.32308
33.1896
[hangs]
```

Tested with a local branch somewhere a few commits after alpha.3, but this is not likely to be a new problem, just something that hadn't been tested before.

jepler, 2020-09-08:
We could simply disable it, as this patch would do:
```From 5c669e0fa3b681bac9d4db5361c5557c2fff0ece Mon Sep 17 00:00:00 2001
From: Jeff Epler <jepler@gmail.com>
Date: Tue, 8 Sep 2020 09:10:31 -0500
Subject: [PATCH] Straw proposal: disable microcontroller.cpu.voltage on all
 SAM E54

---
 ports/atmel-samd/common-hal/microcontroller/Processor.c | 8 +++++++-
 ports/atmel-samd/mpconfigport.h                         | 7 +++++++
 2 files changed, 14 insertions(+), 1 deletion(-)

diff --git a/ports/atmel-samd/common-hal/microcontroller/Processor.c b/ports/atmel-samd/common-hal/microcontroller/Processor.c
index 199f19354..029cb1d34 100644
--- a/ports/atmel-samd/common-hal/microcontroller/Processor.c
+++ b/ports/atmel-samd/common-hal/microcontroller/Processor.c
@@ -61,6 +61,8 @@
  * POSSIBILITY OF SUCH DAMAGE.
  */
 
+#include <math.h>
+
 #include "py/mphal.h"
 #include "common-hal/microcontroller/Processor.h"
 
@@ -276,6 +278,9 @@ float common_hal_mcu_processor_get_temperature(void) {
 }
 
 float common_hal_mcu_processor_get_voltage(void) {
+#if MICROCONTROLLER_VOLTAGE_DISABLE
+    return NAN;
+#else
     struct adc_sync_descriptor adc;
 
     static Adc* adc_insts[] = ADC_INSTS;
@@ -320,6 +325,7 @@ float common_hal_mcu_processor_get_voltage(void) {
     adc_sync_deinit(&adc);
     // Multiply by 4 to compensate for SCALEDIOVCC division by 4.
     return (reading / 4095.0f) * 4.0f;
+#endif
 }
 
 uint32_t common_hal_mcu_processor_get_frequency(void) {
diff --git a/ports/atmel-samd/mpconfigport.h b/ports/atmel-samd/mpconfigport.h
index 2a2428d67..f185142e4 100644
--- a/ports/atmel-samd/mpconfigport.h
+++ b/ports/atmel-samd/mpconfigport.h
@@ -75,6 +75,9 @@
 #define MICROPY_PY_SYS_PLATFORM                     "MicroChip SAMD51"
 #elif defined(SAME54)
 #define MICROPY_PY_SYS_PLATFORM                     "MicroChip SAME54"
+#ifndef MICROCONTROLLER_VOLTAGE_DISABLE
+#define MICROCONTROLLER_VOLTAGE_DISABLE (1)
+#endif
 #endif
 #define SPI_FLASH_MAX_BAUDRATE 24000000
 #define MICROPY_PY_BUILTINS_NOTIMPLEMENTED          (1)
@@ -246,4 +249,8 @@
     CIRCUITPY_COMMON_ROOT_POINTERS \
     mp_obj_t playing_audio[AUDIO_DMA_CHANNEL_COUNT];
 
+#ifndef MICROCONTROLLER_VOLTAGE_DISABLE
+#define MICROCONTROLLER_VOLTAGE_DISABLE (0)
+#endif
+
 #endif  // __INCLUDED_MPCONFIGPORT_H
-- 
2.20.1

```

DavePutz, 2020-09-08:
Is this possibly related to Issue #2893/ PR #3052? Perhaps a longer delay is needed in common_hal_mcu_processor_get_voltage()?

dhalbert, 2020-09-08:
@DavePutz: @jepler mentioned it on discord this morning, and we discussed it including that PR. He tried a longer delay, but it didn't seem to help. I have a SAME54 dev board also and will try some things eventually.

The errata say that the TEMP functionality is broken, without giving any details. I have two different boards with identical chips but different date codes: one was hanging before I added the delay, and one was fine. So it may vary by sample.

tannewt, 2021-07-15:
Fixed by #3383. Thanks @DavePutz

closed 2021-07-15 by tannewt