adafruit/circuitpython #3389: Manual Safe Mode on esp32s2

microDev1, 2020-09-10:
This  is a follow-up to [9-9-20 (CPY Day) esp32s2 Deep Dive](https://youtu.be/0xyeEGdmLKE?t=2662) in which esp32s2 `manual safe mode` support was briefly discussed.

The esp32s2 posses a challenge for `manual safe mode` implementation as it lacks a dedicated reset pin instead the reset button is connected to `CHIP_PU` which when pulled low turns off the power to the chip thus causing a `hard reset`.   

A solution to this problem as discussed during the deep dive is to use `IO0` (boot button) and check its state during CPY startup in order to trigger a `soft reboot` keeping the ram alive.

I was able to boot into safe mode by pressing the boot button momentarily during 700ms loop which is indicated by a steady yellow color of the rgb led on startup. 

As of now I am just returning `MANUAL_SAFE_MODE` instead of `NO_SAFE_MODE` in order to enter safe mode and haven't really figured how this will work alongside other CPY ports and the soft reboot with status stored in ram. 

Code snippet from [safe_mode.c ](https://github.com/microDev1/circuitpython/commit/f2ba3693ca1f8fe44f4d7744f19e52fe49ffebe7).
Tested on [microS2](https://circuitpython.org/board/microdev_micro_s2/).

```
digitalio_digitalinout_obj_t boot_button;
common_hal_digitalio_digitalinout_construct(&boot_button, MICROPY_HW_BUTTON);  //#define MICROPY_HW_BUTTON (&pin_GPIO0) 
common_hal_digitalio_digitalinout_switch_to_input(&boot_button, PULL_UP);

uint64_t start_ticks = supervisor_ticks_ms64();
uint64_t diff = 0;

while (diff < 700) {
       #ifdef MICROPY_HW_LED_STATUS
       // Blink on for 100, off for 100, on for 100, off for 100 and on for 200
       common_hal_digitalio_digitalinout_set_value(&status_led, diff > 100 && diff / 100 != 2 && diff / 100 != 4);
       #endif
        
       if (!common_hal_digitalio_digitalinout_get_value(&boot_button)) {
           return BOARD_USER_SAFE_MODE;
       }

       diff = supervisor_ticks_ms64() - start_ticks;
}  
```

tannewt, 2020-09-10:
Awesome! Thank you!

I think the easiest thing is to add a board define `CIRCUITPY_BOOT_BUTTON` that can be used to include the code you've added.

Want to make a PR? Thanks!

microDev1, 2020-09-14:
@tannewt I'll leave this issue open until `soft_reboot` approach is also implemented.

tannewt, 2020-09-14:
So we need to:

- [ ] Figure out how to reset from software and save 4-bytes across the reset.
- [ ] Detect brown outs and safe mode when it happens.
- [ ] Intercept fatal error from the IDF and safe mode.

PR #3808 (2020-12-09): Enter safe mode after panic or brownout
Uses the IDF's reset reason. Does nothing before reset.

Fixes #3389
commit d0a806d7 (2020-12-09): Enter safe mode after panic or brownout

Uses the IDF's reset reason. Does nothing before reset.

Fixes #3389
closed 2020-12-09 by tannewt