adafruit/circuitpython #3391: Extension: Touch-based graphical user elements for displays

kmatch98, 2020-09-10:
### Objective: To define a CircuitPython friendly method for an easy-to-use set of touch-based graphical controls for displays

Recent discussions highlight the desire for easy-to-use screen based Graphical User Interface elements for CircuitPython displays. 

These requests range from a desire to relate touch input to existing displayio graphics elements (https://github.com/adafruit/circuitpython/issues/1598) and to @makermelissa 's request for [radio buttons and scrolling text boxes](https://youtu.be/JqK911g7O7Q?t=3226).

This issue is created to:
- highlight existing CircuitPython projects that already are executing touch-based user interfaces
- identify python-based GUI definitions that could be relevant
- have a place to capture wish-list of features and implementation ideas
- divvy up work for those who are interested in contributing

#### Existing CircuitPython projects related to GUI elements

- [CircuitPython Touch screen library](https://github.com/adafruit/Adafruit_CircuitPython_Touchscreen).  This is a driver for interfacing with a resistive touchscreen.  Read the location and pressure with the `adafruit_touchscreen.Touchscreen.touch_point` function.
- [PyPortal graphical interface learn guide with splash screen, buttons, pages, bitmaps and text.](https://learn.adafruit.com/making-a-pyportal-user-interface-displayio/overview) by [Richard Albritton](https://learn.adafruit.com/users/richa1)
- [PyPortal MQTT sensor node with buttons](https://learn.adafruit.com/pyportal-mqtt-sensor-node-control-pad-home-assistant) by [Richard Albritton](https://learn.adafruit.com/users/richa1)
- [Pull down "hamburger" menu](https://gist.github.com/FoamyGuy/b11390a020d213b1ee15fb5f02eacfff) and [video]() by @foamyguy
- [Display_layout library that uses JSON files to define graphical page layouts](https://github.com/FoamyGuy/circuitpython_display_layouts) by @foamyguy.  Here is a [related Adafruit forum post.](https://forums.adafruit.com/viewtopic.php?f=60&t=168866#p826821)

Other comments from the discord:
- From @makermelissa: 
    - It's kind of like experiencing the Graphical UI all over again and that's been my inspiration on the UI controls. It's really just a bunch of loose ideas at this point. So as you mentioned, radio button and checkbox. Maybe a scrolling text area (which may have been done on a project or two already). A drop down box, list, tree control, maybe groups. I like to look at some old Visual Basic interfaces with available controls for inspiration. I do like the idea of touch events where each element keeps track of it's own touches it receives.
- From @foamyguy: 
     - @MakerMelissa's answer last night has inspired me to make an attempt at Checkbox and Radio buttons next. I am keenly interested in what is possible on the touch event location front as well. It would be great if the system could tell you what got clicked somehow instead of having to pass the touch event object into a function on each control to check if it got hit or not.
    - Thinking outloud a bit. Maybe the systen could provide a Gamepad-like object where each property represents a screen control instead of a IO pin or similar.
- From @hierophect: 
     - do we have a system for a rotating selection list? Like the kind of thing you have on a 3D printer, where a carrot moves down a list to select menu items, can enter submenus and such. And if the list is larger than the screen, it will cycle into new items once you hit the bottom or top?
    - I put together one on Mbed one time, maybe I'll try doing it for circuitpython.  I'll need it for some upcoming things I think

tannewt, 2020-09-11:
@tg-techie Has been interested in this as well.

joeycastillo, 2020-09-14:
@tannewt suggested I chime in with this project I'm working on, [circuitpyui](https://github.com/joeycastillo/circuitpyui). It's a bit basic at this point but has some of the same goals (adding interaction to displayio groups with touch and joystick control). Might be some useful stuff in there, like locating tapped views in a hierarchy and routing events to views that want to respond. There's an example for the PyPortal [here](https://github.com/joeycastillo/circuitpyui/blob/main/examples/PyPortal/demo.py).

TG-Techie, 2020-09-14:
The graphical side of circuitpython has been growing and, in my opinion,  the lack of a unified user input scheme can be felt.

I think adding featureS like these can be broken down into several categories:
- the ui-tree, this is the data structure constructed by the developer of a ui-application that represents the onscreen data. This seems to be more than covered with the facilities provided by displayio. the only functionality it odes not seem to provide is bi-directional tree traversal.
- the action/behavior conventions, these are the names, interfaces, functions, and properties that are used to indicate what functionality an element in the tree has and can perform.
- the ui-loop or "event loop", I have found that this is one of the hardest parts of the system to implement (for several reasons). These loops can include touch-screen or directional("gamepad") user input that is then used to traverse/search the ui-tree and then call actions(as defined above) based on what it finds.

Each of these needs to interact but will need to be defined separately.

A mix of thoughts:
- I think circuit python runs a far too resource-constrained resource environment to use a full model view controller architecture. moreover, MVC setups are often employed because they allow the decoupling of parts of a program so they can be developed in tandem. In most applications of circuitpython the exchange of parallel development time for increased overhead is not needed.
- I agree that a lightweight event loop would is a good idea. Due to to the complexity of touch or "directional"(gamepad) interface loop I think these would be good routines to be moved into the core as they will be very hot.

- the naming and flagging functionality of the action/behavior conventions are relatively easy to change and I would not sweat about it in the beginning. see below for a table of what I've been using. [1]

- I think a good approach to the ui-loop would to be to have three different control loops, says built-in classes, that take input that is specific to their input type and then manipulate the ui-tree differently based on the input. For example:
    ```
    my_disp = Display() #this is psuedo-code just go with it
    my_disp.set_group(the_ui_tree)

    from Displayio.control_loops import TouchControlLoop

    tcl = TouchControlLoop(my_disp)

    # the main event loop
    while True:
        raw_x, raw_y = touchscreen.call_to_hardware()
        is_point_down = touchscreen.other_call_to_hardware()
        x, y = some_math_to_shift_coords(raw_x, raw_y)
        TouchControlLoop.update_point(is_point_down, x, y)
    ```

[1] my current action/behavior conventions (object or class methods from TG-Gui 4):
- pointdown(): a method on a widget (tree element) that when called tells the widget that it is currently "pointed". rather, the widget is currently selected by a list with directional navigation or a finger is currently on top of it in touchscreen navigation. when this method is called the widget is responsible for altering it's color, test, etc to match.
- pointup(): tells the widget to resume its unselected state.
- tap(x, y): executes the arbitrary action attached to that widget.
- alttap(x, y): executes the  arbitrary action for a right click or long press
- various scrolling methods still in flux ... etc

[_] Some background on where I’m coming from:
I've been using circuit python for a few years and have been working on graphical interfaces for most if not all of the time. This, so far, has all been in user-land python because python is os easy to develop in. I've implemented a graphical user interface a few times that have supported directional/"gamepad" navigation and touch navigation.
here are some of the products:
- https://twitter.com/TG_Techie/status/1253948404914405377?s=20
- or some of my older work on the rest of my twitter page https://twitter.com/TG_Techie

kmatch98, 2020-09-15:
Thanks @joeycastillo and @TG-Techie for the great comments and background.  I like how the discussion is already going.  Please forgive me if the words I use are inaccurate with typical terminology and feel free to correct me and clarify anything that I say here.

I've started thinking about the user experience.  I agree with @TG-Techie that a full Model-View-Controller is probably more complex than most new users will be able to create as a first project.  In particular, I'm concerned about having a giant, complicated `if` statement in the loop that deals with all the interactions.

**Simplifying everything:** I suspect if you asked people to draw what they wanted on their GUI, they would tell you where to put the elements and then tell you want each one does.  Can we make it seem as simple as that?

I think there are several layouts that would cover many of the use cases:
- One-page "dashboard" with some display elements and a few controls
- Drill-down list "tree" with some controls at the "leafs", maybe with a home "dashboard" (e.g. 3D printer with a rotary encoder knob) [they call call this a "table" in iOS]
- Multipage displays, with page navigation with arrow buttons, tabs or "hamburger menu" to change pages
- Some combination of the above where the screen is divided into sections or "windows" (e.g. a display section at the top and multipage tabbed section at the bottom)

I like the way that the visual display is created in this example from [PySimpleGUI](https://pysimplegui.readthedocs.io/en/latest/).  In particular, I like that the display elements are defined in a list, and that list could be laid out to give a visual indication of where they should be on the screen.  However, I don't like that a more complex GUI will require a giant `IF` statement to handle all the work.

[image]

I think some controls could be relatively straightforward to define, such as switch triggers LED to go on or off.  The switch would need the user to define a function to be executed when the state is changed.

I think the heavy-mental thinking will need to go into how the users define touch elements that react with other touch elements.  For example, if you have a analog pressure meter display, and a "range" selector, how in software do you make it obvious how to "hook up" the output of the "range" selector to the pressure meter display?

Can the user define a range selector knob and pressure meter like this?

```python

# Dashboard Row 1
range_knob = KnobWithDetents(1,1000,1000000)
pressure_meter = AnalogMeter(get_voltage(pin3) / range_knob.value)   # where pin 3 is an Analog In pin
pressure_units = TextDisplay( 
    {1: "Pascals", 1000: "kPa", 1000000: "MPa"}(range_knob.value)
    )# use range_knob to reference into a dict to get the pressure units

# Dashboard Row 2
operator_type = Switch(["human", "paranoid android"])

# Dashboard Row 3
evacuation_warning_light = StrobeLight(get_voltage(pin3) >= 42 && operator_type.value == "human") 
                                         # truth table for strobe light to flash

my_dashboard[ 
    [range_knob, pressure_meter, pressure_units],
    [operator_type]
    [evacuation_warning_light]
    ]
touch_handler.activate(my_dashboard)

while True:
    touch_handler()
```

I feel like each display or control element's value should be persistent so that any other element could react to its value.  And somehow each element needs to be "informed" to react to any change in the control.  Rather than try to refresh the display elements every cycle, maybe each element needs the equivalent of "dirty-rectangle" tracking to trigger an update only when the value of a related object was changed.  

When thinking through this, it's almost as if some "machinism" needs to have all these inputs and display elements be "registered" into it so the "machinism" can do all the heavy lifting of knowing the interactions between elements and when each element needs to be updated, and the user just has to solely identify the connections between elements.  Perhaps some of this can be handled with object-oriented programming techniques, but it also feels like there needs to be some kind of master "mechanism" handler that manages the interactions between elements.

### Scoping

I think it is useful to discuss the scope of what we might want to achieve.  

I think that the visual creation of switches, radio buttons, and the different visual elements are probably the easy part of the whole project.  I think the complicated part is how to make the logical interconnections easy.  

For example, we could decide to just make "templates" of some common GUI structures as a starting point (multipage with navigation, dashboard with a switch, a knob and a radio button, etc.).  But that's probably no different than just providing a few good examples.  

However, I'm actually envisioning something that makes the visual layout and logical interconnections easy.  I'll continue to do some more digging to see if there are other examples that may already be doing this.  And we need a simple example to try to see how this all might work.

tannewt, 2020-09-15:
> I feel like each display or control element's value should be persistent so that any other element could react to its value. And somehow each element needs to be "informed" to react to any change in [...]

I would caution against having this be part of what you are trying to achieve because there have been many attempts at this already:

* Callbacks are the oldest way I know of. QT has "signals" and "slots" that tries to do this.
* [Key-value observing](https://developer.apple.com/library/archive/documentation/Cocoa/Conceptual/KeyValueObserving/KeyValueObserving.html) that I believe is no longer a strong pattern because it leads to hard to understand code. "Change a value and have no idea what code that triggers."
* [React](https://en.wikipedia.org/wiki/React_(web_framework)) takes a different, functional approach where you construct your widget hierarchy every data change and it diffs it to change the display.

So, in short. I would encourage you to focus on the widget (button, checkmark, text input) and layout (horizontal, vertical, wrapping item) piece. Let the python code move data from one widget to another.

gmeader, 2020-10-20:
Here is an impressive open-source graphics library written in C for embedded systems. 
https://lvgl.io/
Apparently it works with microPython. Sure would be nice to have these widgets in CircuitPython.

TG-Techie, 2021-01-07:
Taking a leaf out of displayio's book, I think what is needed for circuitpython's core is time sensitive / performance building blocks for a UI loop. 
My first guess at what this should include would be one way to do each of the following:
- A way to make a group as touchable, scrollable, or selectable. 
- A gesture recognition algorithm that takes  inputs from a touchscreen/hardware/buttons and turns them into scroll, touchdown, touchup, press, and hold events (this might several loops for different types of hardware)
- A way to search through the currently shown group on a display to find the widgets that mathc the most recent scroll, touchdown, touchup, press, and hold events
- call some method based on the event.

whatever API we choose to implement as user facing could feature callbacks, diffs, etc as we see fit. or there could be several options to choose from.

there are clearly details to be worked out but it think this would be a good place to start. (and the algorithms  for the touch based events can be quite complex)

FWIW: I’m in the callback camp as it is a very pythonic approach (imo). That being said I also think state based ui is the way to go (look at swifui or flutter).

TG-Techie, 2021-01-07:
Though I agree with @tannewt that choosing between callbacks and the other options would be better to keep out of scope.

kmatch98, 2021-01-28:
Here's a [GUI related topic](https://github.com/adafruit/Adafruit_CircuitPython_DisplayIO_Layout/issues/3) in the libraries.

kmatch98, 2021-02-26:
I think most of this will be handled in the new library https://github.com/adafruit/Adafruit_CircuitPython_DisplayIO_Layout, so I’m closing this here.

closed 2021-02-26 by kmatch98