adafruit/circuitpython #3394: nRF52840-MDK got REPL, but no way to upload sketches

uyoyo117, 2020-09-11:
the board take like a minute to boot, after which REPL is available, but the drive shows 
[image]
i tried to reset the file system and no change, tried on ubuntu and linux
what can i do ?

dhalbert, 2020-09-11:
CIRCUITPY should be available in seconds.
1. Is this the PCA10056 board?
2. Did you install the nRF UF2 bootloader on the board? If not, how are you loading the firmware?
3. What happens on Ubuntu when you plug the board in? Could you start monitoring `/var/log/syslog` from just before you plug in to the time things stabilize, and attach the log here?

uyoyo117, 2020-09-11:
1. it is nRF52840-MDK [https://github.com/makerdiary/nrf52840-mdk](https://github.com/makerdiary/nrf52840-mdk)
2. it dose not support it 
3.`
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.2/1-2.2:1.3/0003:239A:802A.000B/input/input44/event23"
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (II) XINPUT: Adding extended input device "makerdiary nRF52840-MDK Keyboard" (type: KEYBOARD, id 21)
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (**) Option "xkb_model" "pc105"
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (**) Option "xkb_layout" "us"
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (WW) Option "xkb_variant" requires a string value
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (WW) Option "xkb_options" requires a string value
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (II) event23 - makerdiary nRF52840-MDK Keyboard: is tagged by udev as: Keyboard
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (II) event23 - makerdiary nRF52840-MDK Keyboard: device is a keyboard
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (II) config/udev: Adding input device makerdiary nRF52840-MDK Consumer Control (/dev/input/event25)
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (**) makerdiary nRF52840-MDK Consumer Control: Applying InputClass "libinput keyboard catchall"
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (II) Using input driver 'libinput' for 'makerdiary nRF52840-MDK Consumer Control'
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (II) systemd-logind: got fd for /dev/input/event25 13:89 fd 82 paused 0
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (**) makerdiary nRF52840-MDK Consumer Control: always reports core events
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (**) Option "Device" "/dev/input/event25"
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (**) Option "_source" "server/udev"
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (II) event25 - makerdiary nRF52840-MDK Consumer Control: is tagged by udev as: Keyboard
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (II) event25 - makerdiary nRF52840-MDK Consumer Control: device is a keyboard
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (II) event25 - makerdiary nRF52840-MDK Consumer Control: device removed
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (II) libinput: makerdiary nRF52840-MDK Consumer Control: needs a virtual subdevice
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.2/1-2.2:1.3/0003:239A:802A.000B/input/input46/event25"
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (II) XINPUT: Adding extended input device "makerdiary nRF52840-MDK Consumer Control" (type: MOUSE, id 22)
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (**) Option "AccelerationScheme" "none"
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (**) makerdiary nRF52840-MDK Consumer Control: (accel) selected scheme none/0
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (**) makerdiary nRF52840-MDK Consumer Control: (accel) acceleration factor: 2.000
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (**) makerdiary nRF52840-MDK Consumer Control: (accel) acceleration threshold: 4
Sep 11 20:04:36 ubuntu mtp-probe: checking bus 1, device 9: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.2"
Sep 11 20:04:36 ubuntu mtp-probe: bus: 1, device: 9 was not an MTP device
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (II) event25 - makerdiary nRF52840-MDK Consumer Control: is tagged by udev as: Keyboard
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (II) event25 - makerdiary nRF52840-MDK Consumer Control: device is a keyboard
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (**) makerdiary nRF52840-MDK Consumer Control: Applying InputClass "libinput keyboard catchall"
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (II) Using input driver 'libinput' for 'makerdiary nRF52840-MDK Consumer Control'
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (II) systemd-logind: returning pre-existing fd for /dev/input/event25 13:89
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (**) makerdiary nRF52840-MDK Consumer Control: always reports core events
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (**) Option "Device" "/dev/input/event25"
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (**) Option "_source" "_driver/libinput"
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (II) libinput: makerdiary nRF52840-MDK Consumer Control: is a virtual subdevice
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (**) Option "config_info" "udev:/sys/devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2.2/1-2.2:1.3/0003:239A:802A.000B/input/input46/event25"
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (II) XINPUT: Adding extended input device "makerdiary nRF52840-MDK Consumer Control" (type: KEYBOARD, id 23)
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (**) Option "xkb_model" "pc105"
Sep 11 20:04:36 ubuntu /usr/lib/gdm3/gdm-x-session[1819]: (**) Option "xkb_layout" "us"
[... cut ...]
Sep 11 20:06:05 ubuntu udisksd[1642]: Mounted /dev/sdd at /media/ubuntu/DAPLINK on behalf of uid 999
Sep 11 20:06:07 ubuntu dbus-daemon[1825]: [session uid=999 pid=1825] Activating service name='org.gnome.gedit' requested by ':1.99' (uid=999 pid=3331 comm="/usr/bin/nautilus --gapplication-service " label="unconfined")
Sep 11 20:06:07 ubuntu dbus-daemon[1825]: [session uid=999 pid=1825] Successfully activated service 'org.gnome.gedit'
Sep 11 20:06:35 ubuntu tracker-store[3611]: OK
Sep 11 20:06:35 ubuntu systemd[1791]: tracker-store.service: Succeeded.`

uyoyo117, 2020-09-11:
i tried to build it myself and  by [this port ](https://github.com/adafruit/circuitpython/tree/main/ports/nrf/boards/makerdiary_nrf52840_mdk)
and after adding SoftDevice this is my problem

dhalbert, 2020-09-11:
It would be best if you consulted with the MakerDiary folks about this. We accepted their board definition, but do not support it ourselves.

If you were using SoftDevice 7.x, try 6.1.1. They are different sizes and the build has to allow for that.

uyoyo117, 2020-09-11:
already tried to ask them for help, but they said it only REPL that is suppurated is it possible to upload a sketch, or to enable the drive ?

uyoyo117, 2020-09-12:
after a bit of research, how do i config this board to use its internal memory ? and not the flash chip on board ?

dhalbert, 2020-09-12:
I'm having a a hard time reading your screenshot. Make sure you update the git submodules if you've pulled recently. Yes, copy the config from the pca10059 or similar board.

uyoyo117, 2020-09-12:
sorry for the bad image 
[image]

dhalbert, 2020-09-12:
You need to remove the QSPI pin definitions in the `mpconfigboard.h` for this board.

uyoyo117, 2020-09-13:
and it works!!

closed 2020-09-14 by tannewt