adafruit/circuitpython #3400: readthedocs edit  404

CarlFK, 2020-09-12:
https://circuitpython.readthedocs.io/en/latest/shared-bindings/time/#time.monotonic_ns

click "edit on github" 
https://github.com/adafruit/circuitpython/blob/main/shared-bindings/time/index.rst
404

for this one, correct seems to be:
https://github.com/adafruit/circuitpython/edit/main/shared-bindings/time/__init__.c

Jeff E.Today at 11:21 AM
@CarlFK hmmm I don't know what to do about that link 404'ing, can you file an issue so we can at least remember it happens?  It has to do with the way we generate our documentation in several steps, and I guess that works differently than readthedocs needs

jepler, 2020-09-13:
@ciscorn I don't suppose you know if we have a way to resolve this?

julianrendell, 2020-09-16:
Probably not the cause, but from reading the config files and pattern matching, wonder if it's to do with `/docs/autoapi/templates/python/module.rst` - search for `index.rst`. To me the paths that I think this template generates match up to the incorrect links on RTD. 

I've built the docs locally and the resultant structure is quite different than RTD- so I'm lost as to how to reproduce it locally.

But after some googling, I think the GitHub links on RTD are added by their theme, which looks to be here: https://github.com/readthedocs/sphinx_rtd_theme. 

There is a 

```
.. confval:: github_url

    Force the :guilabel:`Edit on GitHub` button to use the configured URL.
```

mentioned in the [configuration docs](https://github.com/readthedocs/sphinx_rtd_theme/blob/c04c214a8591a0290f30026b89bf3215fb1061b4/docs/configuring.rst#id9) page. But that looks to be a hardcoded link, not sure if it is branch aware.

tannewt, 2022-02-02:
Let's close this in favor of  #5763

closed 2022-02-02 by tannewt