adafruit/circuitpython #3403: WiFi Association with specific BSSID

astrobokonon, 2020-09-12:
It'd be nice to be able to specify the specific mac address of the access point you want to connect to as a backup/sanity parameter in multi-access point environments.

When connecting, the exact AP associated with seems to be chosen somewhat at random - sometimes it's the strongest signal, other times it definitely isn't.  In the field with MicroPython based devices, I've seen it try to connect to an AP with an RSSI of -80, rather than the one right next to it at -45.  To get around that, I do an initial scan at boot, sort by strongest signals, and then connect() with the specific SSID and BSSID of the best RSSI from the scan.

Looking at the debug port during startup, it seems to be doing the same random choice - I've got two different APs broadcasting the same SSID, and it connected to the one with the "lower" signal strength (-48 vs. -40).  Everything in here looks normal but I include it here just in case

```
I (2628) wifi:wifi driver task: 3ffff00c, prio:23, stack:3584, core=0
I (2628) system_api: Base MAC address is not set
I (2628) system_api: read default base MAC address from EFUSE
I (2628) wifi:wifi firmware version: 9c89486
I (2638) wifi:wifi certification version: v7.0
I (2638) wifi:config NVS flash: disabled
I (2638) wifi:config nano formating: disabled
I (2648) wifi:Init data frame dynamic rx buffer num: 32
I (2648) wifi:Init management frame dynamic rx buffer num: 32
I (2658) wifi:Init management short buffer num: 32
I (2658) wifi:Init dynamic tx buffer num: 32
I (2668) wifi:Init static rx buffer size: 1600
I (2668) wifi:Init static rx buffer num: 10
I (2678) wifi:Init dynamic rx buffer num: 32
I (2808) phy: phy_version: 603, 72dfd77, Jul  7 2020, 19:57:05, 0, 2
I (2808) wifi:enable tsf
I (2808) wifi:mode : softAP (7c:df:xx:xx:xx:xx)
I (2808) wifi:Total power save buffer number: 16
I (2808) wifi:Init max length of beacon: 752/752
I (2818) wifi:Init max length of beacon: 752/752
I (2928) wifi:mode : sta (7c:df:xx:xx:xx:xx)
I (4328) cp iternext: stop iteration
I (5058) wifi:new:<6,0>, old:<1,1>, ap:<255,255>, sta:<6,0>, prof:1
I (5528) wifi:state: init -> auth (b0)
I (5538) wifi:state: auth -> assoc (0)
I (5548) wifi:state: assoc -> run (10)
I (5558) wifi:connected with <MYSSID>, aid = 12, channel 6, BW20, bssid = 7a:8a:xx:xx:xx:xx
I (5558) wifi:security: WPA2-PSK, phy: bgn, rssi: -48
I (5558) wifi:pm start, type: 1

W (5568) wifi: connected
I (5598) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (7618) esp_netif_handlers: sta ip: 192.168.1.234, mask: 255.255.255.0, gw: 192.168.1.1
W (7618) wifi: got ip
I (7658) wifi:state: run -> init (0)
I (7658) wifi:pm stop, total sleep time: 1485945 us / 2090829 us

I (7658) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
```

Since MAC isn't an available property of ```wifi.Network``` during ```wifi.radio.start_scanning_networks``` I can't do my usual trick here, and it looks like ```wifi.radio.connect()``` doesn't accept BSSID anyways.

Testing with CircuitPython 6.0.0-alpha.3-165-g9256e6b37 on 2020-09-11; Saola 1 w/Wroom with ESP32S2.

tannewt, 2020-09-14:
I suspect it's connecting to the first channel that has an AP with the desired access point. The connect redoes a scan and stops on the first channel it finds. I added `channel` as an option to connect in order to speed up the connection process. That should solve your issue if your APs are on different channels.

Would you like to help add bssid to the Network class? It should be a straightforward edit to https://github.com/adafruit/circuitpython/blob/main/shared-bindings/wifi/Network.c#L78 and the corresponding file in common-hal. Having a BSSID option would still help the chase where both APs are on the same channel.

astrobokonon, 2020-09-16:
@tannewt sure, I think I can get this going.  You're right that bssid to the Network class was pretty easy once I got a better feel for the layout of things, since it's already stuffed into ```wifi_network_obj_t``` along with everything else from the esp-idf's ```wifi_ap_record_t```.  I just need to add it as a parameter to the ```connect``` call and then I'll open a PR.

tannewt, 2020-09-16:
Thanks @astrobokonon! If you need help, the Adafruit Discord is a good spot to get help if you need it: https://adafru.it/discord

PR #3433 (2020-09-19): ESP32S2: Allow connecting to specific bssid
Closes #3403 

I couldn't figure out a way to avoid doing a last minute conditional check in ```common-hal/wifi/Radio.c``` because the ```bssid_set``` flag needs to be set only if you're specifying a specific MAC.  If you're not passing in a bssid, but you happen to still set the ```bssid_set``` flag, the connection will fail and you'll end up returning ```WIFI_RADIO_ERROR_NO_AP_FOUND``` so that's no good (and a bit confusing).

I needed a way to check the bssid length as an initial sanity check against someone passing it in as a string (with colons) [... cut]
closed 2020-09-19 by dhalbert