adafruit/circuitpython #3408: mkqstrdata encoding assumption on windows msys

ladyada, 2020-09-12:
removing line 144 'solves' it!

see https://github.com/adafruit/circuitpython/pull/3219/files for possible reasons

```
ladyada@LimorFried MINGW64 ~/Dropbox/micropython/circuitpython
$ make -C mpy-cross
make: Entering directory '/c/Users/ladyada/Dropbox/micropython/circuitpython/mpy-cross'
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
QSTR updated
Traceback (most recent call last):
  File "../py/makeqstrdata.py", line 455, in <module>
    encoding_table = compute_huffman_coding(translations, qstrs, args.compression_filename)
  File "../py/makeqstrdata.py", line 144, in compute_huffman_coding
    print("//", ord(ch), s, counts[ch], canonical[ch], renumbered)
  File "C:/msys64/mingw32/lib/python3.8/encodings/cp1252.py", line 19, in encode
    return codecs.charmap_encode(input,self.errors,encoding_table)[0]
UnicodeEncodeError: 'charmap' codec can't encode character '\x80' in position 0: character maps to <undefined>
make: *** [../py/py.mk:346: build/genhdr/qstrdefs.generated.h] Error 1
make: *** Deleting file 'build/genhdr/qstrdefs.generated.h'
make: *** Waiting for unfinished jobs....
make: Leaving directory '/c/Users/ladyada/Dropbox/micropython/circuitpython/mpy-cross'
```

jepler, 2020-09-12:
We can either set the standard output encoding to utf-8 on all platforms when running this script:
```
diff --git a/py/makeqstrdata.py b/py/makeqstrdata.py
index 721fa8320..d5502330f 100644
--- a/py/makeqstrdata.py
+++ b/py/makeqstrdata.py
@@ -16,6 +16,8 @@ import collections
 import gettext
 import os.path
 
+sys.stdout.reconfigure(encoding='utf-8')
+
 py = os.path.dirname(sys.argv[0])
 top = os.path.dirname(py)
 
```
which should _ALSO_ solve it for windows .. or `sys.stdout.reconfigure(encoding='ascii')` so that the problem would dog us on Linux / Mac as well.

jepler, 2020-09-13:
I've added a commit to #3398 that will resolve this problem.  If that PR takes long to merge, I can split it out into its own PR.  However, as that open PR is working in the same part of the code, it reduces the chances of needing to resolve merge conflicts.

commit 15964a47 (2020-09-16): makeqstrdata: Avoid encoding problems

Most users and the CI system are running in configurations where Python
configures stdout and stderr in UTF-8 mode.  However, Windows is different,
setting values like CP1252.  This led to a build failure on Windows, because
makeqstrdata printed Unicode strings to its stdout, expecting them to be
encoded as UTF-8.

This script is writing (stdout) to a compiler input file and potentially
printing messages (stderr) to a log or console.  Explicitly configure stdout to
use utf-8 to get consistent behavior on all platforms, and configure stderr so
that if any  [... cut]
closed 2020-09-16 by tannewt