adafruit/circuitpython #3409: Alternative to adjust Brightness on RGB matrix

riffnshred, 2020-09-13:
Hey guys, Sorry if this as been discussed already but I can't find anything using google and search on here.

I've been playing around with the Feather M4 express and an 64x32 RGB matrix. First thing I've been looking for when I started running demos is for a way to reduce the brightness. 

I found that you can only turn on or off the LEDs by setting the brightness attribute to 1.0 or 0.0:
https://circuitpython.readthedocs.io/en/5.3.x/shared-bindings/rgbmatrix/RGBMatrix.html

Is there any way I can hack my way to adjust the brightness? On all my other raspberry pi projects where I run an rgb matrix, I set them to half brightness because I feel like looking at it 100% brightness is way too intense. 

Thank you so much in advance

jepler, 2020-09-13:
You're right that this isn't possible right now.  It would need to be added "upstream", so I've filed a bug there: https://github.com/adafruit/Adafruit_Protomatter/issues/19

Until that feature is added and then made available in Circuitpython, depending on what you're doing, you can
 * just select different color constants, like 0x7f7f7f or (127,127,127) instead of 0xffffff or WHITE or (255,255,255)
 * modify your images before exporting them as bitmaps
 * if you use paletted images, dynamically modify the palette for brightness
For the latter, we have some example code within another guide that you may be able to adapt -- the "Reshader" class https://learn.adafruit.com/ocean-epoxy-resin-lightbox-with-rgb-led-matrix-image-scroller/code-explanation

riffnshred, 2020-09-13:
Sweet, thank you Ill test those out

tvajjhala, 2022-03-21:
Any update on this?

ladyada, 2022-03-22:
unlikely to be added any time soon, it is nontrivial

wb8nbs, 2023-12-17:
Louis Beaudoin has it worked out in the SmartMatrix library. Maybe you could steal his code.
This works for me 64x64 panel, SmartMatrix, Teensy 3.6
void setBrightness(uint8_t brightness);

czei, 2024-02-21:
> just select different color constants, like 0x7f7f7f or (127,127,127) instead of 0xffffff or WHITE or (255,255,255)

FYI, this approach doesn't seem to work on the MatrixPortal S3 with CircuitPython.   The actual brightness of 0x7f7f7f is exactly the same as 0xffffff.  :-(.  

I was hoping this would be a solution, but no dice.  I'm getting complaints about the LED being so bright it's hurting people's eyes, so I'm hoping a workaround will show up somehow.

CedarGroveStudios, 2024-02-21:
I haven’t used the S3 version of the MatrixPortal yet, but suspect that the default color bit depth is set to two bits, just four levels of color brightness, limiting the granularity of intensity.
When instantiating the display in CircuitPython, the bit depth can be set to 6-bits which increases the brightness granularity — at a price of needing more memory to hold the display buffer.
If you are using CircuitPython displayio objects, you could apply something like PaletteFader to help. https://github.com/CedarGroveStudios/CircuitPython_PaletteFader  It’s in the Community Bundle and can be installed using circup.

no3rdw, 2024-02-21:
So, I have a RGBMatrix brightness workaround... you can even use a brightness control to adjust the brightness values of all text/vectorio shapes _and_ displayed BMP images.  I'm defining all color values in HLS instead of Hex, which makes it super simple to change the brightness without changing the color value (just adjust L/level).  For BMPs, feed the pixel_shader into a function that converts the integer colors to HSL, then adjusts their level and inserts the new value back into the pixel_shader.

Relevant code is in this file:
https://github.com/no3rdw/ledmatrix-effects/blob/main/device.py

All colors are defined like this: `device.hls(.85, .3, 1)`, and the hsl function in device.py applies the brightness adjustment.
All BMPs need their shader fed into this function: `device.alphaPalette(bmpimage.pixel_shader)` - which again applies the brightness adjustment to every indexed color used by the BMP.

czei, 2024-02-21:
> I haven’t used the S3 version of the MatrixPortal yet, but suspect that the default color bit depth is set to two bits, just four levels of color brightness, limiting the granularity of intensity. [...]

Yup, that was the problem.   After setting the buffer depth to 6-bits adjusting the RGB values by a scale enabled a decent control over brightness.   Thanks for the suggestion!

sub-zerohm, 2025-04-23:
> Yup, that was the problem. After setting the buffer depth to 6-bits adjusting the RGB values by a scale enabled a decent control over brightness. Thanks for the suggestion!

Just getting a 404 error on that link