adafruit/circuitpython #3410: RFC: routines for handling wrapped tick counts

jepler, 2020-09-13:
_I plan to raise this issue In the Weeds on Monday and if there is a consensus about approach I can be the implementor_

# Problem statement

Libraries and users need a way to manage time durations, and existing solutions are not entirely satisfactory.  Thanks to @dglaude for raising this issue on Discord and [github](https://github.com/adafruit/Adafruit_CircuitPython_Debouncer/issues/22).

For consistency and portability reasons, we have only included compatible time-related functions in the standard `time` module.  When it comes to dealing with durations and times not related to a particular epoch, this means we have

 * time.monotonic(), available on all builds, which returns a floating point number of seconds that increases until the microcontroller is reset.  monotonic() loses precision so that after around 18 hours, intervals of 1/64 second can no longer be distinguished; within several days, intervals of 1/8 second can no longer be distinguished
 * time.monotonic_ns(), which returns an integer number of nanoseconds that increases until the microcontroller is reset.  monotonic_ns() is only available on builds with long integers, because without it the count overflows in less than 2 seconds.

Multiple Adafruit bundle libraries have implemented workarounds of the general form:
```
if (monotonic_ns exists and is usable):
    def custom_timekeeping_function():
        implement in terms of monotonic_ns
else:
    def custom_timekeeping_function():
        implement in terms of monotonic
```
however, due to the problem of time.monotonic granularity, this leads to programs which have to be restarted (reset button, not soft restart) everyday or they misbehave.

Another (resolvable) problem is that due to the evolution of the core, the test (monotonic_ns exists and is usable) is different between 5.3 and 6.0, which means that at the time I write this, the `adafruit_debouncer` library does not work on the Trinket M0.  The other workaround I know of, in `adafruit_led_animation`, is what I would consider the canonical workaround.

The `ble_midi` library uses monotonic_ns with no workaround, but we probably do not anticipate any devices that support ble but are too resource-constrained to have long ints.

I did not survey libraries that use time.monotonic only; there are a large number of them and there are probably many "misbehaves after hours/days" class bugs there, such as
```
adafruit_esp32spi.py:        while (time.monotonic() - times) < 0.1:
```
In all, 89 files in the Adafruit bundle and 14 files in the community bundle seem to use `time.monotonic()`.

# Requirements

I think we can only address this by adding new code to CircuitPython.  This is a dicey proposition but right this second we do have space in our most constrained builds due to improvements in message compression, so we should be able to make it work for 6.0.

 * The new APIs are not placed in a desktop Python module like time
    * microcontroller and rtc are two plausible places
    * microcontroller already has sleep_us
 * work with the data type of (short) integers only
 * is feasible to implement in terms of time.monotonic_ns, so it can be implemented in blinka and for backwards compatibility on long int builds
 * is feasible to implement in terms of time.monotonic, without worsening the granularity problem

There is a solution, though it means giving up these two properties:
 * The clock increases forever, never wrapping around
 * the units of the clock are one of the python-"comptaible" units, seconds
   or nanoseconds

# Proposal

Pull some select routines from micropython.utime into microcontroller, possibly choosing more expressive names:
```
microcontroller.ticks_ms()
microcontroller.ticks_add()
microcontroller.ticks_diff()
```
    
ticks_ms counts up, but each time it would count past a maximum value, it wraps back to zero.  If you have two values from ticks_ms(), you can add them (or subtract them, by negating one):
```
start = ticks_ms()
wait_for_event()
end = ticks_ms
duration_ms = ticks_add(end, -start)
```
You can use ticks_diff to find out if you're before or after a deadline:

```
start = ticks_ms()
deadline = ticks_add(start, 100)
while ticks_diff(deadline, ticks_ms()) > 0:
    do_something_quick()
```
In practical terms, ticks_ms() will wrap every 2**29 milliseconds, which is approximately 6 days.  Determining "before or after" works when the times are within half of the wrapping period, or about 3 days.  When it comes to a program task like "count down to Halloween", it is not the appropriate API, and date-based functions are more appropriate.  When it comes to a program task like "make the LED animation advance as close to every 1/20s as possible, forever", it is adequate indefinitely.

elaborating on some choices I made:
 * the TICKS_PERIOD of 2**29 is chosen to simplify the backwards-compatibility implementation of ticks_add and ticks_diff on platforms without long ints; we can add or subtract any two numbers where the absolute values are less than TICKS_PERIOD without overflow.
 * TICKS_PERIOD and other constants in the `adafruit_ticks` library should not be accessed and are considered implementation defined.  The minimum guarantee would be that ticks wouldn't wrap around for at least 2 days, allowing differences of at least 1 day to be represented.  (I can't think of a reason we'd want to do smaller than 2**29 either)

 * ms is the best choice for granularity: seconds are too long, microseconds are too short, and other granularities like 1/1024s, .001s, etc., feel too arbitrary

# Backwards compatible helper library: `adafruit_ticks`

This backwards compatible implementation is the one that shold be used (rather than directly using microcontroller.ticks_ms etc), because it will function on a range of CircuitPython versions from 5.x to 6.0 as well as on Adafruit Blinka.  It makes a best effort implementation, but cannot solve the granularity problem on builds without long ints.
```
_MS_PER_NS = const(1000000)
_MS_PER_S = const(1000)
_TICKS_MAX = const(536870911)
_TICKS_PERIOD = const(536870912)
_TICKS_HALFPERIOD = const(268435456)

try:
    from microcontroller import ticks_ms, ticks_add, ticks_diff
    ticks_ms()
    ticks_add(0, -1)
    ticks_diff(1, 100)
except (ImportError, NotImplementedError):
    import time

    try:
        def ticks_ms():
            return (time.monotonic_ns() // MS_PER_NS) & _TICKS_MAX
        ticks_ms()
    except (NameError, NotImplementedError):
        def ticks_ms():
            return round(time.monotonic() * MS_PER_S % _TICKS_PERIOD)

    def ticks_add(ticks, delta):
        return (a + b) % _TICKS_PERIOD

    def ticks_diff(end, start):
        diff = (end - start) & _TICKS_MAX
        diff = ((diff + _TICKS_HALFPERIOD) & _TICKS_MAX) - _TICKS_HALFPERIOD
        return diff
```

_this library could be rewritten so that the availablity of ticks_ms is considered separately from ticks_add and ticks_diff, reducing the in-core implementation footprint from 3 functions to 1_

dhalbert, 2020-09-14:
There is a `LONGLONG` implementation of long ints which we don't use, is simply 64 bits of integer, not indefinite length. Unfortunately, it builds to be even larger than the regular MPZ implementation. But this is due to it dragging in double-precision float routines, which are huge and which should not be necessary. I am looking at how to prevent this, and perhaps LONGLONG could be included even in the smallest builds, which would make `monotonic_ns` be available for all ports.

dhalbert, 2020-09-14:
See #3416 about whether using `LONGLONG` for longints for the smallest builds is practical. The answer is it only fits for the smaller translations.

snej, 2020-09-14:
It's pretty trivial to implement basic 64-bit integer operations using (int, int) tuples as the representation. Especially for the kind of operations that are usually done on timestamps: addition, subtraction, comparison.

If you interpret those tuples as 32.32 fixed-point, you get a range of about ±60 years with sub-nanosecond precision.

emard, 2020-09-14:
If I can vote, I prefer having montonic_ns() to be 32 or 64 bit integer
and better not having ticks_add sub and or not mul mod etc
functions for time arithmetic

endtime = montonic_ns() + some_time
while montonic_ns()-endtime < 0:
  pass

snej, 2020-09-14:
> better not having to use special ticks_add sub and or not mul mod etc functions

Python has operator overloading. Whatever the type is, you should to be able to use `+`, `-`, `<`, etc. on it, that's not a problem.

jepler, 2020-09-21:
Thanks, I appreciate all the comments.

I think that an object-based interface that has operator overloading is a fine idea for the hypothetical `adafruit_ticks` module coded in Python.  As long as this object has some kind of "update"-like method that is called at least once per day, it can deal with overflows in whatever way is required.

An additional advantage of the Micropython API which only takes and returns ints is that it operates without allocations.  This would not be true if an object instance or a tuple was returned.

An additional justification for why it is "okay" to restrict ticks to smaller timescales, and not pretend that it's suitable for general timekeeping:  Especially when not on USB, the monotonic clock of a microcontroller is not particularly accurate.  I measured a 240kHz nominal PWM output from a Trinket M0.  When it's on USB, a frequency meter I trust measures 240.005 with the last digit varying  -- but when it's not, I get about 235.900kHz with the last three digits varying.  That's an error of nearly half an hour per day.  Thus, even if there is no problem with wraparound, the "ticks" are not appropriate for accurately measuring intervals of 1 day to "human precision".  You need a separate RTC chip for this, and a CircuitPython-adapted version of standard Python's "datetime" module for arithmetic on dates.

(This also means that if you try to use *ANY* API to blink a LED 24 times a minute -- or, say, to govern a 24FPS image display -- you'd get 23.5FPS, no matter how sophisticated the "ticks" API is, so in any case your program as a whole has to be able to behave gracefully when ticks collide with "real world" time)

I chose 240kHz because the Trinket M0 operates at (nominal) 48MHz, so the PWM frequency is a nice integer divisor 200 = 8*5*5.  The frequency is more accurate when attached by USB because the Trinket makes its own timekeeping match the USB system's timekeeping; otherwise, it uses an inaccurate internal oscillator which is only good to within a few percent.  In this case the host USB system is a Pi Zero W.  Results will vary from Trinket to Trinket and from host to host.   The Pi is running systemd-timesyncd, but I don't think that this affects the accuracy of its USB bus frequency.

Breefield, 2020-11-16:
I am 100% in support of the inclusion of the time `ticks` methods. I actually just hit a small wall yesterday with https://github.com/micropython-IMU/micropython-fusion/blob/master/deltat.py#L53 because it relies on `utime.ticks_diff`
It also uses `time.ticks_us()` but it's trivial to convert from `time.ticks_ms()` to microseconds.

Following, and hopeful this could be included.

todbot, 2021-04-26:
I upvote any solution that provides an equivalent of an integer `millis()` or `ticks()` function.
Adding either to `microcontroller` seems good if you don't want to pollute the `time` namespace. 
(super jealous of `_pew.get_ticks()` I'd take that if it was available on Trinket / QT Py M0 boards)

ktritz, 2021-04-26:
nitpick, but if you want to be mathematically accurate the first constant should be _NS_PER_MS.

dhalbert, 2021-05-15:
I thought about this, and propose adding a simple new native-C module `interval`. Its implementation will be very small so it can fit on the smallest boards, which don't have longints. Because it only uses short ints and is not a class, it generates no garbage, and is fast. Pseudo-code is below:

interval.py:
```py
"""Time intervals < 2**30 milliseconds (about 298 hours)."""

MAX_MS = 2**30 - 1
"""Maximum available interval: about 298 hours."""

def now_ms() -> int:
    """Return the value of a clock that counts milliseconds
    and wraps around above interval.MAX_MS."""

    # Truncate the current tick count to a 30-bit positive integer.
    # supervisor_ticks_ms32() is in supervisor/shared/tick.c
    return supervisor_ticks_ms32() & MAX_MS

def diff_ms(later: int, earlier: int) -> int:
    """Return the difference in milliseconds between two now_ms() values.
    The value returned is always >= 0. If later < earlier,
    assume that now_ms() has wrapped, and adjust the difference accordingly.
    For instance, if later == 1 and earlier == (MAX_MS + 1) - 2, return 3"""

    if later >= earlier:
        return later - earlier
    else:
        # This arithmetic will be done in C with 32 bits, so there's no overflow problem.
        return later + (MAX_MS + 1) - earlier

def has_elapsed_ms(start: int, length: int) -> bool:
    """Return True if at least length milliseconds have elapsed since start."""

    return diff_ms(now_ms(), start) >= length
```
`interval.has_elapsed()` exists to make the check be as fast as possible, and to cover the most common use case.

A typical way this would be used is:

```py
import interval

start = interval.now_ms()
while not interval.has_elapsed(start, 10):
    # do something for 10 msecs
```

Comments welcome.

jepler, 2021-05-15:
@dhalbert I think that's fine, it has a slightly smaller "surface" than mine but is a bit "more different" from the micropython implementation.

To allow implementing micropython's ticks_diff and ticks_add with a minimum of fuss, consider making the wrap occur 2**29 rather than 2**30 (and do put the wrap value in the public API):

> the TICKS_PERIOD of 2**29 is chosen to simplify the backwards-compatibility implementation of ticks_add and ticks_diff on platforms without long ints; we can add or subtract any two numbers where the absolute values are less than TICKS_PERIOD without overflow.

and also more above where I argue that this API is not useful for lengthy intervals anyway, so going from 300 hours to 150 hours "range" is not a big sacrifice.

dhalbert, 2021-05-15:
I worked `interval` out from scratch while not being able to sleep, without reference to the MicroPython `utime` routines. The first thing I thought of was `has_elapsed_ms()`, and added `diff_ms()` later. I didn't realize how close it was to `utime`. My `diff_ms()` does not return negative values (it always assumes there might be a wrap) and does not depend on `utime`s `TIME_PERIOD/2` constraint. I think that it is a little less confusing.

dhalbert, 2021-05-15:
I wrote `interval` up without fully re-reading the original comments -- sorry. I'd be OK with adding some kind of `add` function too, and including the possibility of negative `diff`. But I do like the idea of putting this in its own module. Adding a `done` or `has_elapsed` function is also convenient. The main thing is that it should be quite small, so it still fits on low-flash-size boards.

PR #4936 (2021-07-12): supervisor: Add a function to get "wrapping" milliseconds
As discussed in #3410, this is a micropython-compatible (though differently named) function that allows creation of useful time and deadline functions in pure python.     

Closes: #3410 

An alternative to #4935 which only adds the wrapped ticks and doesn't change the monotonic reference time.

## supervisor.ticks_ms() → int

Return the time in milliseconds since an unspecified reference point, wrapping after 2**29ms.

The value is initialized so that the first overflow occurs about 65 seconds after power-on, making it feasible to chec [... cut]
closed 2021-07-12 by jepler