adafruit/circuitpython #3414: Lilygo T8 ESP32s2 with 4MB Flash, 8MB RAM

mallagant, 2020-09-14:
Hello,

I have tried to create my own board spec for Lilygo module, but I am getting following error. First I have got error about mismatching 2MB ram but installed 8MB which I have changed in configuration. But it probably caused that partitions does not match, which is what I understood from error invalid SPI size. I have even tried expand partition 4 to 512K, but it does not look like it helped.

Could you help me with that? Is there better place to discuss than here? It is very nice module with IPS display on board.

I (48) boot: ESP-IDF v4.2-dev-2554-gde733cdab 2nd stage bootloader
I (48) boot: compile time 08:24:52
I (48) boot: chip revision: 0
I (52) boot.esp32s2: SPI Speed      : 40MHz
I (56) boot.esp32s2: SPI Mode       : DIO
I (61) boot.esp32s2: SPI Flash Size : 4MB
I (66) boot: Enabling RNG early entropy source...
I (71) boot: Partition Table:
I (75) boot: ## Label            Usage          Type ST Offset   Length
I (82) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (90) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (97) boot:  2 ota_0            OTA app          00 10 00010000 00160000
I (104) boot:  3 ota_1            OTA app          00 11 00170000 00160000
I (112) boot:  4 uf2              factory app      00 00 002d0000 00080000
I (120) boot:  5 user_fs          Unknown data     01 81 00350000 00080000
I (127) boot: End of partition table
I (131) boot: Defaulting to factory image
E (136) esp_image: image at 0x2d0000 has invalid magic byte
W (142) esp_image: image at 0x2d0000 has invalid SPI mode 255
W (149) esp_image: image at 0x2d0000 has invalid SPI size 15
E (155) boot: Factory app partition is not bootable

There is also second issue with library for the display, https://github.com/devbis/st7789_mpy which is created for MicroPython, but it does not compile for CircuitPython. Link to lib machine_spi.h not found. I expect this may be no longer similar to original MicroPython library. But at least hint where should I look to modify would help a lot.

mall@mall-pc:~/circuitpython/ports/esp32s2$ make BOARD=lilygo USER_C_MODULES=/home/mall/st7789_mpy/
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
Including User C Module from /home/mall/st7789_mpy//st7789
/home/mall/st7789_mpy/st7789/st7789.c:31:10: fatal error: extmod/machine_spi.h: No such file or directory
 #include "extmod/machine_spi.h"
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [../../py/mkrules.mk:55: build-lilygo/st7789/st7789.o] Error 1

Thank you very much
Richard

tannewt, 2020-09-14:
Hi Richard! Thanks for adding a new ESP32-S2 board to CircuitPython. We're always happy to see more boards supported. The best place to get support developing with CircuitPython is the Adafruit Discord server, the #circuitpython channel specifically. You can join it by going to https://adafru.it/discord . Discord is blocked in China, so you can add me (tannewt) on WeChat if you are there.

The Factory app partition error is a red herring because we intend on placing the TinyUF2 bootloader there. Until then, the second stage bootloader should be able to try OTA0 and boot CircuitPython from it.

I don't recommend compiling in the ST7789 driver. We have a CircuitPython driver for it here that should work: https://github.com/adafruit/Adafruit_CircuitPython_ST7789 For boards with built-in displays, we tend to init them by default but the info on this board I found doesn't have it on the board. So, initing from user Python code should be fine.

tannewt, 2020-09-14:
Here is a guide we have for adding a board in case you haven't seen it yet: https://learn.adafruit.com/how-to-add-a-new-board-to-circuitpython

skieast, 2020-11-07:
HI Richard, 
If you haven't gotten this working here is a link to my various board files for what I believe is the same board. I also tested the board with some CircuitPython code to drive the display. I'll have to search around and see if I can find that, I probably just used the driver Scott mentioned above
https://github.com/skieast/circuitpython/tree/main-copy/ports/esp32s2/boards/lilygo_ttgo_t8_s2

https://gist.github.com/skieast/8c00274f557113cad4d5d377cfd57a9d

expandme-tech, 2020-11-10:
Hi All,

Can we somehow collaborate on it ?

Double checking - Are we speaking about these: 

LILYGO® TTGO T8 ESP32-S2 V1.1
http://www.lilygo.cn/prod_view.aspx?TypeId=50033&Id=1300&FId=t3:50033:3

LILYGO® TTGO T8 ESP32-S2 V1.1 ST77789 1.14 Inch LCD Display
http://www.lilygo.cn/prod_view.aspx?TypeId=50033&Id=1321&FId=t3:50033:3

These modules has USB-C functionality selectable by 4 dip switches  (Serial or chip USB), so does circuitpython need both functionality at once ?

Ideal will be to have it supported in this HW config: USB in ESP32-S2 mode, crystal connected and TF support enabled.

PS: I just see on Mouser new ESP32-S2 WROVER (2GB PSRAM) boards with 2 USB ports :-)

Have a nice day
Rudolf

tannewt, 2020-11-10:
@expandme-tech We're more than happy to have board definitions for these in CircuitPython. The biggest hurdle for most boards is having a unique USB PID for each. Open source hardware can get one for free from https://pid.codes. Companies are expected to purchase a USB VID from the USB-IF (https://www.usb.org/getting-vendor-id).

@Xinyuan-LilyGo Do you have USB PIDs for each of these boards?

expandme-tech, 2020-11-10:
So this board will be best fit for sample code with right free USB ID:  https://github.com/eggsampler/ESP32-S2-Breakout (It's even has USB to MicroUSB now)

It has right license (MIT) license 👍

Rudolf

xmmpru83, 2020-12-18:
@skieast ur fork is broken. tinyusb outdated, some mistake on the board.h path as well...
just clone the latest CP, copy those 2 folders of the lily, modify board.c to point into supervisor/board.h like all other boards.
build flash and go.

skieast, 2020-12-18:
I haven’t been looking closely at my fork. Mostly looking at some Wifi i2c
issues.  I’ll take a look and see  what I can do. Thanks

xmmpru83, 2020-12-18:
i think this board is too fresh.. very unsupported, im not able to make it run..

SPIWP:0xee
mode:DIO, clock div:2
load:0x3ffe6100,len:0x8
load:0x3ffe6108,len:0x17ec
load:0x4004c000,len:0x930
load:0x40050000,len:0x2cc8
entry 0x4004c1b4
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4002ec30
0x4002ec30: _xt_highint4 at /root/circuitpython/ports/esp32s2/build-lilygo_ttgo_t8_s2_st7789/esp-idf/../../esp-idf/components/freertos/xtensa/xtensa_vector_defaults.S:100

@skieast  any idea ?

Larrxi, 2021-01-02:
Hey people!
I have the LCD variant and with the code from @skieast I successfully build and make and wrote circuitpyhon.
However when I change pins to OTG mode then I cant see anything not even on dmesg. 
I am connected board USB C to my computer.
Am I missing something? Cant the circuitpython drive mount work this way?

Larrxi, 2021-01-05:
I got it working by using the board files from espressif_saola_1_wroom.

For the LCD example to work add the LCD pins from pins.c from board files from skieast. 

```
     // 1.14 inch LCD ST7789
    { MP_ROM_QSTR(MP_QSTR_LCD_MISO), MP_ROM_PTR(&pin_GPIO4) },
    { MP_ROM_QSTR(MP_QSTR_LCD_MOSI), MP_ROM_PTR(&pin_GPIO35) },
    { MP_ROM_QSTR(MP_QSTR_LCD_CS), MP_ROM_PTR(&pin_GPIO34) },
    { MP_ROM_QSTR(MP_QSTR_LCD_CLK), MP_ROM_PTR(&pin_GPIO36) },
    { MP_ROM_QSTR(MP_QSTR_LCD_RST), MP_ROM_PTR(&pin_GPIO38) },
    { MP_ROM_QSTR(MP_QSTR_LCD_BL_CTR), MP_ROM_PTR(&pin_GPIO33) },
    { MP_ROM_QSTR(MP_QSTR_LCD_D_C), MP_ROM_PTR(&pin_GPIO37) },
```

If using skieast LCD example then comment out the second display initiation:

```
display = ST7789(
	display_bus, rotation=270, width=240, height=135, rowstart=40, colstart=53
)

"""
display = displayio.Display(
	display_bus, INIT_ST7789, rotation=270, width=240, height=135, rowstart=40, colstart=53, 
)
"""
```

Add the library files for
```
adafruit_display_text
adafruit_st7789
```

And it will say Hello World! on the LCD.

netroy, 2021-01-06:
Since I was already using work shared by you folks in this conversation, I've create a [pull-request](https://github.com/adafruit/circuitpython/pull/3943).
Please let me know if any of you prefer to be the author of the commit. I can change that & update the PR.
Thanks for sharing your updates here.

skieast, 2021-01-07:
Thanks for all your work. I originally did not continue working on the port
due to issues with obtaining a unique VID/PID pair for the USB device. I
have been using Espressif VID for my internal use but that is obviously not
useful publicly. Long term the various boards using Espressif devices from
some vendors will not be easy to support.  I had some hope that Espressif
might allocate PID from their own VID but that also seems unlikely. Maybe
some large purchaser of Espressif parts can ask them directly. Other
semiconductor manufacturers currently do this for products using their
parts.

I will incorporate your changes into my own fork/branch of circuitpython at
some point in the future.  Thanks again.

Bruce Segal
bsegal@skieast.net

ladyada, 2021-01-07:
@skieast please email lilygo, they have to purchase a VID for their products just like every other manufacturer. if you email them, the support team there will be able to assign a PID. we are not customers so we cannot email them.

skieast, 2021-01-07:
Thanks will do.

skieast, 2021-01-16:
Response received from  Xinyuan-LilyGO, seems like a no-go

[image]

ladyada, 2021-01-16:
they need to be a USB organization, they are selling hardware with native USB. please continue to push them to get a vendor ID. other companies selling electronics with USB have one.

skieast, 2021-01-16:
The whole situation is not ideal. Many companies  do not have USB Vendor IDs with native usb products. Not being a lawyer or even pretending to be one I'm not commenting :)

ladyada, 2021-01-16:
please continue to push them to get a vendor ID, it is required and is a one time fee that other companies have paid. if enough customers tell them to do it, they will.

skieast, 2021-01-16:
Will do, thanks for taking the time here.

tannewt, 2021-01-20:
Another option would be to ask Espressif to get a VID and give PIDs for products with their chip. This is commonly done by other vendors. (USB-IF may not allow this anymore though.)

netroy, 2021-01-20:
Looks like Espressif already has a VID `0x303A`, and they [recently](https://github.com/espressif/usb-pids) started giving out PIDs. 
I can send them a pull request later today.

tannewt, 2021-01-20:
> Looks like Espressif already has a VID `0x303A`, and they [recently](https://github.com/espressif/usb-pids) started giving out PIDs.
> I can send them a pull request later today.

Awesome! That's great news! Thanks for the link.

netroy, 2021-01-27:
Another attempt https://github.com/adafruit/circuitpython/pull/4077

PR #4077 (2021-02-01): Add support for Lilygo TTGO T8 ESP32-S2
fixes #3414

[example code](https://gist.github.com/netroy/d1bff654fa58b884d63894ca2c890fc6)

continued from #3943 

VIDs [requested from Espressif](https://github.com/espressif/usb-pids/pull/6)
closed 2021-02-01 by tannewt