adafruit/circuitpython #3429: No REPL on cxd56 spresense

dhalbert, 2020-09-16:
@kamtom480 I have a Spresense board and camera, and thought I would try out your PR when reviewing it. I loaded the bootloader and other firmware, v2.0.000 successfully. I built the firmware, first from your PR branch and then from `main` after the merge. In both cases I built just fine, and did `make BOARD=spresense flash`. But after flashing, I did not get a `/dev/ttyACM0`, but only `/dev/ttyUSB0`. I did unplug and plug in the board after flashing when it did not work immediately after flashing.

This is on Ubuntu 20.04. The submodules are all up to date. Am I missing something? Thanks.

tannewt, 2020-09-16:
The native USB is on the breakout board I think. Could you post a picture of your setup?

dhalbert, 2020-09-16:
Thanks, you are right, it was hiding under the board, which I had mounted on a holder.

closed 2020-09-16 by dhalbert