adafruit/circuitpython #3439: Crash calling time.localtime() with no input on ESP32-S2

askpatrickw, 2020-09-18:
I was able to reproduce this on Kaluga and the FeatherS2. 
Build is off of adafruit/main as of Midnight 9/18 (Pacific)

In the REPL
```
>>> import time
>>> dir(time)
['__class__', '__name__', 'localtime', 'mktime', 'monotonic', 'monotonic_ns', 'sleep', 'struct_time', 'time']
>>> time.localtime()
```

KALUGA Debug

```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008e9aa  PS      : 0x00060430  A0      : 0x8008ea36  A1      : 0x3ffdd1e0
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3ffd12d0
A6      : 0xffffffff  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x400b0580  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffd12d0
A14     : 0x3ffd10ec  A15     : 0x00000001  SAR     : 0x0000000c  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x3ffd12d0  LCOUNT  : 0x400290dd

Backtrace:0x4008e9a7:0x3ffdd1e0 0x4008ea33:0x3ffdd200 0x4008d48a:0x3ffdd220 0x4008d4e9:0x3ffdd240 
0x4008d540:0x3ffdd260 0x400aab70:0x3ffdd290 0x400b058d:0x3ffdd2c0 0x4009150b:0x3ffdd2f0 
0x4008d26d:0x3ffdd320 0x4008d36d:0x3ffdd340 0x4009ab46:0x3ffdd360 0x400914a6:0x3ffdd400 
0x4008d26d:0x3ffdd430 0x4008d29a:0x3ffdd450 0x400a8732:0x3ffdd470 0x400a8a55:0x3ffdd520 
0x4009e144:0x3ffdd560 0x4009e1a9:0x3ffdd580 0x4009e39f:0x3ffdd5a0 0x4002fa2c:0x3ffdd5c0

ELF file SHA256: 4059e40b5f14639e
CPU halted.
```

`python tools/decode_backtrace.py espressif_kaluga_1`
```
espressif_kaluga_1
? 0x4008e9a7:0x3ffdd1e0 0x4008ea33:0x3ffdd200 0x4008d48a:0x3ffdd220 0x4008d4e9:0x3ffdd240 
0x4008d540:0x3ffdd260 0x400aab70:0x3ffdd290 0x400b058d:0x3ffdd2c0 0x4009150b:0x3ffdd2f0 
0x4008d26d:0x3ffdd320 0x4008d36d:0x3ffdd340 0x4009ab46:0x3ffdd360 0x400914a6:0x3ffdd400 
0x4008d26d:0x3ffdd430 0x4008d29a:0x3ffdd450 0x400a8732:0x3ffdd470 0x400a8a55:0x3ffdd520 
0x4009e144:0x3ffdd560 0x4009e1a9:0x3ffdd580 0x4009e39f:0x3ffdd5a0 0x4002fa2c:0x3ffdd5c0

got ['0x4008e9a7', '0x4008ea33', '0x4008d48a', '0x4008d4e9', '0x4008d540', '0x400aab70', '0x400b058d', 
'0x4009150b', '0x4008d26d', '0x4008d36d', '0x4009ab46', '0x400914a6', '0x4008d26d', '0x4008d29a', '0x400a8732', 
'0x400a8a55', '0x4009e144', '0x4009e1a9', '0x4009e39f', '0x4002fa2c']
/Users/askpatrickw/code/github/circuitpython/ports/esp32s2/../../py/obj.c:52
/Users/askpatrickw/code/github/circuitpython/ports/esp32s2/../../py/obj.c:46
/Users/askpatrickw/code/github/circuitpython/ports/esp32s2/../../py/runtime.c:1060
/Users/askpatrickw/code/github/circuitpython/ports/esp32s2/../../py/runtime.c:1093
/Users/askpatrickw/code/github/circuitpython/ports/esp32s2/../../py/runtime.c:931
/Users/askpatrickw/code/github/circuitpython/ports/esp32s2/../../shared-bindings/rtc/__init__.c:46
/Users/askpatrickw/code/github/circuitpython/ports/esp32s2/../../shared-bindings/time/__init__.c:242
/Users/askpatrickw/code/github/circuitpython/ports/esp32s2/../../py/objfun.c:129
/Users/askpatrickw/code/github/circuitpython/ports/esp32s2/../../py/runtime.c:624
/Users/askpatrickw/code/github/circuitpython/ports/esp32s2/../../py/runtime.c:639
/Users/askpatrickw/code/github/circuitpython/ports/esp32s2/../../py/vm.c:1017
/Users/askpatrickw/code/github/circuitpython/ports/esp32s2/../../py/objfun.c:286
/Users/askpatrickw/code/github/circuitpython/ports/esp32s2/../../py/runtime.c:624
/Users/askpatrickw/code/github/circuitpython/ports/esp32s2/../../py/runtime.c:598
/Users/askpatrickw/code/github/circuitpython/ports/esp32s2/../../lib/utils/pyexec.c:114
/Users/askpatrickw/code/github/circuitpython/ports/esp32s2/../../lib/utils/pyexec.c:520
/Users/askpatrickw/code/github/circuitpython/ports/esp32s2/../../main.c:439
/Users/askpatrickw/code/github/circuitpython/ports/esp32s2/../../main.c:504
/Users/askpatrickw/code/github/circuitpython/ports/esp32s2/supervisor/port.c:198
/Users/askpatrickw/code/github/circuitpython/ports/esp32s2/build-espressif_kaluga_1/esp-idf/../../esp-idf/components/freertos/xtensa/port.c:511
```

NOTE: Calling it correctly does work
```
>>> import time
>>> time.mktime((2020, 9, 14, 23, 22, 15, 0, -1, -1))
1600125735
>>> time.localtime(1600125735)
struct_time(tm_year=2020, tm_mon=9, tm_mday=14, tm_hour=23, tm_min=22, tm_sec=15,
tm_wday=0, tm_yday=258, tm_isdst=-1)
```

tannewt, 2020-09-18:
@askpatrickw want to take a crack at this? I have a feeling we're trying to get the time from an uninitialized RTC object.

askpatrickw, 2020-09-21:
I looked at this over the weekend just to give a try and since I don't know C, I didn't figure it out. I'll tell you what I tried in case its meaningful

I looked at [`common-hal/time.c`](https://github.com/adafruit/circuitpython/blob/main/ports/esp32s2/common-hal/time/__init__.c) and tried adding includes for RTC but that didn't change anything.

```
#include "shared-bindings/rtc/RTC.h"
#include "shared-bindings/time/__init__.h"
```

Like I said though, I don't know C and was just trying stuff. Someone who knows C would be better to fix this.

tannewt, 2020-09-21:
@hierophect Want to take a look at this? I suspect the internal RTC object is null.

anecdata, 2020-09-22:
`time.time()` crashes too, even after initializing RTC:
```
Adafruit CircuitPython 6.0.0-beta.0 on 2020-09-21; Saola 1 w/Wrover with ESP32S2
>>> import time
>>> import rtc
>>> rtc.RTC().datetime = time.localtime(946684800)
>>> time.time()

```
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008e9e6  PS      : 0x00060930  A0      : 0x8008ea72  A1      : 0x3ffdd1e0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x3ffd12c0  
A6      : 0xffffffff  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x3f014600  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x3ffd12c0  
A14     : 0x3ffd10dc  A15     : 0x00000001  SAR     : 0x0000000c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x3ffd12c0  LCOUNT  : 0x400290dd  

Backtrace:0x4008e9e3:0x3ffdd1e0 0x4008ea6f:0x3ffdd200 0x4008d4c6:0x3ffdd220 0x4008d525:0x3ffdd240 0x4008d57c:0x3ffdd260 0x400aabd4:0x3ffdd290 0x400b0633:0x3ffdd2c0 0x4009142a:0x3ffdd2f0 0x4008d2a9:0x3ffdd310 0x4008d3a9:0x3ffdd330 0x4009ab82:0x3ffdd350 0x400914e2:0x3ffdd3f0 0x4008d2a9:0x3ffdd420 0x4008d2d6:0x3ffdd440 0x400a8796:0x3ffdd460 0x400a8ab9:0x3ffdd510 0x4009e180:0x3ffdd550 0x4009e1e5:0x3ffdd570 0x4009e3db:0x3ffdd590 0x4002fa28:0x3ffdd5b0

ELF file SHA256: 2720569dd55ad93f

CPU halted.
```

anecdata, 2020-09-22:
PyPortal, for comparison:
```
Adafruit CircuitPython 6.0.0-beta.0 on 2020-09-21; Adafruit PyPortal with samd51j20
>>> import time
>>> import rtc
>>> rtc.RTC().datetime = time.localtime(946684800)
>>> time.time()
946684800
>>> time.time()
946684813
```

askpatrickw, 2020-09-22:
Possibly related where I implemented RTC
https://github.com/adafruit/circuitpython/pull/3422

microDev1, 2020-09-22:
Got a fix for this. Getting a PR ready.

tannewt, 2020-09-22:
Fixed by #3451. Thanks @microDev1

closed 2020-09-22 by tannewt