adafruit/circuitpython #3444: Implement reset-to-dfu on STM MCUs without UF2 bootloaders

jepler, 2020-09-20:
Following the example of micropython we should be able to reset into the DFU bootloader from code: https://github.com/micropython/micropython/blob/0986675451edbdcbe31d90ddacf8f6dc3327a4ae/stmhal/modmachine.c#L197-L223
```
STATIC NORETURN mp_obj_t machine_bootloader(void) {
    pyb_usb_dev_deinit();
    storage_flush();

    HAL_RCC_DeInit();
    HAL_DeInit();

#if defined(MCU_SERIES_F7)
    // arm-none-eabi-gcc 4.9.0 does not correctly inline this
    // MSP function, so we write it out explicitly here.
    //__set_MSP(*((uint32_t*) 0x1FF00000));
    __ASM volatile ("movw r3, #0x0000\nmovt r3, #0x1FF0\nldr r3, [r3, #0]\nMSR msp, r3\n" : : : "r3", "sp");

    ((void (*)(void)) *((uint32_t*) 0x1FF00004))();
#else
    __HAL_REMAPMEMORY_SYSTEMFLASH();

    // arm-none-eabi-gcc 4.9.0 does not correctly inline this
    // MSP function, so we write it out explicitly here.
    //__set_MSP(*((uint32_t*) 0x00000000));
    __ASM volatile ("movs r3, #0\nldr r3, [r3, #0]\nMSR msp, r3\n" : : : "r3", "sp");

    ((void (*)(void)) *((uint32_t*) 0x00000004))();
#endif

    while (1);
}
```
This would go in supervisor/port.c in reset_to_bootloader, ideally paired with some alternate "#if UF2" code to preferentially enter the uf2 bootloader.  I don't have any STM boards with UF2 bootloader so I don't know what that sequence is; it differs between samd and nrf so I don't think there's a standard.

Additional discussion: https://stackoverflow.com/questions/26891432/jump-to-bootloader-in-stm32-through-application-i-e-using-boot-0-and-boot-1-pins

flom84, 2022-04-22:
Hi!

On my `nucleo-f446re board` from ST, I've been trying to add support for resetting the MCU in DFU mode.
Because circuitpython does not yet support this board, I decided to give it a shot and try running it on this board.

I've managed to execute the 'REPL' from a serial console so far.
My local fork has a change that adds support for the board: https://github.com/flom84/circuitpython/commit/8bc5752e1f647f6ffe74a287becdbf303fbc2079.

I also implemented reset to DFU mode once I had something functioning.  Commit can be found here:
 commit: https://github.com/flom84/circuitpython/commit/9f6949bf678871899e96b0bffbc6f49e45a54461
The python code bellow have been used to make the CPU enter in DFU mode.
`
import microcontroller
microcontroller.on_next_reset(microcontroller.RunMode.BOOTLOADER)
microcontroller.reset()
`

If this may be used as a starting point for adding DFU mode, any feedback or suggestions are welcome .

Thanks.

skerr92, 2022-10-16:
[PR Merged](https://github.com/adafruit/circuitpython/pull/6919) on this issue 28 days ago, is this fixed?

dhalbert, 2022-10-16:
I think so! Thanks, fixed by #6919.

closed 2022-10-16 by dhalbert