adafruit/circuitpython #3460: Fix pin printing

tannewt, 2020-09-23:
When a pin is listed in `board`, it should print out with its first `board` name instead of `microcontroller.pin`.

Code is here: https://github.com/adafruit/circuitpython/commit/168aa394dba31449cdb267444992b4f636906aaa#diff-df3dad56701b150d4292baceaa18e2c2R77

With a pin definition like:

```c
STATIC const mp_rom_map_elem_t board_global_dict_table[] = {
    { MP_ROM_QSTR(MP_QSTR_A0), MP_ROM_PTR(&pin_GPIO17) },
    { MP_ROM_QSTR(MP_QSTR_D14), MP_ROM_PTR(&pin_GPIO17) },
}
```

This

```
>>> print(board.A0)
microcontroller.pin.A0
>>> print(board.D14)
microcontroller.pin.A0
```

should be

```
>>> print(board.A0)
board.A0
>>> print(board.D14)
board.A0
```

askpatrickw, 2020-09-23:
@tannewt Is it really just this simple?

```diff
diff --git a/shared-bindings/microcontroller/Pin.c b/shared-bindings/microcontroller/Pin.c
index 1294b78f0..409d998ee 100644
--- a/shared-bindings/microcontroller/Pin.c
+++ b/shared-bindings/microcontroller/Pin.c
@@ -71,7 +71,7 @@ STATIC void mcu_pin_print(const mp_print_t *print, mp_obj_t self_in, mp_print_ki
     qstr name;

     get_pin_name(self, &package, &module, &name);
-    mp_printf(print, "%q.%q.%q", MP_QSTR_microcontroller, MP_QSTR_pin, name);
+    mp_printf(print, "%q.%q.%q", MP_QSTR_board, MP_QSTR_pin, name);
 }

 const mp_obj_type_t mcu_pin_type = {
```

tannewt, 2020-09-24:
@askpatrickw I don't think so because you need to account for printing objects that aren't in board as well.

That will also print `board.pin.A0` I believe too.

askpatrickw, 2020-09-24:
Ok, I played around with this on my Kaluga

```diff
diff --git a/shared-bindings/microcontroller/Pin.c b/shared-bindings/microcontroller/Pin.c
index 1294b78f0..5fc402316 100644
--- a/shared-bindings/microcontroller/Pin.c
+++ b/shared-bindings/microcontroller/Pin.c
@@ -71,7 +71,7 @@ STATIC void mcu_pin_print(const mp_print_t *print, mp_obj_t self_in, mp_print_ki
     qstr name;

     get_pin_name(self, &package, &module, &name);
-    mp_printf(print, "%q.%q.%q", MP_QSTR_microcontroller, MP_QSTR_pin, name);
+    mp_printf(print, "%q.%q.%q", module, MP_QSTR_pin, name);
 }

 const mp_obj_type_t mcu_pin_type = {
```

```python
>>> print(microcontroller.pin.GPIO0)
board.pin.IO0
>>> print(board.IO0)
board.pin.IO0
```

I see the scenario where printing microcontroller.pin shows up as board.
But I'm struggling with a test where there's a non-board pin to verify it works 100%.

askpatrickw, 2020-09-25:
Commented out  IO0

```diff
diff --git a/ports/esp32s2/boards/espressif_kaluga_1/pins.c b/ports/esp32s2/boards/espressif_kaluga_1/pins.c
index c1657d6c0..bd095d287 100644
--- a/ports/esp32s2/boards/espressif_kaluga_1/pins.c
+++ b/ports/esp32s2/boards/espressif_kaluga_1/pins.c
@@ -1,7 +1,7 @@
 #include "shared-bindings/board/__init__.h"

 STATIC const mp_rom_map_elem_t board_global_dict_table[] = {
-    { MP_ROM_QSTR(MP_QSTR_IO0), MP_ROM_PTR(&pin_GPIO0) },
+    // { MP_ROM_QSTR(MP_QSTR_IO0), MP_ROM_PTR(&pin_GPIO0) },
     { MP_ROM_QSTR(MP_QSTR_IO1), MP_ROM_PTR(&pin_GPIO1) },
     { MP_ROM_QSTR(MP_QSTR_IO2), MP_ROM_PTR(&pin_GPIO2) },
     { MP_ROM_QSTR(MP_QSTR_IO3), MP_ROM_PTR(&pin_GPIO3) },
diff --git a/shared-bindings/microcontroller/Pin.c b/shared-bindings/microcontroller/Pin.c
index 1294b78f0..5fc402316 100644
--- a/shared-bindings/microcontroller/Pin.c
+++ b/shared-bindings/microcontroller/Pin.c
@@ -71,7 +71,7 @@ STATIC void mcu_pin_print(const mp_print_t *print, mp_obj_t self_in, mp_print_ki
     qstr name;

     get_pin_name(self, &package, &module, &name);
-    mp_printf(print, "%q.%q.%q", MP_QSTR_microcontroller, MP_QSTR_pin, name);
+    mp_printf(print, "%q.%q.%q", module, MP_QSTR_pin, name);
 }

 const mp_obj_type_t mcu_pin_type = {
```

No IO0

```python
>>> dir(board)
['__class__', 'AUDIO_BT_ADC', 'AUDIO_I2S0_BCLK', 'AUDIO_I2S0_LRCK', 'AUDIO_I2S0_MCLK', 'AUDIO_I2S0_SDI', 'AUDIO_I2S0_SDO', 'AUDIO_I2S1_BCLK_DAC2', 'AUDIO_I2S1_LRCK_DAC1', 'AUDIO_I2S1_MCLK', 'AUDIO_I2S1_SDI', 'AUDIO_I2S1_SDO', 'AUDIO_PA_CTRL', 'AUDIO_RST', 'AUDIO_SCL', 'AUDIO_SDA', 'AUDIO_SPI_CS', 'AUDIO_SPI_MISO', 'AUDIO_SPI_MOSI', 'AUDIO_SPI_SCK', 'AUDIO_WAKE_INT', 'BT_ARRAY_ADC', 'CAMERA_D2', 'CAMERA_D3', 'CAMERA_D4', 'CAMERA_D5', 'CAMERA_D6', 'CAMERA_D7', 'CAMERA_D8', 'CAMERA_D9', 'CAMERA_HREF', 'CAMERA_PCLK', 'CAMERA_SIOC', 'CAMERA_SIOD', 'CAMERA_VSYNC', 'CAMERA_XCLK', 'IO1', 'IO10', 'IO11', 'IO12', 'IO13', 'IO14', 'IO15', 'IO16', 'IO17', 'IO18', 'IO19', 'IO2', 'IO20', 'IO21', 'IO26', 'IO3', 'IO33', 'IO34', 'IO35', 'IO36', 'IO37', 'IO38', 'IO39', 'IO4', 'IO40', 'IO41', 'IO42', 'IO43', 'IO44', 'IO45', 'IO46', 'IO5', 'IO6', 'IO7', 'IO8', 'IO9', 'LCD_BL_CTR', 'LCD_CLK', 'LCD_CS', 'LCD_D_C', 'LCD_MISO', 'LCD_MOSI', 'LCD_RST', 'LCD_TP_BUSY', 'LCD_TP_CS', 'LCD_TP_IRQ', 'LCD_TP_MISO', 'LCD_TP_MOSI', 'LCD_TP_SCK', 'NEOPIXEL', 'RX', 'SCL', 'SDA', 'TOUCH1', 'TOUCH10', 'TOUCH11', 'TOUCH12', 'TOUCH13', 'TOUCH14', 'TOUCH2', 'TOUCH3', 'TOUCH4', 'TOUCH5', 'TOUCH6', 'TOUCH7', 'TOUCH8', 'TOUCH9', 'TX']
>>> print(board.IO0)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'IO0'
```

print shows up as pin. not board. so we know we're printing what we're passed... 

```python
>>> dir(microcontroller.pin)
['__class__', 'GPIO0', 'GPIO1', 'GPIO10', 'GPIO11', 'GPIO12', 'GPIO13', 'GPIO14', 'GPIO15', 'GPIO16', 'GPIO17', 'GPIO18', 'GPIO19', 'GPIO2', 'GPIO20', 'GPIO21', 'GPIO26', 'GPIO27', 'GPIO28', 'GPIO29', 'GPIO3', 'GPIO30', 'GPIO31', 'GPIO32', 'GPIO33', 'GPIO34', 'GPIO35', 'GPIO36', 'GPIO37', 'GPIO38', 'GPIO39', 'GPIO4', 'GPIO40', 'GPIO41', 'GPIO42', 'GPIO43', 'GPIO44', 'GPIO45', 'GPIO46', 'GPIO5', 'GPIO6', 'GPIO7', 'GPIO8', 'GPIO9']
>>> print(microcontroller.pin.GPIO0)
pin.pin.GPIO0
```

askpatrickw, 2020-09-25:
I played around with this a bit and came up with a conditional statement that outputs what I think is correct, but I'm not sure if there is some legacy behavior or other requirement about _always_ having "pin" in there, I could add that back in for when the package is not present. 

```diff
diff --git a/ports/esp32s2/boards/espressif_kaluga_1/pins.c b/ports/esp32s2/boards/espressif_kaluga_1/pins.c
index c1657d6c0..bd095d287 100644
--- a/ports/esp32s2/boards/espressif_kaluga_1/pins.c
+++ b/ports/esp32s2/boards/espressif_kaluga_1/pins.c
@@ -1,7 +1,7 @@
 #include "shared-bindings/board/__init__.h"

 STATIC const mp_rom_map_elem_t board_global_dict_table[] = {
-    { MP_ROM_QSTR(MP_QSTR_IO0), MP_ROM_PTR(&pin_GPIO0) },
+    // { MP_ROM_QSTR(MP_QSTR_IO0), MP_ROM_PTR(&pin_GPIO0) },
     { MP_ROM_QSTR(MP_QSTR_IO1), MP_ROM_PTR(&pin_GPIO1) },
     { MP_ROM_QSTR(MP_QSTR_IO2), MP_ROM_PTR(&pin_GPIO2) },
     { MP_ROM_QSTR(MP_QSTR_IO3), MP_ROM_PTR(&pin_GPIO3) },
diff --git a/shared-bindings/microcontroller/Pin.c b/shared-bindings/microcontroller/Pin.c
index 1294b78f0..6ba02a0e7 100644
--- a/shared-bindings/microcontroller/Pin.c
+++ b/shared-bindings/microcontroller/Pin.c
@@ -71,7 +71,11 @@ STATIC void mcu_pin_print(const mp_print_t *print, mp_obj_t self_in, mp_print_ki
     qstr name;

     get_pin_name(self, &package, &module, &name);
-    mp_printf(print, "%q.%q.%q", MP_QSTR_microcontroller, MP_QSTR_pin, name);
+    if (package){
+        mp_printf(print, "%q.%q.%q", package, module, name);
+    } else {
+        mp_printf(print, "%q.%q", module , name);
+    }
 }

 const mp_obj_type_t mcu_pin_type = {
```

```python
>>> print(microcontroller.pin.GPIO0)  # pin not mapped in pins.c
microcontroller.pin.GPIO0
>>> print(microcontroller.pin.GPIO1)  # pin mapped in pins.c
board.IO1

>>> print(board.IO1)
board.IO1
>>> print(board.TOUCH1)
board.IO1
>>> print(board.TOUCH2)
board.IO2
```

closed 2020-09-25 by tannewt