adafruit/circuitpython #3461: "edit on github" link for canio docs is a broken link

siddacious, 2020-09-23:
on the page:
https://circuitpython.readthedocs.io/en/latest/shared-bindings/canio/index.html#canio.Message 
this link:
[image]
points here, which doesn't exist:
https://github.com/adafruit/circuitpython/blob/main/shared-bindings/canio/index.rst

tannewt, 2020-09-24:
I'm not sure this is doable because we generate index.rst based on the .c files in shared-bindings.

siddacious, 2020-09-24:
I don't mind if there is a link to actually edit the docs or not, I'd just don't like broken links.

Can you link to the processes and/or scripts that are responsible for generating the .rst files? I assume it's run by Actions?

jepler, 2020-09-24:
See also #3400

tannewt, 2020-09-25:
I think we should close this in favor of #3400 

> Can you link to the processes and/or scripts that are responsible for generating the .rst files? I assume it's run by Actions?

https://github.com/adafruit/circuitpython/blob/main/.github/workflows/build.yml#L70-L77

`make stubs` strips the `//|` lines out of the `.c` files in `shared-bindings` to create `.pyi` files. `sphinx-build` then uses the `autoapi` plugin to convert `.pyi` to `.rst` which is then rendered to the docs.

closed 2020-09-25 by tannewt