adafruit/circuitpython #3468: displayio(?): sometimes doesn't repaint when objects change/move

PaintYourDragon, 2020-09-25:
Not 100% certain whether the underlying issue is in displayio or RGBmatrix. Regardless: sometimes after changing elements of a displayed group, nothing happens on the display, or only some of the changes take effect. Is there a way to force a full repaint for good measure?

This is sometimes seen on the Moon clock. It might stick on the splash screen (though I’ve confirmed it’s doing Moon math and updating elements in the active displayed group), or might only redraw a subset of elements on the time/phase display (again, same, clock is doing its thing, display just isn’t updating all elements, sometimes repeated calls help).

I don’t have a 100% reliable way to reproduce this yet, it’s an intermittent thing that just happens sometimes.

kevinjwalters, 2020-09-25:
For what it's worth, I've done a fair bit of `displayio` using various 5.x versions on CLUE and TFT Gizmo and haven't seen bugs like this. The one thing I have seen is surprising visual (and audio) effects of handling the pads which can happen with the TFT Gizmo depending on how you hold it: [Circuit Playground Bluefruit with TFT Gizmo vs fingers - curious video artefacts](https://www.youtube.com/watch?v=h5VPk8JsDXk) - this doesn't sound anything like what's being described in issue ticket, though.

kevinjwalters, 2020-09-25:
You may already be aware of this but there's a way of dumping the screen to a file described in [https://learn.adafruit.com/saving-bitmap-screenshots-in-circuitpython](Adafruit Learn: Saving CircuitPython Bitmaps and Screenshots) using the `adafruit_bitmapsaver` library. That might be useful for debugging to gather a bit more information on where the problem is likely to lie.

tannewt, 2020-09-25:
> Not 100% certain whether the underlying issue is in displayio or RGBmatrix. Regardless: sometimes after changing elements of a displayed group, nothing happens on the display, or only some of the changes take effect. Is there a way to force a full repaint for good measure?

You can force a full repaint by switching the top level group to a different group and back. However, I don't think we should do that. Instead, we should figure out what the bug is.

How are you changing the elements of the group? The code to do it is here: https://github.com/adafruit/circuitpython/blob/main/shared-module/displayio/Group.c

kmatch98, 2020-09-25:
There is a new `refresh` option to refresh automatically of any dirty rectangles https://github.com/adafruit/circuitpython/pull/3366

But if you need a hard refresh of everything on the screen then @tannewt suggestion of `display.show(None)` and then back to your Group, will work too.

tannewt, 2020-10-08:
@PaintYourDragon Are you still seeing this issue?

PaintYourDragon, 2020-10-09:
Clock seems to be behaving now, and a simpler “demonstrate the issue” program is working fine, so it appears to be resolved and I’ll close this item. Thanks for taking a look!

closed 2020-10-09 by PaintYourDragon