adafruit/circuitpython #3473: Memory corruption: feather nrf52840 + sharp memory display 400x240

jepler, 2020-09-25:
I have a large-ish program (RPN calculator with decimal arithmetic) that runs on this hardware combination.  After reloading it an indeterminate number of times, I get an inexplicable traceback like
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 4, in <module>
  File "__init__.py", line 5433, in <module>
RuntimeError: Corrupt raw code
```
or I'll get a full hang after trying to enter safe mode.  The safe mode entry looks like nonsense:
```
Breakpoint 1, reset_into_safe_mode (reason=reason@entry=HARD_CRASH)
    at ../../supervisor/shared/safe_mode.c:95
95	void __attribute__((noinline,)) reset_into_safe_mode(safe_mode_t reason) {
(gdb) n
96	    if (current_safe_mode > BROWNOUT && reason > BROWNOUT) {
(gdb) where
#0  reset_into_safe_mode (reason=reason@entry=HARD_CRASH)
    at ../../supervisor/shared/safe_mode.c:96
#1  0x0004457c in HardFault_Handler () at supervisor/port.c:347
#2  <signal handler called>
#3  0x00033226 in mp_map_lookup (map=0x200389a4, index=0x22080f00, lookup_kind=<optimized out>)
    at ../../py/obj.h:171
#4  0x000551bc in irq_handler (p_reg=0x200127ac <usb_callback>, instance_id=279173, 
    channel_count=283785) at nrfx/drivers/src/nrfx_rtc.c:308
#5  0x200389a4 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)

```
As yet, I haven't captured any better debugging information than this.  I also haven't distilled down a reproducer smaller than my full calculator program.

I have jumped to the possibly-incorrect conclusion that it's something about the Sharp display or framebufferio that allows memory corruption to occur during board reset, but I don't have proof yet.  Nothing else that my program does is new core functionality.

DavePutz, 2020-09-28:
@jepler , are you able to attach the program you are running so I can test?

jepler, 2020-09-29:
Sadly my program is too big and ugly to post as-is.  If I find a reduced test case I'll post it.

closed 2020-10-01 by tannewt