adafruit/circuitpython #3477: ESP32-S2 mounted SD not accessible from PC?

emard, 2020-09-26:
Recent ESP32-S2 release can mount SD card,
after setting pinout and SPI, this works
storage.mount(vfs, "/sd")
os.listdir("/sd") lists files stored on the SD card.
['KERNEL.SYS', 'COMMAND.COM', 'INVADERS.COM', etc..

Now I want to copy a file from PC to the mounted SD card 
but linux PC usb-storage interface doesn't "see" mounted
"/sd" directory

cp README.md /media/guest/CIRCUITPY/sd/
cp: cannot create regular file '/media/guest/CIRCUITPY/sd/': Not a directory

tannewt, 2020-09-28:
We don't support the SD card over USB. The USB Mass Storage protocol works on the blocks below the filesystem. So, we'd need to expose the SD card as a second drive (logical unit in USB parlance). It would have the same read/write constraints as the internal flash.

emard, 2020-09-28:
OK, let SD-over-USB be a feature request,
it's ok for me to appear as another drive.

Intuitively, I have expected this to "just work".

Now I need to first upload file from USB to internal flash,
write my own copy.py application, upload it too, then
enter python prompt and finally copy this file from flash to SD.

emard, 2020-12-23:
Can you provide some boot or commandline option
for usb-storage to be directed towards SD card instead
of internal flash?

tannewt, 2020-12-23:
> Can you provide some boot or commandline option
> for usb-storage to be directed towards SD card instead
> of internal flash?

It's not a high priority for us and it's not simple. The native sdcard support is a good start though.

calcut, 2021-06-26:
@tannewt I asked about this yesterday in the deep dive chat.

From the discussion here https://github.com/adafruit/circuitpython/issues/2206 with @dhalbert 
I thought the new CP7 USB endpoint customisation might make this easier.

But sounds like there is another subtlety that makes it tricky.
Appreciate it isn't high priority just now.

dhalbert, 2021-06-26:
> From the discussion here #2206 with @dhalbert
> I thought the new CP7 USB endpoint customisation might make this easier.

It doesn't really help that much, though the internal structure makes it a bit easier to add another MSC device. But there still has to be code behind that to make the SD card visible, and that's not easy.

Paul-iKarma, 2021-08-02:
Newbie question.
Was hoping to connect this like a USB memory stick, write jpeg files from the PC and have those changes trigger the posting of file names and sizes to a remote server. Is this possible? Perhaps with some onboard program that might, for example, switch connection of the SD card back and forth between the unit and the PC so the card could take a peek at the new files? Or completely impossible given the current specs? Thanks.

dhalbert, 2021-08-02:
See #2006 and #1764 for why this is hard.

lesamouraipourpre, 2021-08-06:
For transferring **text files** to and from the SD card, you can use `Thonny` (and maybe other editors, eg `Mu`, but I haven't tried)

Once `code.py` has run and mounted the SD card, the `/sd/` directory shows up when you try to read or write a file to the CircuitPython device. I don't understand how it actually works but it does.

I don't think this can be used for transferring binary files but someone might be able to figure out something.

rgbpcb, 2022-07-03:
> See #2206 and #1764 for why this is hard.

[A newer comment](https://github.com/adafruit/circuitpython/issues/2206#issuecomment-892167126) in #2206 says:

> This is not an issue with endpoint count. USB MSC can have multiple drives over the same endpoints by using a logical unit (LUN) index.

Is implementing this the hard part? If so, could you please add the option to show the SD card **_instead of_** the usual CIRCUITPY filesystem?

For example, `boot.py` could mount the SD card and read some file from it that says whether the SD card should be shown to the computer instead of the usual CIRCUITPY.

Giving the computer easy access to the SD card (which is by far the largest storage of the device) seems kind of important. :) Thanks!

dhalbert, 2022-07-03:
We are working on making multiple drives available. There are some issues about making it work properly on all operating systems, which we're working through.

rgbpcb, 2022-07-03:
That's great, thank you! Looking forward to the Windows solution :sweat_smile:

hathach, 2022-07-04:
just submit an PR for this #6555

PR #10129 (2025-04-02): Automount SD cards
And make them available over USB MSC. They can be remount to make them writable (slowly) from the host PC.

Fixes #9954. Fixes #8678. Fixes #3477
commit d0c84009 (2025-04-02): Automount SD cards

And make them available over USB MSC. They can be remount to make
them writable (slowly) from the host PC.

Fixes #9954. Fixes #8678. Fixes #3477
closed 2025-04-02 by dhalbert