adafruit/circuitpython #3479: readthedocs removes file extension from URLs

PTS93, 2020-09-27:
If you got to f.e. https://circuitpython.readthedocs.io/en/latest/CONTRIBUTING.html
Clicking on any link that is supposed to go to a markdown file in the github repo yields a 404 because it removes the file extension.
F.e. instead of linking to `https://github.com/adafruit/circuitpython/blob/main/CODE_OF_CONDUCT.md` (notice the **.md**)
it links to `https://github.com/adafruit/circuitpython/blob/main/CODE_OF_CONDUCT`

The markdown file this page is based on is correct though, it includes the file ending in the URL. So I assume this gets lost in the processing somewhere.

sw23, 2020-10-24:
I explored this issue as part of hacktoberfest and found that this is actually an issue with the recommonmark package, and has been fixed (though a new package version has yet to be released):
https://github.com/readthedocs/recommonmark/pull/181
https://github.com/readthedocs/recommonmark/commit/e821106417bddaa76b1808cf4fce7d65d2459dd2

It's probably worth waiting for a new package version.

PTS93, 2020-10-24:
Great find, thanks. I nudged them if they might consider a release, this seems to have been the only commit since may.

tannewt, 2020-10-26:
@sw23 thank for looking into this! Please feel free to mention me on any PRs to circuitpython or libraries that you want to count for Hacktoberfest.

theacodes, 2020-12-13:
We could install directly from source in our requirements.txt:

```
git+https://github.com/readthedocs/recommonmark.git@e821106#egg=recommonmark
```

I'm happy to send a PR to do this if y'all want.

tekktrik, 2022-04-20:
Has this been fixed?  It looks like the CODE OF CONDUCT now links to an online page (HTML) instead of 404'ing.  Or am I misunderstanding?

prcutler, 2022-05-31:
> Has this been fixed? It looks like the CODE OF CONDUCT now links to an online page (HTML) instead of 404'ing. Or am I misunderstanding?

tekktrik appears to be right.  I tested both the original link and the only other Markdown I could file, and both pages load the HTML page and no longer 404:

* https://docs.circuitpython.org/en/latest/docs/common_hal.html (common_hal.md)

Unless I'm misunderstanding too, this can be marked fixed and closed.

--Paul

tekktrik, 2022-07-21:
@adafruit/circuitpythonlibrarians I think this can potentially be closed

closed 2022-07-21 by jepler