adafruit/circuitpython #3506: CircuitPython core code crashed hard. Whoops!

wifijt, 2020-10-02:
CircuitPython 6 Beta 1 

Matrix Portal M4

Trying to run the Moon Phase Clock for Adafruit Matrix Portal

Crashes hard and I have to reload CircuitPython from the bootloader to recover

[boot_out.txt](https://github.com/adafruit/circuitpython/files/5319504/boot_out.txt)

[code.txt](https://github.com/adafruit/circuitpython/files/5319509/code.txt)

jepler, 2020-10-02:
I have reproduced this problem on my matrixportal.  I ran the following lines at the REPL and got the same behavior (safe mode reset):
```
>>> from adafruit_matrixportal.matrix import Matrix
>>> MATRIX = Matrix(bit_depth=4)
```

closed 2020-10-03 by ladyada

jepler, 2020-10-03:
Thanks for the report @wifijt -- you should now be able to grab a fixed firmware by using the "absolute latest" link at https://circuitpython.org/board/matrixportal_m4/ or wait for the next prerelease to come out.  Either way, please let us know your results!

PR #3507 (2020-10-03): displayio: Fix matrixportal crash
An RGBMatrix has no bus and no bus_free method.  It is always possible to refresh the display.

This was not a problem before, but the fix I suggested (#3449) added a call to core_bus_free when a FramebufferDisplay was being refreshed. This was not caught during testing.

This is a band-aid fix and it brings to light a second problem in which a SharpDisplay + FrameBuffer will not have a 'bus' object, and yet does operate using a shared SPI bus.  This kind of display will need a "bus-free" like function to be added, or it can have problems like #3309.

Closes:  [... cut]