adafruit/circuitpython #3508: Fix FrameBufferDisplay, displayio_display_core_start_refresh & bus_free

jepler, 2020-10-02:
#3507 papers over a problem where we called `self->bus_free(self->bus);` when `bus_free` was a NULL function pointer.  In the case of RGBMatrix, this is fine.  In the case of SharpDisplay, however, there is the possibility that the underlying SPI bus will be locked and we need to deal with it.

This _could_ be done by
 * adding a new Framebuffer protocol to check for bus free (default/NULL function indicates bus is always free)
    * and implementing it for SharpDisplay, checking that the bus is in fact free
 * assigning this protocol to display_core.bus_free
 * removing the NULL check I added for `self->bus`

Presumably, without this fix, it is possible for a lockup to happen with an auto-refresh SharpDisplay and another foreground use of the SPI bus, including SD cards.  However, I did not actually test anything in this area.

hatchman, 2020-10-09:
I can confirm some issues with the sharp display not able to access spi in certain conditions. I have a project with a bluefruit sense nrf52840 and a 2.7 sharp memory display. I have code that runs through a menu to sepect and run sub files, and each sub file has a routine with supervisor to issue a reload command. Oddly if i use displayio.release_displays() and issue the reboot, on soft reboot it claims spi bus is busy when the display is initialized. If i hard reboot, the code functions as intended. 

I also leveraged the gc library to check memory at certain points in my code and found that on a hard reboot i get roughly 7,000 more bytes of free memory available on initialization. on EVERY soft reboot, i loose 000-10,000 bytes of memory at the start of my code where only libraraies have been called. Not sure what is being held on to but the impact is significant where certain code can only run on a fresh power cycle and not on a ctrl d or softreboot with code.

dhalbert, 2023-10-28:
@tannewt Is this still an issue?

closed 2023-11-20 by tannewt