adafruit/circuitpython #3515: /devices.h: plz Add P25Q32H 4 MiB flash chip support.

hanyelkomy, 2020-10-07:
Hi, Could you please add support to this external flash cause i cant get my hand on any of the supported flash devices with uson 2x3package and greater than 2MB except this one.
Thanks in advance!

Datasheet: 
[PUYA-P25Q32H-UXH-IR.pdf](https://github.com/adafruit/circuitpython/files/5339842/PUYA-P25Q32H-UXH-IR.pdf)

hanyelkomy, 2020-10-07:
[image]

hanyelkomy, 2020-10-07:
[image]

veng1, 2020-10-07:
While doing this, why not add the support for the full line of GigaDevice 25QCTxxx?  It appears that the 32M will work.

And it looks like the board could be routed for the wider package.  Then up to 128M would fit.

tannewt, 2020-10-07:
@hanyelkomy The device definitions are here: https://github.com/adafruit/circuitpython/blob/main/supervisor/shared/external_flash/devices.h Please make a PR for it. I don't have that flash so I can't test it.

@veng1 You are welcome to add more definitions as well.

hanyelkomy, 2020-10-15:
@tannewt if u can guide me maybe i can be of help to test this flash ship and get it working.

bd34n, 2020-10-15:
@hanyelkomy, take a look here for great instructions:

https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github

If you're already familiar with git, it'll be a breeze. If not, the instructions are detailed so you shouldn't have any problems.

Basically you'll fork the CircuitPython repo, check out your fork to your local system, create a branch to make your changes on, build and test them, push your changes to your fork, then submit a pull request to get them merged.

-Brian Dean

hanyelkomy, 2020-10-15:
@bd34n Thanks again for your answer, but i meant how can i test this spi flash and get its settings?

bd34n, 2020-10-15:
Unless the chip is really weird, which on first glance it doesn't appear to be, it's a matter of reading the data sheet for the various parameters needed for the definition block. You've got the device ID above in your pasted image - 0x85, 0x60, and 0x16 - that will allow it to be detected and configured based on the other parameters. Tthen verify how many status registers, startup time, speed, etc, and update the values in the definition. That should all be available in the datasheet. If there's another chip from the same mfg in devices.h, it's sometimes helpful to spot check against those since most mfgs are fairly consistent within their own chip lines. Sorry I'm not more help - that's the general process.

dhalbert, 2022-10-16:
Fixed by https://github.com/adafruit/nvm.toml/pull/8. Thanks @skerr.

deshipu, 2023-07-22:
Closing, as this should be fixed.

closed 2023-07-22 by deshipu