adafruit/circuitpython #3528: Auto-reload works only occasionally on file save

todbot, 2020-10-09:
Host PC: Macbook Pro running macOS 10.15.7 
Board: QT Py, with no peripherals  (also happens on Trinket M0)
CircuitPython version: 6-beta2

I've been noticing in CircuitPython 6 that the auto-reload doesn't seem to work much after saving a file.  For instance, in the CP program below (which plays random colors on the built-in Neopixel and prints out the current `time.monotonic()` and a string), I can edit the "hello" string, save the file, and auto-reload does not happen.

```py
import board
import time
import neopixel
from random import randint
pixel = neopixel.NeoPixel(board.NEOPIXEL, 1, brightness=0.2)
while True:
    print(time.monotonic(), "hello")
    pixel.fill( (randint(0,255), randint(0,255), randint(0,255) ))
    time.sleep(0.2)
```
Here's a screencast showing what I'm seeing.  
[image]

(also at https://imgur.com/gallery/zAHgrNg)

In the screencast, there are two terminals: the bottom one shows the output of CP's USB Serial, in the top one I use `vim` to edit `code.py`, save it, and quit the editor.  In the serial terminal, nothing changes, until I Ctrl-C and Ctrl-D to stop & restart execution.

educ8s, 2020-10-09:
I also face a similar problem. I am using a Seeeduino Xiao board. When using the latest CircuitPython 6.0.0 Beta 2 the program does not autoload when saving. Any program, even the blink program does not run on save. I have to disconnect and reconnect the board in order for the program to run. When using CircuitPython 5.3.1 everything works as expected.

tannewt, 2020-10-09:
Anyone tried this on an M0 Express board?

gamblor21, 2020-10-09:
I just tried this on an M0 Basic (and just changed the print as no neopixel) and saw the exact same behavior. Hitting Ctrl-S in Mu did not auto-reload. Ctrl-C & Ctrl-D reloaded it.

tannewt, 2020-10-09:
Ok, I suspect that this is an interaction with the time.sleep and the autoreload timing.

todbot, 2020-10-09:
ahh, yes!  If I create a similar small script, but do not use `time.sleep()`, auto-reload works fine. E.g.
```py
import board
import time
import neopixel
from random import randint
pixel = neopixel.NeoPixel(board.NEOPIXEL, 1, brightness=0.2)
lasttime = time.monotonic()
while True:
    if time.monotonic() - lasttime > 0.2:
        lasttime=time.monotonic()
        print(time.monotonic(), "hello")
        pixel.fill( (randint(0,255), randint(0,255), randint(0,255) ))
```

todbot, 2020-10-09:
Verified on Trinket M0 that #3532 fixes auto-reload.

PR #3532 (2020-10-09): Fix autoreload on SAMD21
The issue was that a time.sleep() would set the RTC wake up
further into the future even if we wanted to tick every ms. Ticking
every ms is used to time the autoreload delay and without it,
autoreload doesn't work.

Fixes #3528
commit 3ccf644d (2020-10-09): Fix autoreload on SAMD21

The issue was that a time.sleep() would set the RTC wake up
further into the future even if we wanted to tick every ms. Ticking
every ms is used to time the autoreload delay and without it,
autoreload doesn't work.

Fixes #3528
closed 2020-10-09 by tannewt