adafruit/circuitpython #3533: Bluefruit Sense nrf52840 board.A6 vbat voltage pin missing

hatchman, 2020-10-10:
running the latest 6.0.0 beta 2, I get an error trying to read the battery voltage internally, as referenced here:
https://learn.adafruit.com/adafruit-feather-sense/pinouts
I should be able to just read pin A6 and multiply it out to convert to a voltage, however.

AttributeError: 'module' object has no attribute 'A6'

when running the following:

```
import board
import analogio

vbatval = analogio.AnalogIn(board.A6)
```

Am I missing something painfully obvious?

jerryneedell, 2020-10-10:
Use board.VOLTAGE_MONITOR instead of board.A6.

hatchman, 2020-10-10:
Thank you Jerry! that was it! looking at the code you have linked, both variables board.VOLTAGE_MONITOR, as well as board.BATTERY give the same results.

closed 2020-10-10 by hatchman