adafruit/circuitpython #3552: MatrixPortal M4 A0 in use error

makermelissa, 2020-10-13:
When attempting to use board.A0 with CircuitPython 6.0 Beta 2 with NeoPixels, I am seeing this error:
```
code.py output:
Traceback (most recent call last):
  File "code.py", line 4, in <module>
ValueError: A0 in use
```

It can be reproduced with this simple bit of code
```
import board
import digitalio

pin = digitalio.DigitalInOut(board.A0)
```

makermelissa, 2020-10-13:
Oh, looks like it requires a soft reboot.

makermelissa, 2020-10-14:
FYI, I also tried this with the absolute newest image and the behavior is still there, so it hasn't been fixed yet.

tannewt, 2020-10-14:
Can you reproduce this on a Metro M4? I don't have a MatrixPortal yet.

makermelissa, 2020-10-14:
I'll give it a try.

makermelissa, 2020-10-14:
It did not have the same issue despite them both using PA02 for A0.

tannewt, 2020-10-14:
Does it work after a soft reboot but not in the first run of code.py?

makermelissa, 2020-10-14:
It works when it first starts up, but the pin doesn't work after a soft reboot like it's not getting reset or something.

makermelissa, 2020-10-14:
It also does the same thing with A2, but A1, A3, and A4 act normal.

tannewt, 2020-10-14:
I have no idea why this would be the case for *just* this board. I'll get one in a few days and see if I can reproduce it. Thanks!

tannewt, 2020-10-19:
Ok, I see this on my MatrixPortal too.

closed 2020-10-22 by jepler

dhalbert, 2021-10-08:
@martingosnell Answerd in #5440.

PR #3574 (2020-10-20): Correct pins to not reset.
They must have the PORT_ prefix otherwise they mask the wrong pins.

Fixes #3552

PRing to 6.0.x and then will merge from 6.0.x into main.
commit eb139c9b (2020-10-20): Correct pins to not reset.

They must have the PORT_ prefix otherwise they mask the wrong pins.

Fixes #3552