adafruit/circuitpython #3553: PID for new boards

Targett363, 2020-10-14:
Hi I'm going through Kattni's How to "Add a New Board to CircuitPython" for my ESP32S2 Module Clip board and I'm pretty much there except for needing a VID&PID, well 2 I guess as you can attach either WROOM or WROVER variants to the board.

The guide suggests asking for help here...

dhalbert, 2020-10-14:
Could you give us a "manufacturer" name and official board names?

Targett363, 2020-10-14:
I am the Manufacturer "Targett" and the Board names are "ESP32S2 Module Clip" Wroom and Wrover.
I'd propose the name targett_module_clop_wroom and targett_module_clip_wrover in my fork.

Targett363, 2020-10-14:
https://www.tindie.com/products/20752/

tannewt, 2020-10-14:
Here is the other PR: https://github.com/pidcodes/pidcodes.github.com/pull/564

Targett363, 2020-10-14:
Scott, tank you so much. Git newbie, I thought I'd done something fatally wrong over there.

closed 2020-10-14 by Targett363