adafruit/circuitpython #3558: (Solved) update correct values for Macronix MX25R1635F 8MiB SPI flash

hanyelkomy, 2020-10-15:
Credits to Brian — Brian Dean bsd@bdmicro.com BDMICRO, LLC https://www.bdmicro.com

According to the datasheet for the MX25R1635F, the JDEC should be 0xc2, 0x28, 0x15. In the devices.h file, the capacity parameter is specified as 0x18 for that chip. So the value is incorrect in devices.h

/ Settings for the Macronix MX25R1635F 8MiB SPI flash.
// Datasheet: https://www.macronix.com/Lists/Datasheet/Attachments/7595/MX25R1635F,%20Wide%20Range,%2016Mb,%20v1.6.pdf
// In low power mode, quad operations can only run at 8 MHz.
#define MX25R1635F  {\
    .total_size = (1 << 21), /* 2 MiB */ \
    .start_up_time_us = 800, \
    .manufacturer_id = 0xc2, \
    .memory_type = 0x28, \
    .capacity = 0x15, \
    .max_clock_speed_mhz = 33, /* 8 mhz for dual/quad */ \
    .quad_enable_bit_mask = 0x80, \
    .has_sector_protection = false, \
    .supports_fast_read = true, \
    .supports_qspi = true, \
    .supports_qspi_writes = true, \
    .write_status_register_split = false, \
    .single_status_byte = true, \

tannewt, 2020-10-15:
Hi! Please make a pull request with the change. We'd love to see it fixed. Thanks!

More info on how to make a pull request here: https://learn.adafruit.com/contribute-to-circuitpython-with-git-and-github

microDev1, 2021-09-06:
Closing... fixed in #3838.

closed 2021-09-06 by microDev1