adafruit/circuitpython #3559: Request VID/PID for DynOSSAT-EDU

ecasadod, 2020-10-15:
Hi,

I'd love to get a couple of USB VID/PID for our project **DynOSSAT-EDU**, the world's first open source nanosatellite educational kit compatible with CircuitPython and Arduino :)

Our CircuitPython fork: https://github.com/BHDynamics/circuitpython (auto build fails because of duplicate VID/PID since it's set to 0)
UF2 Bootloader fork: https://github.com/BHDynamics/uf2-samdx1

The hardware repos are set on private because we'll release later this month/early november, but I can give you access if you want to check them out.

You can read more info about the project here: https://medium.com/bhdynamics/dynossat-edu-newspace-at-your-reach-9c2f88ca52e6

Thank you,

Enrique

tannewt, 2020-10-15:
Hi Enrique! We're trying to minimize giving out Adafruit PIDs. When your source is released, you can get a PID from [pid.codes](http://pid.codes/). They also have testing PIDs that you can use temporarily.

Note that we already have other satellite platforms:
* https://circuitpython.org/board/kicksat-sprite/
* https://circuitpython.org/board/pycubed/ and https://circuitpython.org/board/pycubed_mram/

ecasadod, 2020-10-16:
Hi Scott,

Thanks! We will then release the source and ask for a PID there. The moment we have it, we will submit a PR to add DynOSSAT-EDU to your repos.

About the satellite platforms, noted! Although they are great, they do not fulfill the same role as ours. Ours is a full PocketQube (5x5 cm) with a modular design, PyCubed is a payload board for a CubeSat (10x10cm).

Best regards,

Enrique

tannewt, 2020-10-16:
PIDs acquired from MicroChip: #3564

closed 2020-10-16 by tannewt