adafruit/circuitpython #3572: ESP32S2 I2C hangs with bno08x

jerryneedell, 2020-10-19:
Not sure if this is a library issue or a Circuitpython Issue.
I have been testing out a bno08x IMU
I tried running the examples on both a grandcentral m4 and on an esp32s2 (saola_wrover and unexpectedmaker_feathers2)
both the bno08x_simpletest and the bno08x_more_reports examples run normally on the grandcentral m4
On the esp32s2 (both boards) only the bno08x_simpletest runs
When I try the bno08x_more_reports example, it prints out the word
Acceleration:

but then hangs.

I  am trying to create a simple example of the issue, but I am also finding that invogin this result leads to a loss of access to the REPL and then if i reset the board while the repo is frozen, my Linux computer reboots ....  several things going on here...

I wanted to create this issue to see if anyone can reproduce it and if you think it is a library or CP issue.
To reproduce, 
try to run the bno8x_more_reports.py script on an esp32s2.

ladyada, 2020-11-15:
i want to piggy back on this because i think i2c in *general* is causing issues. ive had no problems with USB until i started trying accessing the I2C peripheral. then getting hangs, disconnects, erased drive etc :(

once code runs, i2c is working fine...it just struggles to get there

on a magtag (built in i2c accel)
```
import time
import board
import digitalio
import busio
import adafruit_lis3dh

i2c = busio.I2C(board.SCL, board.SDA)
lis3dh = adafruit_lis3dh.LIS3DH_I2C(i2c, address=0x19, int1=None)
lis3dh.range = adafruit_lis3dh.RANGE_2_G

while True:
    x, y, z = [
        value / adafruit_lis3dh.STANDARD_GRAVITY for value in lis3dh.acceleration
    ]
    print("x = %0.3f G, y = %0.3f G, z = %0.3f G" % (x, y, z))
    # Small delay to keep things responsive but give time for interrupt processing.
    time.sleep(0.1)
```

gamblor21, 2020-11-15:
I discovered the issue seems to occur once my program ends or aborts with Ctrl-C. The program will execute fine as long as it does not reach the REPL.

I was testing with:
```
import time
import board
import adafruit_bme280

time.sleep(5)

bme280_sensor = adafruit_bme280.Adafruit_BME280_I2C(board.I2C())

while True:
    print(bme280_sensor.temperature)
    print(bme280_sensor.humidity)
    time.sleep(5)
```
I also tested removing the **while** and letting it drop to the REPL. In either case 90%+ of the time I locked up.

When I ran the same code typing it all into the REPL it never hangs, no matter how many times I abort or stop it.

jepler, 2020-11-17:
Here's what I'm testing with:
```
import time

import board
import busio
import supervisor

i = busio.I2C(sda=board.SDA, scl=board.SCL)
i.try_lock()
i.scan()

supervisor.reload()

####################################################################
####################################################################
####################################################################
####################################################################
####################################################################
####################################################################
####################################################################
####################################################################
####################################################################
####################################################################
####################################################################
####################################################################
####################################################################
```
This rapidly reloads itself, but I also repeatedly save it from my text editor.  The trailing "#" may or may not be important -- it's still a bit inconsistent for me, but it seems to happen more often if the file is big enough to span multiple 512-byte disk blocks.

Usually after a handful of writes from my text editor, a crash-or-something occurs.

tannewt, 2020-11-18:
@jepler Want to close this? We can reopen if it crops up again.

closed 2020-12-23 by jepler