adafruit/circuitpython #3576: ESP32S2: Wifi unresponsive after soft reboot and causes hard crash

astrobokonon, 2020-10-20:
I'm still confused by this, but #3484 seems to have caused a regression.  That was my PR, but I don't yet know what went wrong!

Testing out the additions with this simple code

```python
import wifi
import ipaddress

def cheapoHex(inBytes, delim=':'):
    res = ""
    for i, b in enumerate(inBytes):
        # print("byte %d %s (%02x)" % (i, b, b))
        res += "%02x" % b 
        if i < len(inBytes) - 1:
            res += delim

    return res

for network in wifi.radio.start_scanning_networks():
    pstr = "%02d %s %d %s" % (network.channel, cheapoHex(network.bssid), 
                              network.rssi, network.ssid)
    print(pstr)
wifi.radio.stop_scanning_networks()
print("")
print("Done scanning")

print("")
print("Connecting ...")
wifi.radio.connect(<yourssid>, <yourpassword>)
pstr = "Using chan. %02d with %s (%s) at RSSI %d" % (wifi.radio.ap_info.channel, 
                                                     wifi.radio.ap_info.ssid, 
                                                     cheapoHex(wifi.radio.ap_info.bssid), 
                                                     wifi.radio.ap_info.rssi)
print(pstr)
print("Network info:")
print("IP: %s on %s" % (wifi.radio.ipv4_address, cheapoHex(wifi.radio.mac_address)))
print("Gateway: %s" % (wifi.radio.ipv4_gateway))
print("DNS: %s" % (wifi.radio.ipv4_dns))
print("Subnet: %s" % (wifi.radio.ipv4_subnet))
print("")
ipv4 = ipaddress.ip_address("8.8.4.4")
print("Pinging %s ... " % (ipv4), end='')
pval = wifi.radio.ping(ipv4)
if pval is not None:
    print("%.4f ms" % (pval))
print("Done pinging")
```

works once, but after a soft reboot, the scan returns nothing, and the connection attempt hangs indefinitely, and when I control-C it the CIRCUITPY drive disconnects so it seemed like something was causing a hard crash.  To check that I made a build with ```DEBUG=1``` and I can indeed see some sort of crash

```
W (32598) wifi: connected
I (32638) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (35028) esp_netif_handlers: sta ip: 192.168.1.234, mask: 255.255.255.0, gw: 192.168.1.1
W (35028) wifi: got ip
I (84578) wifi:state: run -> init (0)
I (84578) wifi:pm stop, total sleep time: 46400266 us / 51984995 us

I (84578) wifi:new:<10,0>, old:<10,0>, ap:<255,255>, sta:<10,0>, prof:1
W (84578) wifi:hmac tx: stop, discard
W (84588) wifi: disconnected
W (84588) wifi: reason 8 0x08
I (84618) wifi:flush txq
I (84618) wifi:stop sw txq
I (84618) wifi:lmac stop hw txq
I (84618) wifi:Deinit lldesc rx mblock:10

abort() was called at PC 0x400b33e1 on core 0

Backtrace:0x4002f586:0x3ffdbdd0 0x4002fabd:0x3ffdbdf0 0x400374da:0x3ffdbe10 0x400b33e1:0x3ffdbe80 0x4009e6b7:0x3ffdbea0 0x4009e2dc:0x3ffdbec0 0x4009e574:0x3ffdbee0 0x4009e5e4:0x3ffdbf00 0x4009e7f3:0x3ffdbf20 0x4002facc:0x3ffdbf40

ELF file SHA256: edd7b0afe347928f

CPU halted.
```

which translated to

```
circuitpython/ports/esp32s2/build-espressif_saola_1_wrover/esp-idf/../../esp-idf/components/esp_system/panic.c:360
circuitpython/ports/esp32s2/build-espressif_saola_1_wrover/esp-idf/../../esp-idf/components/esp_system/system_api.c:104
circuitpython/ports/esp32s2/build-espressif_saola_1_wrover/esp-idf/../../esp-idf/components/newlib/abort.c:46
circuitpython/ports/esp32s2/common-hal/wifi/__init__.c:140 (discriminator 1)
circuitpython/ports/esp32s2/supervisor/port.c:123
circuitpython/ports/esp32s2/../../main.c:237
circuitpython/ports/esp32s2/../../main.c:443
circuitpython/ports/esp32s2/../../main.c:508
circuitpython/ports/esp32s2/supervisor/port.c:213
circuitpython/ports/esp32s2/build-espressif_saola_1_wrover/esp-idf/../../esp-idf/components/freertos/xtensa/port.c:535
```

jerryneedell, 2020-10-20:
I am seeing the same behavior on both a saola_wrover and umfeathers2
Works on Hard reset, but not after soft reboot.

astrobokonon, 2020-10-20:
@jerryneedell thanks for confirming - I went round and round for a while convinced that I had broken my esp-idf submodule when jumping up and down branches the last few days, or that the problem was really just a bug in there, but no such luck.  Wondering now if the ```Network``` instance I stuff into ```wifi.radio.ap_info``` is the cause and isn't getting free'ed like I assumed it was?  Still not sure what went wrong with this one.

tannewt, 2020-10-20:
Are we resetting sta_mode when we reset the wifi?

astrobokonon, 2020-10-20:
@tannewt I think you've found it!  The sta_mode flag gets set the first time through ```start_station()``` but after a soft reboot it's still True, causing that function to immediately return and not actually set up station mode.  Things get worse very quickly from there.  I had assumed that whole struct would get trashed after a soft reboot so it wouldn't matter, but I guess not!

I tested it quickly just now by commenting out the immediate check/return of ```self->sta_mode``` and things work again after soft reboots.  I'll clean up that flag and test a little more carefully, but I should be able to get a PR in to fix this fairly quickly if someone doesn't beat me to it since this is an annoying one.

jerryneedell, 2020-10-20:
@astrobokonon -- did a quick test on both saola_wrover and umfeathers2 with this change - comment out the auto return.
```
diff --git a/ports/esp32s2/common-hal/wifi/Radio.c b/ports/esp32s2/common-hal/wifi/Radio.c
index 275bae31a..d55416046 100644
--- a/ports/esp32s2/common-hal/wifi/Radio.c
+++ b/ports/esp32s2/common-hal/wifi/Radio.c
@@ -42,9 +42,9 @@
 #define MAC_ADDRESS_LENGTH 6
 
 static void start_station(wifi_radio_obj_t *self) {
-    if (self->sta_mode) {
-        return;
-    }
+//    if (self->sta_mode) {
+//        return;
+//    }
     wifi_mode_t next_mode;
     if (self->ap_mode) {
         next_mode = WIFI_MODE_APSTA;
```

works on both after soft reboot!  Thanks!

PR #3580 (2020-10-22): ESP32S2: Reset sta_mode and ap_mode flags
Should close #3576.

I opted to set both ```ap_mode``` and ```sta_mode``` flags in ```common_hal_wifi_init``` to False just to start from a clean slate every time, since at that point things aren't properly configured for either mode anyways.  I think *technically* setting ```sta_mode``` to True might be legit after the call to ```esp_netif_create_default_wifi_sta``` but it's functionally not useful until after you load in the config so to me it seems debatable.  I at least added a comment to that effect as a reminder for the next brave soul mucking ab [... cut]
closed 2020-10-22 by tannewt