adafruit/circuitpython #3579: Cannot CTRL-C out of sleep on ESP32S2

tannewt, 2020-10-20:


DavePutz, 2020-10-26:
The issue is that we are sitting in port_sleep_until_interrupt() until the sleep duration expires. Looking at port_interrupt_after_ticks(), which sets the duration, I see that portTICK_PERIOD_MS = 10. Looks like that is being configured to "1000 / configTICK_RATE_HZ " in esp-idf/components/freertos/xtensa/include/freertos/portmacro.h. If we just use 1000 in port_interrupt_after_ticks() instead of portTICK_PERIOD_MS then ctrl-c works OK; but I am not sure how that might affect anything else.

tannewt, 2020-10-26:
@DavePutz I think the issue may be that USB doesn't interrupt us because it is handled in a separate task. I think we'll need a freertos primitive to wait on with a timeout that matches the sleep time. The primitive can be used by the USB task to wake us up early.

We'll want to rely on FreeRTOS to wait so that it can automatically light sleep.

DavePutz, 2020-10-28:
@tannewt - I think I understand what you are saying, but doesn't going into a FreeRTOS light sleep defeat the purpose of the loop in mp_hal_delay_ms() that allows us to check for runnable background tasks while sleeping?

tannewt, 2020-10-28:
The delay_ms loop is meant to run background tasks after every interrupt. I'm not sure how that works with FreeRTOS though.

tannewt, 2020-11-17:
Fixed by #3647

tannewt, 2020-12-10:
@mightyohm is having trouble with beta.2 on a feather s2. Could someone retest please? Thanks!

tannewt, 2020-12-10:
Nevermind. It was a realterm issue on Windows.

closed 2020-12-10 by tannewt