adafruit/circuitpython #3581: board.SPI lost after being used by display

cwalther, 2020-10-20:
While playing around with creating and destroying displays to exercise the supervisor heap, I came across some strange behavior: When a `board.SPI()` object has ever been used by a `FourWire` or a `SharpMemoryFramebuffer` that has since been deinitialized, subsequent sessions can never reconstruct it, because its pins are still in use.

Steps to reproduce:

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-rc.0-61-g1a677406b-dirty on 2020-10-21; CWtest with samd51G19
>>> import board, displayio
>>> displayio.FourWire(board.SPI(), command=board.RX, chip_select=board.TX)
<FourWire>
>>> displayio.release_displays()
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Hello World!

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-rc.0-61-g1a677406b-dirty on 2020-10-21; CWtest with samd51G19
>>> import board
>>> board.SPI()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: SCK in use
>>> 
```

or

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-rc.0-61-g1a677406b-dirty on 2020-10-21; CWtest with samd51G19
>>> import board, displayio, sharpdisplay
>>> sharpdisplay.SharpMemoryFramebuffer(board.SPI(), board.RX, 400, 240)
<SharpMemoryFramebuffer>
>>> displayio.release_displays()
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Hello World!

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-rc.0-61-g1a677406b-dirty on 2020-10-21; CWtest with samd51G19
>>> import board
>>> board.SPI()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: SCK in use
>>> 
```

What happens is:

* `common_hal_displayio_fourwire_construct()` and `common_hal_sharpdisplay_framebuffer_construct()` call `common_hal_busio_spi_never_reset(self->bus)`.
* `release_displays()` doesn’t do anything with the SPI bus.
* When the session ends, `cleanup_after_vm()` → `reset_port()` → `reset_all_pins()` does not reset the pins because they were marked as `never_reset`.
* `cleanup_after_vm()` → `reset_board_busses()` sets `spi_singleton = NULL;` so that the next call to `board.SPI()` needs to recreate it, but it can’t because the pins are still in use.

I’m not sure how this is supposed to work, but there seems to be a hole in the logic somewhere. I suppose someone should call `common_hal_busio_spi_deinit()` at some point so that the pins are made resettable again? Maybe `reset_board_busses()` before it effectively leaks the resources used by the `spi_singleton`, but then `reset_port()` and `reset_board_busses()` come in the wrong order?

Some of the observations made by @hatchman in https://github.com/adafruit/circuitpython/issues/3508#issuecomment-706211348 seem related (but probably unrelated to the actual issue of #3508, so I’m opening a new one).

jepler, 2020-10-21:
I have seen similar behavior with the Sharp Memory Display.  I'm not sure what the fix is, but I hope you will consider making a pull request to fix this.  @tannewt may have a suggestion.

cwalther, 2020-10-21:
Finding a good fix will require more analysis on my part, so I was hoping someone of you maintainers would have an idea more quickly. But I can give it a try. No promises though. It’s not urgent as far as I am concerned.

hatchman, 2020-10-21:
@cwalther yes! this seems to be the case, thank you for a more clear description of what is going on....

unfortunate to hear its not that urgent, I wish I had more expertise to help with the sharp displays. Adafruit seems to not have anyone to test these. I got myself invested into a project where I have a keyboard button array, a sharp memory display and an nrf based board hoping to use circuit python. but in my case, its several limited hitting issues with memory not being freed on reboots, where the frame buffer is just sitting on extra memory, and in cases where speed is needed, the board cannot be reset even properly after a reboot and call back to the display. its made everything I spent developing into a brick which is sad because the information of these issues seems to be something nouns ever tested. in my opinion the issues here make the display unusable and unstable in projects and really has been a cloud on my circuit python experience, I am sure many others, and as such the use of sharp memory displays doesnt seem fair to be claimed as stable to less practiced coders such as myself.

cwalther, 2020-10-21:
Regarding the memory leaks, could you try a recent [“absolute newest”](https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/feather_bluefruit_sense/) build (newer than 6.0.0-rc.0)? #3482 should help with these, although I’m not sure if the first commit of #3498 is also needed. (@jepler, maybe you could rebase that and leave out the second commit that is obsoleted by #3542, to get a new build to test?)

hatchman, 2020-10-21:
in regard to memory leaks, if anything my experience has got worse. I just flashed adafruit-circuitpython-feather_bluefruit_sense-en_US-20201020-1a67740.uf2 and here is my experience:

prior to this build, up to 6.0.0 beta 2 when I save my files, it would reboot my code, and have about 72,000 bytes avail after freshboot init sequence, and displays this on screen. 

anytime a software reboot is launched, saving a file, hitting the reset button once, or calling a soft reboot with code, it would reboot with the same sequence and yield 53,000 on average. the only way around is to double press  reset to get into bootloader mode then once more for a 'fresh' boot. 

now, my code (same stuff used in prior builds) launches with 66,000 bytes after init, but exhibits odd behavior. files that previously loaded aren't loading which is likely due to the 6,000 bytes missing, but most of my programs dont run.for reference, I am running a modified Adafruit clue library adding my hardware callouts for keypresses using adafruits matrix keypad library, an encoder using rotaryio, expanded the onboard neopixel by adding less off the output pin on the onboard neopixel to run a strip (powered and buffered appropriately with resistors and caps) and changing the display driver out for the sharp frame buffered display, then compiling to .mpy. then I have a main menu using clue simpletextdisplay. that reads and executes other python files onboard my storage.

what is MORE strange, is that if I then issue a single button reboot, or save a file to trigger a reboot, my code doesnt make it to even load the main menu, which should load with 66,000 bytes free which is the case on a fresh boot.

hatchman, 2020-10-21:
@cwalther I am happy to help test any of this and learn more as long as there is attention, but the recent builds have definitely got worse for my hardware combination. im close to switching back to the Arduino side even through it slows my ability to develop about 10x, at least id have room to program more than display 5 words in a tiny byte font and blink a light :/

cwalther, 2020-10-21:
As far as my contributions are concerned, I am grateful for any testing with Sharp Memory Displays, because I don’t have one. I can test memory management by pretending there was a display, but I can’t check if anything sensible shows up on the display. (I’m also unfamiliar with nRF52 hardware. Does the reset button really do a soft reboot there? That’s not what I would expect from a reset button.)

Do you think you could distill your testing procedure into a series of simple steps that someone else could follow, and show us the output from the different builds you’ve been testing with, similarly to what I did in the first post?

hatchman, 2020-10-21:
yeah I totally get that! Adafruit should be shipping this hardware out to hard working folks like you! especially since not many seems to be championing the development of a product Adafruit sells thats still newer.

I can certainly work to optimize my test methods and describe them better, but I would need a bit of guidance. I understand that I have a unique combination of hardware and weird software modifications like to the .mpy libraries. I just dont know how fast I can do something thats more straight forward. I will try to write something quickly using the graphic libraries I suppose.

as far as reset button functionality, I was under the impression that the reset button isnt doing a full reboot ever? for example circuit python still holds on to the display to show serial commands if not released manually on a reset button press.

in code, I call
import supervisor
supervisor.reload()
to soft reboot. seems to be functionally similar to a single reset button press but maybe not under the hood.

hatchman, 2020-10-21:
so best I can do quickly, is use this standalone piece of graphics code I wrote. it doesnt max out memory, but does point to about 10,000 bytes being used up between soft reboots. attached is my code, and the modified clue library with the sharp memory display swap, plus my additional sensors:

code.py:
`import gc
import time
from adafruit_clue import clue
import displayio
import supervisor
from vectorio import Circle, Polygon, VectorShape
print(gc.mem_free(),'bytes free --> import success')

clue.pixels.fill((0,0,0,0))

a = 1/30
live = displayio.Group(max_size=2, x=200, y=120, scale=1)
cblack = displayio.Palette(2)
cblack.make_transparent(0)
cblack[1] = 0x000000
cwhite = displayio.Palette(2)
cwhite.make_transparent(0)
cwhite[1] = 0xFFFFFF
print(gc.mem_free(),'bytes free --> initialize display variables')
clue.display.show(live)
oc = Circle(radius=78)
ic = Circle(radius=39)
ahm = Polygon(points=[(-22,5), (-17,-5),(-18,-14),(-14,-11),(-6,-27),
					 (2,-11), (10,-27), (13,-21),   (5,-5),  (10,5),
					 (18,-11), (21,-5),  (13,11),  (14,20), (10,17),
					 (5,27),    (2,21),   (7,11),    (2,1), (-3,11),
					 (-2,20),  (-6,17),  (-11,27),(-14,21), (-9,11),
					 (-14,1),(-19,11)])
ahs = Polygon(points=[(-11,-5),(-6,-15),(-1,-5),(-6,5)])
cblack = displayio.Palette(2)
cblack.make_transparent(0)
cblack[1] = 0x000000
cwhite = displayio.Palette(2)
cwhite.make_transparent(0)
cwhite[1] = 0xFFFFFF

ahoutline = VectorShape(shape=oc, x=0, y=0, pixel_shader=cwhite)
ahcircle = VectorShape(shape=ic, x=0, y=0, pixel_shader=cblack)
ahmain = VectorShape(shape=ahm, x=0, y=0, pixel_shader=cwhite)
ahsub = VectorShape(shape=ahs, x=0, y=0, pixel_shader=cblack)

AHlogo = displayio.Group(max_size=4, x=0, y=0, scale=1)
AHlogo.append(ahoutline)
AHlogo.append(ahcircle)
AHlogo.append(ahmain)
AHlogo.append(ahsub)
live.append(AHlogo)
time.sleep(a)
AHlogo.scale = 2
time.sleep(a)
AHlogo.scale = 3
time.sleep(a)
print(gc.mem_free(),'bytes free --> low res logo')

live.pop()
time.sleep(a)
oc2 = Circle(radius=234)
ic2 = Circle(radius=118)
ahm2 = Polygon(points=[(-66,15), (-51,-15),(-54,-42),(-42,-33),(-18,-81),
					 (6,-33), (30,-81), (39,-63),   (15,-15),  (30,15),
					 (54,-33), (63,-15),  (39,33),  (42,60), (30,51),
					 (15,81),    (6,63),   (21,33),    (6,3), (-9,33),
					 (-6,60),  (-18,51),  (-33,81),(-42,63), (-27,33),
					 (-42,3),(-57,33)])
ahs2 = Polygon(points=[(-33,-15),(-18,-45),(-3,-15),(-18,15)])

ahoutline2 = VectorShape(shape=oc2, x=0, y=0, pixel_shader=cwhite)
[... 1232 lines cut ...]
   from adafruit_clue import clue
"""
`

to elaborate, I just reran this code on a fresh boot, and a full power cycle, full power cycle I got 
75680 bytes free --> low res logo
74288 bytes free --> hi res logo
^ quickest commands I see when opening the serial

EVERY soft reboot or reset button press, or code save reload, the same point of code yields:
63584 bytes free --> low res logo
62192 bytes free --> hi res logo

thats a lot to loose, and idk how to fix it. 

furthermore, If I try to do anything to deinitialize the display at any point using the function I added in adafruitclue.py, I get SPI errors, and can no longer use the display until a full power cycle. heres that specific function:

def deinitdisp(self):
        displayio.release_displays()
        del self._framebuffer
        digitalio.board.A5.deinit()

even tried deleting the frame buffer object which I know doesnt actually do what I think it does in effort to release some pins. but I know its more going on with issues mentioned earlier.

let me know how else I can be of value. Thank you for taking the time!

tannewt, 2020-10-21:
@cwalther I think the bug is setting the singleton back to NULL while it's being used by the display still. It should be preserved by `release_displays()` in case it's referenced before it. I suspect https://github.com/adafruit/circuitpython/pull/3378 is related.

@hatchman `supervisor.reload()` is not the same as pressing the reset button. "reloads" don't change display state because the display will be used by CircuitPython after the VM completes.

tannewt, 2020-10-21:
Actually, why isn't this working: https://github.com/adafruit/circuitpython/blob/main/shared-module/board/__init__.c#L162

cwalther, 2020-10-21:
@hatchman:
I was thinking of something a little shorter. :) Could you try to strip down this clue library to just the display-related parts needed for this application? I need to remove at least these imports to get past the import section on my board: neopixel, adafruit_apds9960.apds9960, adafruit_bmp280, adafruit_lis3mdl, adafruit_lsm6ds.lsm6ds33, adafruit_sht31d, audiobusio, audiopwmio, adafruit_matrixkeypad.

Regarding the original issue with the inability to reinitialize board.SPI, the workaround is to call `board.SPI().deinit()` between `release_displays()` and the soft reboot.

@tannewt: I’m not sure I follow. In my case, there is no display using it anymore at the time of the soft reboot.

dhalbert, 2020-10-21:
If you want to do a hard reset, you can use `microcontroller.reset()`.

jepler, 2020-10-21:
Having a decreased amount of memory the second time you use a sharp memory display is hopefully going to be addressed by #3498, which is an in-progress pull request.

Assuming that what you are seeing is that the amount of available memory decreases from the first to second run, but stays broadly similar in second and following runs, I believe that is what you are seeing.

cwalther, 2020-10-21:
> I think the bug is setting the singleton back to NULL while it's being used by the display still. It should be preserved by `release_displays()` in case it's referenced before it.

I’m not sure about that. The object is `spi_obj` and still exists, so any Python references should remain valid, `spi_singleton` that is set to NULL is just a pointer to it and can be identically restored by calling `board.SPI()` again (at least until the end of the session, due to the bug).

hatchman, 2020-10-21:
@tannewt @dhalbert yes, thank you for the clarifications on all this. I did change my code to use the microcontroller function, which is more what I was looking for. 

@cwalther yeah sorry about that, that was just something quick, ill prune the file and add it here to test with the Adafruit sense specific stuff removed and my other hardware. the quick fix of calling board.SPI().deinit() after releasing displays and resetting is just the trick for what I need. now ill see if I can easily reinit the display in the same boot, and id be real happy.

@jepler I will try to get something together for @cwalther to test, and reproduce this more regularly, my approach is now changing, but the issue will persist in other use cases.

to each of you: has anyone told you you are a beautiful human lately? BIG THANKS!

hatchman, 2020-10-21:
@cwalther try this for 
code.py: displays my logo at a few sizes. prints during operation to serial. press the onboard button to issue a supervisor reload.
```
import gc
import time
from adafruit_clue import clue
import displayio
import supervisor
from vectorio import Circle, Polygon, VectorShape
print(gc.mem_free(),'bytes free --> import success')

clue.pixels.fill((0,0,0,0))

a = 1/30
live = displayio.Group(max_size=2, x=200, y=120, scale=1)
cblack = displayio.Palette(2)
cblack.make_transparent(0)
cblack[1] = 0x000000
cwhite = displayio.Palette(2)
cwhite.make_transparent(0)
cwhite[1] = 0xFFFFFF
print(gc.mem_free(),'bytes free --> initialize display variables')
clue.display.show(live)
oc = Circle(radius=78)
ic = Circle(radius=39)
ahm = Polygon(points=[(-22,5), (-17,-5),(-18,-14),(-14,-11),(-6,-27),
					 (2,-11), (10,-27), (13,-21),   (5,-5),  (10,5),
					 (18,-11), (21,-5),  (13,11),  (14,20), (10,17),
					 (5,27),    (2,21),   (7,11),    (2,1), (-3,11),
					 (-2,20),  (-6,17),  (-11,27),(-14,21), (-9,11),
					 (-14,1),(-19,11)])
ahs = Polygon(points=[(-11,-5),(-6,-15),(-1,-5),(-6,5)])
cblack = displayio.Palette(2)
cblack.make_transparent(0)
cblack[1] = 0x000000
cwhite = displayio.Palette(2)
cwhite.make_transparent(0)
cwhite[1] = 0xFFFFFF

ahoutline = VectorShape(shape=oc, x=0, y=0, pixel_shader=cwhite)
ahcircle = VectorShape(shape=ic, x=0, y=0, pixel_shader=cblack)
ahmain = VectorShape(shape=ahm, x=0, y=0, pixel_shader=cwhite)
ahsub = VectorShape(shape=ahs, x=0, y=0, pixel_shader=cblack)

AHlogo = displayio.Group(max_size=4, x=0, y=0, scale=1)
AHlogo.append(ahoutline)
AHlogo.append(ahcircle)
AHlogo.append(ahmain)
AHlogo.append(ahsub)
live.append(AHlogo)
time.sleep(a)
AHlogo.scale = 2
time.sleep(a)
AHlogo.scale = 3
time.sleep(a)
print(gc.mem_free(),'bytes free --> low res logo')

live.pop()
time.sleep(a)
oc2 = Circle(radius=234)
ic2 = Circle(radius=118)
ahm2 = Polygon(points=[(-66,15), (-51,-15),(-54,-42),(-42,-33),(-18,-81),
					 (6,-33), (30,-81), (39,-63),   (15,-15),  (30,15),
					 (54,-33), (63,-15),  (39,33),  (42,60), (30,51),
					 (15,81),    (6,63),   (21,33),    (6,3), (-9,33),
					 (-6,60),  (-18,51),  (-33,81),(-42,63), (-27,33),
					 (-42,3),(-57,33)])
ahs2 = Polygon(points=[(-33,-15),(-18,-45),(-3,-15),(-18,15)])

ahoutline2 = VectorShape(shape=oc2, x=0, y=0, pixel_shader=cwhite)
[... 159 lines cut ...]
        title=None,
        title_color=(255, 255, 255),
        title_scale=1,
        text_scale=1,
        font=None,
        colors=None,
        add_height=0,
    ):
        return _ClueSimpleTextDisplay(
            title=title,
            title_color=title_color,
            title_scale=title_scale,
            text_scale=text_scale,
            font=font,
            colors=colors,
            add_height=add_height
        )

clue = Clue()

```
in my experience, the first time the code loads,  I get 

129216 bytes free --> low res logo on a fresh power cycle consistently, and
115952 bytes free --> low res logo consistently every time that soft reboot is shown in the serial.

tannewt, 2020-10-22:
@cwalther Huh, weird. Maybe this is a case where we mark the SPI as "never reset" but when we deinit the display we don't mark it as "ok to reset". Usually this is done by deinit but that's not quite what we want...

cwalther, 2020-10-22:
@hatchman: I have now been able to run your code. The decrease in free memory between a power cycle and subsequent soft reboots is indeed fixed by #3498 for me, which should be merged soon (or you can download a build [here under _Artifacts_](https://github.com/adafruit/circuitpython/pull/3498/checks?check_run_id=1294096916)).

However, I do not see a decrease in free memory between 6.0.0-beta.2 and 1a67740. After a power cycle, they match exactly, and after soft reboots I get 256 bytes more in 1a67740.

cwalther, 2020-10-24:
Addendum to the original issue: The only reason this does not happen with I2C displays is that `reset_board_busses()` neglects to check `displays[i].bus_base.type` to see whether that display slot is still in use and then compares stale data and wrongly concludes that the I2C bus is still in use by the display that has actually been deinited already.

So a proper solution will apply to both SPI and I2C. (Working on it.)

hatchman, 2020-10-25:
@cwalther Sorry it has taken me so long to test and get back, 

I tested the build with the #3498 fix, and seems to eliminated the issue when I run the code I had sent you.

I do have some concerns however that you have not seen the same memory losses I saw on 6.0.0 beta 2 and previous. the only difference here are some libraries which shouldn't have held onto anything, but perhaps some of the I2C devices may be the culprit. I am reworking my code, and will report back if what I am experiencing comes back when I start working with i2c again, but for now, I assume everything is fixed and I am just a janky coder ;) thank all for the help and dedication to the project!

deshipu, 2020-10-25:
Importing libraries will always use up some unrecoverable memory, because strings such as variable names and string constants are getting interned globally.

tannewt, 2020-10-26:
> Importing libraries will always use up some unrecoverable memory, because strings such as variable names and string constants are getting interned globally.

What do you mean by unrecoverable? Interned strings should be reset on `reload` because the string pool lives on the heap.

dhalbert, 2020-10-29:
@cwalther Do you consider that this bug is now fixed by #3498? Thanks.

cwalther, 2020-10-29:
Yes, but 6.0.x hasn’t been merged into main yet, I assume that’s why it hasn’t been automatically closed yet.

cwalther, 2020-10-29:
Er, wait. It’s not fixed by #3498 but by #3603. Possible confusion?

dhalbert, 2020-10-29:
> Er, wait. It’s not fixed by #3498 but by #3603. Possible confusion?

Oops, yes, sloppy reading, sorry.

PR #3603 (2020-10-27): Fix lost board.SPI and board.I2C
The **first commit** brings `board.I2C` into the same state as `board.SPI` by fixing something that was clearly wrong, but by itself caused no severe problems. It thereby _introduces_ the problem described in #3581 for `board.SPI` also for `board.I2C`. See commit message for details.

The **second commit** fixes #3581 for both SPI and I2C.

I found no reason in the code or in the Git history for `reset_port()` to come before `reset_board_busses()` in `cleanup_after_vm()`, so I reversed them.

For symmetry, since setting `spi_singleton` to non-NULL occurs togeth [... cut]
commit f4f80e07 (2020-10-27): Fix lost board.SPI and board.I2C after being used by display.

Fixes #3581.

Pins were marked as never_reset by common_hal_displayio_fourwire_construct() and common_hal_sharpdisplay_framebuffer_construct(), but these marks were never removed, so at the end of a session after displayio.release_displays(), {spi|i2c}_singleton would be set to NULL but the pins would not be reset. In the next session, board.SPI() and board.I2C() were unable to reconstruct the object because the pins were still in use.

For symmetry with creation of the singleton, add deinitialization before setting it to NULL in r [... cut]
closed 2020-10-31 by dhalbert