adafruit/circuitpython #3594: Accessing an element in a ulab array with a float causes a hard fault

kevinjwalters, 2020-10-22:
On a CLUE using 6.0.0-rc.0, haven't tried other versions:

```
Adafruit CircuitPython 6.0.0-rc.0 on 2020-10-20; Adafruit CLUE nRF52840 Express with nRF52840
>>>
>>>
>>> import ulab
>>> import array
>>> li = [1.0] * 10
>>> ar = array('f', li)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 'module' object is not callable
>>> ar = array.array('f', li)
>>> ular = ulab.ones(10)
>>>
>>> li[0.0]  # good
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: list indices must be integers, not float
>>> ar[0.0]  # good
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: array indices must be integers, not float
>>> ular[0.0]
```

USB serial disconnected, firing up a new terminal quickly shows:

```
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode: something unanticipated happened.
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
                                 Press any key to enter the REPL. Use CTRL-D to reload.
```

Maybe this issue may end up with a fix via https://github.com/v923z/micropython-ulab

tannewt, 2020-10-23:
@v923z Any ideas?

v923z, 2020-10-23:
On the `unix` port of `micropython`, I get an `IndexError`:

```python
import ulab
>>> a = ulab.ones(10)
>>> a[1]
1.0
>>> a[1.0]
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
IndexError: too many indices
```
and in fact, this error message comes from https://github.com/v923z/micropython-ulab/blob/5c75f08118dc871bd3b6c143db6fbbd5ce4d1651/code/ndarray.c#L800. This function (`ndarray_get_slice()`) is not entirely correct, because it assumes that, if the index is not an integer or list, then it is a tuple. I wonder, whether that might be the root of the problem. I can patch it up in the next couple of minutes, though, I don't see why it works in `micropython`.

v923z, 2020-10-23:
@jepler, @tannewt, I think https://github.com/v923z/micropython-ulab/releases/tag/0.54.4 should fix the issue. 

@kevinjwalters thanks for bringing the problem to the fore!

v923z, 2020-10-24:
@jepler @tannewt After @kevinjwalters pointed out an incompatibility with CPython, I have updated the code, and removed the 0.54.4 release. You should now work from https://github.com/v923z/micropython-ulab/releases/tag/0.54.5.

jepler, 2020-10-30:
Oddly this PR has not been closed even though its linked issue was merged.

closed 2020-10-30 by jepler

dhalbert, 2020-10-30:
>Oddly this PR has not been closed even though its linked issue was merged.

@cwalther noted in another issue that also was not closed automatically that this was probably because the PR was against `6.0.x`, not against `main`.

PR #3602 (2020-10-26): ulab: Incorporate bugfixes (update to tag 0.54.5, was 0.54.0)
In particular, this closes #3594.