adafruit/circuitpython #3597: adafruit_requests.get() results in OSError:0 from socketpool.getaddrinfo()

askpatrickw, 2020-10-23:
In a simple `adafruit_requests.get()` I am consistently getting a failure.
It seems to be failing in SocketPool not requests so I'm opening this in CP.

I also tried calling `print(pool.getaddrinfo('wifitest.adafruit.com', 80))` in my code.py right before the get and it worked 1 time `[(0, 0, 0, '', ('104.236.193.178', 80))]`, but never again... I consistently get the same ESError: 0

I don't seem to be out of memory nor am I getting a memory error as in #3562.

### Tested

- My own build off of adafruit/main as of 2020-10-23 with latest requests (py or mpy) on Kaluga and FeatherS2_PreRelease
- adafruit-circuitpython-unexpectedmaker_feathers2_prerelease-en_US-20201023-2581d09.bin with latest requests (py)

### Failure
```cli
[tio 00:50:05] Connected
ip: 192.168.1.35
hostname: espressif
Traceback (most recent call last):
  File "code.py", line 27, in <module>
  File "adafruit_requests.py", line 555, in get
  File "adafruit_requests.py", line 535, in request
  File "adafruit_requests.py", line 415, in _get_socket
OSError: 0
```

### Test Code
```Python
import socketpool
import ssl
import wifi
import adafruit_requests

# Get wifi details and more from a secrets.py file
try:
    from secrets import secrets
except ImportError:
    print("WiFi secrets are kept in secrets.py, please add them there!")
    raise

print("Configuring Network")
for network in wifi.radio.start_scanning_networks():
    if network.ssid == secrets['ssid']:
        wifi.radio.connect(secrets['ssid'], secrets['wifi_password'])
wifi.radio.stop_scanning_networks()
print(f"ip: {wifi.radio.ipv4_address}")
print(f"hostname: {wifi.radio.hostname}")

radio = wifi.radio
pool = socketpool.SocketPool(radio)
requests = adafruit_requests.Session(pool, ssl.create_default_context())

response = requests.get("http://wifitest.adafruit.com/testwifi/index.html")
print(response.status_code)
print(response.text)
```

askpatrickw, 2020-10-23:
Because the RADIO in the network setup is not the same in the request... therefore... No sockets.

askpatrickw, 2020-10-23:
I am re-opening this because I have a consistent repo of a simple case

anecdata, 2020-10-23:
Works for me on Saola-R. If I read the debug log right, the station is disconnecting (8==disassoc) prior to the abort, but I see no reason why it should.

askpatrickw, 2020-10-23:
I'll try one more time with builds off the S3 bucket.

askpatrickw, 2020-10-23:
@anecdata are you seeing https://github.com/adafruit/circuitpython/pull/3580 ?

**Kaluga** worked better with  adafruit-circuitpython-espressif_kaluga_1-en_US-20201023-c676385.bin
I only saw 3 failures out of 20 attempts, and only after a Reset, not a CTRL-D. I wasn't pounding away I waited 5-10 seconds before any reload\reset.

**UMFeatherS2_PreRelease**  adafruit-circuitpython-unexpectedmaker_feathers2_prerelease-en_US-20201023-c676385.bin
Initially, It failed more frequently >50% of the time and not any noticeable improvement between a reload and a reset. But after a few minutes it improve and worked as well as the Kaluga did.
Also, I had previously tried the build just before ^^ that one (2581d09) and it failed all the time as before.

Verified in my git log that I'm at the same commit level (?) as main c67638520 so unclear why local build would behave differently.

Check IDF ... there were updates and there's a wifi related update in updated IDF submodules. ⬅️ 🤦‍♂️
Built and flashed FeatherS2. Failed on initial reset after flash, on soft reboot (I didn't do that one) and on reload 2 times.
Reset. It worked. Reload it worked. 10 Reloads in a row worked.
Reset. It worked.

OK.. I still thing there is something here, but It is also WAY improved. I think there are other open issues tracking "weird things", so I'm calling this my bad build environment and closing.

closed 2020-10-23 by askpatrickw

anecdata, 2020-10-23:
@askpatrickw I haven't spent much time testing #3580. I'm mostly still using rc.0, seeing some "weird things" on certain boards, but need to run them down more.