adafruit/circuitpython #3598: Inconsistent behavior of supervisor.reload() in except block

catequalsgood, 2020-10-23:
I came across this odd behavior but for all I know this might be completely working as intended. It does seem a little unintuitive though so I will post regardless.

Using the following code (on a Feather Sense running CircuitPython 6.0.0-rc.0):
```
import supervisor
import time

time.sleep(1)
print('start')
try:
    try:
        raise TypeError()
    except:
        print('inner-except-1')
        #supervisor.reload()
        raise
        print('inner-except-2')
except:
    print('outer-except-1')
    supervisor.reload()
    print('outer-except-2')
print('end')
```

The output is:
```
start
inner-except-1
outer-except-1
soft reboot
```

Which is what I would expect from reading the documentation. There it says "supervisor.reload() - Reload the main Python code and run it (equivalent to hitting Ctrl-D at the REPL)". As soon as `supervisor.reload()` is called the reload happens.

But when the call of `supervisor.reload()` happens in the inner except block like so:
```
import supervisor
import time

time.sleep(1)
print('start')
try:
    try:
        raise TypeError()
    except:
        print('inner-except-1')
        supervisor.reload()
        #raise
        print('inner-except-2')
except:
    print('outer-except-1')
    #supervisor.reload()
    print('outer-except-2')
print('end')
```

The output becomes:
```
start
inner-except-1
outer-except-1
outer-except-2
end
soft reboot
```

Calling `supervisor.reload()` seems to function like `raise` and the reload happens only once the end of code is reached.
Is this intended or am I missing something obvious that would explain this behavior?

deshipu, 2020-10-23:
I think this is because the reload is internally implemented as an exception.

deshipu, 2020-10-23:
Note that you should never have a "bare" except like that in your production code, always catch the exception you actually except, or ``Exception`` if you want them all. This is because a lot of internals, like ``KeyboardInterrupt``, ``MemoryError``, ``SystemExit`` are implemented as exceptions, but you don't want to be catching them.

catequalsgood, 2020-10-23:
> Note that you should never have a "bare" except like that in your production code

Obviously not, I just wanted to make the example as simple as possible.

> I think this is because the reload is internally implemented as an exception.

I assumed something like that was going on. I guess the lesson is to be mindful of that and think about how that exception is going to be handled. 
One situation in which `supervisor.reload()` seems to behave differently than a regular exception is when `raise` is called immediately after it in the inner except block. Doing this actually reloads immediately but unfortunately doing the same in the outer except block sends the board into safe mode. Therefore simply always having `raise` after `supervisor.reload()` is not an option.

tannewt, 2020-10-23:
@catequalsgood Could you post the version that safe modes? We should make sure it doesn't.

I think we should add to the documentation of `supervisor.reload()` that it raises `ReloadException`. (It used to be KeyboardInterrupt but we split it out.)

In general, I'd suggest using explicit excepts even in examples because it's easier to understand and a better habit.

catequalsgood, 2020-10-23:
> In general, I'd suggest using explicit excepts even in examples because it's easier to understand and a better habit.

Totally makes sense, will do in the future.

As for the code that triggers safe mode: Even a simple:
```
import supervisor

supervisor.reload()
raise
```

results in:
```
Auto-reload is off.
Running in safe mode! Not running saved code.

You are in safe mode: something unanticipated happened.
CircuitPython core code crashed hard. Whoops!
MicroPython NLR jump failed. Likely memory corruption.
Press any key to enter the REPL. Use CTRL-D to reload.
```

This seems to happen as long as the code is not within a try block. So I do not know why someone would run into this. Something else I noticed was that the following code is actually fine. 'test' is not printed and the code reloads infinitely.
```
import supervisor

supervisor.reload()
print('test')
raise
```

But when inserting something else, the board goes into safe mode again. For example using this:
```
import supervisor

supervisor.reload()
a = 0
raise
```

cwalther, 2020-10-24:
I remember a similar bug from a while ago where `supervisor.reload()` would not take effect immediately, but only once further execution hit certain bytecode instructions, like only on function calls or only on `if`s or something like that. But that was fixed at some point. Maybe it wasn’t totally fixed or has come back.