adafruit/circuitpython #3620: RuntimeError: SDA or SCL needs a pull up

gabetav, 2020-10-29:
Hello,

So I'm using a trinket m0 and a MLX90614 with a breakout.
SDA from sensor connected to SDA trinket m0
SCL from sensor connected to SCL trinket m0
GND from sensor connected to GND trinket m0
VIN from sensor connected to 3v trinket m0

The sensor works sometimes ,but then after a couple minutes it gives this error that SDA or SCL needs a pull up. what is that?

deshipu, 2020-10-29:
That error happens when your sensor is not connected, so I suspect loose wires or otherwise bad connection.

gabetav, 2020-10-29:
> That error happens when your sensor is not connected, so I suspect loose wires or otherwise bad connection.

I soldered to it, should I resolder?

ladyada, 2020-10-29:
@gabetav this sort of question is best posted in the adafruit support forums, as this is not a bug report in core circuitpython

tannewt, 2020-10-30:
@gabetav Please post here: https://forums.adafruit.com/viewforum.php?f=60 and include a picture of your setup.

closed 2020-10-30 by tannewt