adafruit/circuitpython #3626: Possible new lightsleep causing issues with delays

mytechnotalent, 2020-10-31:
The new lightsleep feature might be causing issues with delays of a variety of types.  Today was helping to debug a user's issue when he presented the following.
```
USER at 3:59 AM
I've noticed that using play_tone() with the Circuit Playground Express (see example below), CircuitPython 6.0.0_rc.0 adds a significant pause after each note compared to 5.3.1. Has anyone else experienced this?

from adafruit_circuitplayground.express import cpx
scale = [587, 659, 740, 880, 987, 1174]
while True:
  for note in scale:
    cpx.play_tone(note, 0.25)
```
It is possible `time.sleep()` which shuts down some things might be taking longer to come back and causing an issue with user experience.

robertgallup, 2020-10-31:
The following code has similar results even not using the internal timing for notes:

```
import time
from adafruit_circuitplayground.express import cpx
scale = [587, 659, 740, 880, 987, 1174]

while True:
    for note in scale:
        cpx.start_tone(note)
        start_time = time.monotonic()
        while time.monotonic() - start_time < .2:
            pass
        cpx.stop_tone()
```

dhalbert, 2020-10-31:
This seems to have to do with audio playing. The basic sleep delays are still accurate in 5.3.1 or 6.0.0-rc.2. But the audio play intervals on the SAMD21 are wildly off, and need further investigation.

Test program:
```python
from adafruit_circuitplayground import cp
import time

for test in ("sleep", "tone"):
    for length in (0.05, 0.1, 0.25, 0.5, 1):
        start = time.monotonic()
        if test == "sleep":
            time.sleep(length)
        else:
            cp.play_tone(440, length)
        print(test, length, time.monotonic() - start)
    print()
```
Circuit Playground Express 5.3.1 output. Note the too-long tone lengths. The sleep lengths are fine. The output is pretty consistent over multiple runs.
```
sleep 0.05 0.0500488
sleep 0.1 0.100098
sleep 0.25 0.25
sleep 0.5 0.5
sleep 1 1.0

tone 0.05 0.198975
tone 0.1 0.261963
tone 0.25 0.437012
tone 0.5 0.637939
tone 1 1.13599
```
Circuit Playground Express 6.0.0-rc.0. Short tones are even longer. 
```
sleep 0.05 0.0500031
sleep 0.1 0.0999985
sleep 0.25 0.25
sleep 0.5 0.500999
sleep 1 1.0

tone 0.05 0.362999
tone 0.1 0.444
tone 0.25 0.639999
tone 0.5 0.780998
tone 1 1.288
```
Circuit Playground Bluefruit 5.3.1. Tone lengths are much more accurate. (Note that audio is generated in a completely different way on this board.):
```
sleep 0.05 0.0499954
sleep 0.1 0.0999985
sleep 0.25 0.25
sleep 0.5 0.5
sleep 1 1.0

tone 0.05 0.0600052
tone 0.1 0.110001
tone 0.25 0.259995
tone 0.5 0.510002
tone 1 1.01
```
Circuit Playground Bluefruit 6.0.0-rc.0. Still fine:
```
sleep 0.05 0.0500031
sleep 0.1 0.100006
sleep 0.25 0.25
sleep 0.5 0.5
sleep 1 1.0

tone 0.05 0.0599976
tone 0.1 0.112
tone 0.25 0.261002
tone 0.5 0.512009
tone 1 1.01199
```

robertgallup, 2020-11-02:
Great to have this confirmed. FWIW, it may not be obvious from the numbers, but this issue has significant audio impact on sound effects using `play_tone()`, `start_tone()` and 'stop_tone()` on the CPX.

tannewt, 2021-08-11:
This is not a core bug. The reason for the extra delay is the time it takes to go to and from the quiescent_value. These ramps are done to prevent clicking. The CP library could reuse the AudioOut to reduce the timing impact of these ramps to and from 0. Furthermore, #1623 would help stop samples at the quiescent_value and not require the ramp on stop.

[image]

I just tested with a 7.0.0 and the numbers match 5.3.0 much better than the 6.x numbers. This may be due to timing fixes that are incorporated in my build.

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
sleep 0.05 0.0500488 4.88311e-05
sleep 0.1 0.098877 -0.00112304
sleep 0.25 0.25 0.0
sleep 0.5 0.5 0.0
sleep 1 1.0 0.0

tone 0.05 0.220947 0.170947
tone 0.1 0.266113 0.166113
tone 0.25 0.422852 0.172852
tone 0.5 0.680908 0.180908
tone 1 1.14209 0.14209
```

My test code:
```python
from adafruit_circuitplayground import cp
import time

for test in ("sleep", "tone"):
    for length in (0.05, 0.1, 0.25, 0.5, 1):
        start = time.monotonic()
        if test == "sleep":
            time.sleep(length)
        else:
            cp.play_tone(440, length)
        diff = time.monotonic() - start
        print(test, length, diff, diff - length)
    print()
```

robertgallup, 2021-08-11:
Very interesting, and thanks for looking into this! Are the ramps built in? Or, does the CP library have control over them? As I recall, tones play sine wave samples generated by the CP library. With #1623, ramps may not be required at all if the sample begins/ends on zero.

Also, are different tones generated by generating different samples, rather than using the same sample with different sampling rates? (I may be mistaken, but I seem to remember noticing this when I looked at the code last year)? If so, that seems like it could add overhead to changing tones.

Finally, not related to timing, but it's interesting that a sine wave is used for tones. Is there a reason a triangle or square wave isn't used? (easier to calculate and arguably more sonically interesting)

tannewt, 2021-08-12:
The ramps are built in. The library can set the `quiescent_value` which sets the destination. The init and deinit go to zero so setting `quiescent_value=0` would only lead to ramps on stop.

We discussed this a bit in a meeting yesterday and thought the long term solution is to remove the ramps in favor of controlling speaker enable. This would allow us to disconnect the speaker and let it settle on its own.

For the library questions, I think its better to open a new issue. Generally we only want to provide one option. Folks can write their own code to have more options.

tannewt, 2021-08-12:
I'm going to close this issue because the timings now match 5.3.0 again.

closed 2021-08-12 by tannewt

robertgallup, 2021-08-20:
Awesome! I finally tried this out with 7.0.0 Alpha 6 and it's such a relief to have tones again that don't stagger drunkenly along!  :-)