adafruit/circuitpython #3646: GrayScale OLED not working in RC0

makermelissa, 2020-11-05:
I tested using I2C. There were no errors, it just didn't display anything. It does work with the absolute latest, however when it does an autoreload it does not run until you unplug and replug in the board. It seems like the I2C port is not being reset.

makermelissa, 2020-11-05:
I just noticed RC.1 is available and will try it with that.

makermelissa, 2020-11-05:
Oops, just realized I had originally noticed the issue in rc.1.

tannewt, 2020-11-05:
Please post your test code and also note what board you are using. Does it work on other ports?

makermelissa, 2020-11-05:
I'm using a Feather M4 Express and my test code is at https://github.com/adafruit/Adafruit_CircuitPython_SSD1327/blob/master/examples/ssd1327_simpletest.py

makermelissa, 2020-11-05:
What do you mean ports? Like different pins or do you mean a different board?

tannewt, 2020-11-06:
I mean a different board with a different chip. (Not just another SAMD51)

tannewt, 2020-11-10:
This works for me on 6.0.0-rc.1 on the Feather M4 Express and the latest SSD1327 library (1.2.0). I just commented out the display_text bits to reduce my dependencies.

dhalbert, 2020-11-11:
@makermelissa Do you still see this with the same or different test conditions as @tannewt? This does not seem to be a showstopper for 6.0.0 any more unless you are doing something different.

makermelissa, 2020-11-11:
Oh right. I'll check again.

makermelissa, 2020-11-11:
I just tried again and it's working.

closed 2020-11-11 by makermelissa

dhalbert, 2020-11-11:
OK, great! Thanks.