adafruit/circuitpython #3649: metro esp32s2 after NameError

jepler, 2020-11-05:
1. Load 6.0.0-rc1 via esptool
 1. `storage.erase_filesystem()`
 1. Copy **adafruit_ahtx0.mpy** and **adafruit_bus_device** to CIRCUITPY/lib -- I used adafruit-circuitpython-bundle-6.x-mpy-20201105.zip the latest at time of writing.
 1. At the repl, try to import/use it.  Somehow the import appears to succeed, but the imported module still doesn't exist.  Trying to import it a second time does a reset that bypasses safe mode.
Note: Repeating these steps also tends to trigger the Linux USB bug and will crash your Linux computer after a few iterations.  UGH.
```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-rc.1 on 2020-11-03; Metro ESP32S2 with ESP32S2
>>> 
>>> import adafruit_ahtx0
>>> adafruit_ahtx0
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'adafruit_ahtx0' is not defined
>>> import adafruit_ahtx0

[tio 16:15:59] Disconnected
[tio 16:16:01] Connected
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
```

askpatrickw, 2020-11-07:
I was able to reproduce this on a Kaluga with a release and a debug build.

```
I (630108) gpio: GPIO[45]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008d4de  PS      : 0x00060430  A0      : 0x8008eb0f  A1      : 0x3ffdbb40
A2      : 0x00000006  A3      : 0x72666164  A4      : 0x3ffdba00  A5      : 0x3f00bc94
A6      : 0x00000000  A7      : 0x3fd8bda0  A8      : 0x72000004  A9      : 0x00000000
A10     : 0x6f706d69  A11     : 0x00000000  A12     : 0x0000000a  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x00000019  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000018  LBEG    : 0x0000000a  LEND    : 0x00000000  LCOUNT  : 0x400290f5

Backtrace:0x4008d4db:0x3ffdbb40 0x4008eb0c:0x3ffdbb70 0x4008ec89:0x3ffdbba0 0x4008ec0d:0x3ffdbbc0 0x40090f3d:0x3ffdbbf0 0x4008d235:0x3ffdbc10 0x4009a6d4:0x3ffdbc30 0x40091c0b:0x3ffdbcd0 0x4008d565:0x3ffdbd00 0x4008d592:0x3ffdbd20 0x400abcd5:0x3ffdbd40 0x400abf85:0x3ffdbdf0 0x4009e9a8:0x3ffdbe30 0x4009ea20:0x3ffdbe50 0x4009ec2f:0x3ffdbe70 0x4002fb60:0x3ffdbe90

ELF file SHA256: 21b3b32c81f568b7

CPU halted.
```
Backtrace

```
got ['0x4008d4db', '0x4008eb0c', '0x4008ec89', '0x4008ec0d', '0x40090f3d', '0x4008d235', '0x4009a6d4', '0x40091c0b', '0x4008d565', '0x4008d592', '0x400abcd5', '0x400abf85', '0x4009e9a8', '0x4009ea20', '0x4009ec2f', '0x4002fb60']
/circuitpython/ports/esp32s2/../../py/runtime.c:262 (discriminator 3)
/circuitpython/ports/esp32s2/../../py/map.c:226
/circuitpython/ports/esp32s2/../../py/map.c:136
/circuitpython/ports/esp32s2/../../py/map.c:289
/circuitpython/ports/esp32s2/../../py/objdict.c:598
/circuitpython/ports/esp32s2/../../py/runtime.c:207
/circuitpython/ports/esp32s2/../../py/vm.c:410
/circuitpython/ports/esp32s2/../../py/objfun.c:286
/circuitpython/ports/esp32s2/../../py/runtime.c:624
/circuitpython/ports/esp32s2/../../py/runtime.c:598
/circuitpython/ports/esp32s2/../../lib/utils/pyexec.c:114
/circuitpython/ports/esp32s2/../../lib/utils/pyexec.c:520
/circuitpython/ports/esp32s2/../../main.c:443
/circuitpython/ports/esp32s2/../../main.c:510
/circuitpython/ports/esp32s2/supervisor/port.c:213
/circuitpython/ports/esp32s2/build-espressif_kaluga_1/esp-idf/../../esp-idf/components/freertos/xtensa/port.c:535
```

jepler, 2020-11-10:
This seems to have to do with the repl.  Debugging shows that the memory used by `rl.line.buf` (readline) and `mp_state_ctx.thread.dict_globals.map.table` can end up overlapping (which shouldn't happen)

```Hardware watchpoint 4: -location mp_state_ctx.thread.dict_globals.map.table[0].key
(gdb) c
Continuing.
esp32s2: Target halted, PC=0x40084A60, debug_reason=00000002

Thread 1 received signal SIGTRAP, Trace/breakpoint trap.
vstr_ins_char (vstr=0x3ffdbea0, char_pos=0, chr=97) at ../../py/vstr.c:212
212        *s = chr;
(gdb) p s
$32 = 0x3fd8bc10 "~\001"
(gdb) p &mp_state_ctx.thread.dict_globals.map.table[0].key
$33 = (mp_obj_t *) 0x3fd8bc10
(gdb) where
#0  vstr_ins_char (vstr=0x3ffdbea0, char_pos=0, chr=97) at ../../py/vstr.c:212
#1  0x400a80ec in readline_process_char (c=97) at ../../lib/mp-readline/readline.c:242
#2  0x400a84df in readline (line=0x3ffdbea0, prompt=0x3f013138 "81\001?") at ../../lib/mp-readline/readline.c:482
#3  0x400ab982 in pyexec_friendly_repl () at ../../lib/utils/pyexec.c:455
#4  0x4009e60f in run_repl () at ../../main.c:443
#5  0x4009e687 in main () at ../../main.c:510
(gdb) p rl.line.buf
$36 = 0x3fd8bc10 "~\001"
(gdb) p mp_state_ctx.thread.dict_globals.map.table
$37 = (mp_map_elem_t *) 0x3fd8bc10
```
Adding additional debugging prints shows that
 * The initial value ever assigned to `line.buf` is that `0x3fd8bc10` value
 * `line` resides on the stack of `pyexec_friendly_repl`
 * (presumably the GC accidentally thinks that the memory pointed to by `line.buf` can be freed)
 * The globals map table ends up pointed at the same storage

PR #3668 (2020-11-10): esp32s2: Correct port_stack_get_top()
The main stack is bigger than `CONFIG_ESP_MAIN_TASK_STACK_SIZE` -- an "extra" size is added to it (`TASK_EXTRA_STACK_SIZE`).  This total size is available as `ESP_TASK_MAIN_STACK`.  Presumably `TASK_EXTRA_STACK_SIZE` is additional stack that can be used by the esp-idf runtime.  But what's important for us is that some very outermost stack frames, such as `pyexec_friendly_repl`, could lie inside the "extra" area and be invisible to the garbage collector.  

Closes #3649
commit 2d8ebfcf (2020-11-10): esp32s2: Correct port_stack_get_top()

Closes #3649
closed 2020-11-10 by tannewt