adafruit/circuitpython #3651: ESP32S2 - Order of requests triggers SSL handshake failure

jerryneedell, 2020-11-06:
Using this test program on a metro_esp32s2 with freshly built main and freshly downloaded adafruit_requests

```
import ipaddress
import wifi
import adafruit_requests
import time
import ssl
import socketpool

print(wifi.radio.connect('<ssid>', '<password>))

print('ip', wifi.radio.ipv4_address)

ipv4 = ipaddress.ip_address('8.8.8.8')
print('ping', wifi.radio.ping(ipv4))

pool = socketpool.SocketPool(wifi.radio)

https = adafruit_requests.Session(pool, ssl.create_default_context())

TEXT_URL = "https://httpbin.org/get"
JSON_GET_URL = "https://httpbin.org/get"
JSON_POST_URL = "https://httpbin.org/post"

print("Fetching text from %s" % TEXT_URL)
response = https.get(TEXT_URL)
print("-" * 40)

print("Text Response: ", response.text)
print("-" * 40)
response.close()

print("Fetching JSON data from %s" % JSON_GET_URL)
response = https.get(JSON_GET_URL)
print("-" * 40)

print("JSON Response: ", response.json())
print("-" * 40)
response.close()

data = "31F"
print("POSTing data to {0}: {1}".format(JSON_POST_URL, data))
response = https.post(JSON_POST_URL, data=data)
print("-" * 40)

json_resp = response.json()
# Parse out the 'data' key from json_resp dict.
print("Data received from server:", json_resp["data"])
print("-" * 40)
response.close()

json_data = {"Date": "July 25, 2019"}
print("POSTing data to {0}: {1}".format(JSON_POST_URL, json_data))
response = https.post(JSON_POST_URL, json=json_data)
response = https.post(JSON_POST_URL, json=json_data)
print("-" * 40)

json_resp = response.json()
# Parse out the 'json' key from json_resp dict.
print("JSON Data received from server:", json_resp["json"])
print("-" * 40)
response.close()

response = https.get("https://api.github.com/repos/adafruit/circuitpython")
print(response.status_code)
print("circuitpython open issues", response.json()["open_issues_count"])
print("circuitpython stars", response.json()["stargazers_count"])
response.close()
```
it runs successfully
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Hello World!

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-rc.1-192-gec12477f5 on 2020-11-06; Metro ESP32S2 with ESP32S2
>>> import wifi_https
None
ip 10.0.0.30
ping 0.035
Fetching text from https://httpbin.org/get
----------------------------------------
Text Response:  {
  "args": {}, 
  "headers": {
    "Host": "httpbin.org", 
    "User-Agent": "Adafruit CircuitPython", 
    "X-Amzn-Trace-Id": "Root=1-5fa5937b-28fbeb523e7a377e74f14ed5"
  }, 
  "origin": "73.61.89.123", 
  "url": "https://httpbin.org/get"
}

----------------------------------------
Fetching JSON data from https://httpbin.org/get
----------------------------------------
JSON Response:  {'url': 'https://httpbin.org/get', 'headers': {'User-Agent': 'Adafruit CircuitPython', 'Host': 'httpbin.org', 'X-Amzn-Trace-Id': 'Root=1-5fa5937b-575ca0ec60c767155d54c2d6'}, 'args': {}, 'origin': '73.61.89.123'}
----------------------------------------
POSTing data to https://httpbin.org/post: 31F
----------------------------------------
Data received from server: 31F
----------------------------------------
POSTing data to https://httpbin.org/post: {'Date': 'July 25, 2019'}
----------------------------------------
JSON Data received from server: {'Date': 'July 25, 2019'}
----------------------------------------
200
circuitpython open issues 337
circuitpython stars 1887
>>> 
```

But, if I move 
```
response = https.get("https://api.github.com/repos/adafruit/circuitpython")
print(response.status_code)
print("circuitpython open issues", response.json()["open_issues_count"])
print("circuitpython stars", response.json()["stargazers_count"])
response.close()
```
from the last request being first , then I get this failure on the next request after it.I

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-rc.1-192-gec12477f5 on 2020-11-06; Metro ESP32S2 with ESP32S2
>>> import wifi_https
None
ip 10.0.0.30
ping 0.047
200
circuitpython open issues 337
circuitpython stars 1887
Fetching text from https://httpbin.org/get
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "wifi_https.py", line 32, in <module>
  File "adafruit_requests.py", line 555, in get
  File "adafruit_requests.py", line 535, in request
  File "adafruit_requests.py", line 429, in _get_socket
  File "adafruit_requests.py", line 425, in _get_socket
OSError: Failed SSL handshake
>>> 
```

jerryneedell, 2020-11-06:
another example that always produces an SSL handshake erro is:
```
import ipaddress
import wifi
import socketpool
import time
import adafruit_requests
import ssl
import espidf

import gc

print("cp mem free", gc.mem_free())

print("idf total mem", espidf.heap_caps_get_total_size())
print("idf mem free", espidf.heap_caps_get_free_size())
print("idf largest block", espidf.heap_caps_get_largest_free_block())

for network in wifi.radio.start_scanning_networks():
    print(network, network.ssid, network.rssi, network.channel)

wifi.radio.stop_scanning_networks()

print(wifi.radio.connect("<ssid>", "<password>"))

print("ip", wifi.radio.ipv4_address)

ipv4 = ipaddress.ip_address("8.8.4.4")
print(ipv4)
print("ping", wifi.radio.ping(ipv4))

pool = socketpool.SocketPool(wifi.radio)
requests = adafruit_requests.Session(pool, ssl.create_default_context())

response = requests.get("https://api.thingspeak.com/channels/1417/feeds.json?results=1")
print(response.status_code)
print(response.json())

print("done")

```
I have never been able to get this to work on the ESP32S2
```

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 6.0.0-rc.1-192-gec12477f5 on 2020-11-06; Metro ESP32S2 with ESP32S2
>>> import wifi_cheer
cp mem free 2035200
idf total mem 165768
idf mem free 88780
idf largest block 52348
<Network> Needell Airport -65 6
<Network> Needell Airport -47 1
<Network> WYZE_FBAB62DE71811B38 -65 11
<Network> 31A -82 8
None
ip 10.0.0.30
8.8.4.4
ping 0.05
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "wifi_cheer.py", line 33, in <module>
  File "adafruit_requests.py", line 555, in get
  File "adafruit_requests.py", line 535, in request
  File "adafruit_requests.py", line 429, in _get_socket
  File "adafruit_requests.py", line 425, in _get_socket
OSError: Failed SSL handshake
>>> 
```

BennyE, 2020-11-07:
Hi @jerryneedell,

I've tried this now for a number of hours and my adafruit_requests.py looks like a collection of print() statements. :D 
I tried the same URL, but I have a different result:

RuntimeError: Sending request failed

The same code runs just fine for https://httpbin.org/get fails with the above for https://api.thingspeak.com/channels/1417/feeds.json?results=1

Are you on Adafruit_CircuitPython_Requests 1.7.5?

I noticed that DNS resolution gives two IPs for api.thingspeak.com while api.github.com only gives one IP. From the print outputs, I couldn't find any changing IP or similar. I also noticed that api.thingspeak.com may respond with status code 304 for your second query, but as there is no "location" in the response headers, it shouldn't cause an issue (as it is not actually a redirection, but just the info that nothing changed).

I also tried if the length of the URL is the issue, and just used "https://54.210.227.170", but even that fails with the same error message as above.

My feeling is that the dictionary of "free sockets" is not working as expected. I know that I end up in the retry < 5 loop, that fails to get "any(free_sockets)".

I would have loved to be of actual help, maybe this helps anybody with more experience on this to figure out what is wrong.

Benny

jerryneedell, 2020-11-07:
I updated to 1.7.5 -- was on 1.7.4  and the ordering issue appears to have need resolved.
Now like you, I get 
RuntimError: Sending request failed on the "thingspeak" request

BennyE, 2020-11-08:
Sorry to hear that you now have the same issue, @jerryneedell, but I think I figured out what is going wrong.
I got lured into believing it is a socket issue, but it is not.

[image]

It simply has to do with **what** the ESP32S2 or CircuitPython believes is a valid TLS certificate. I used my server, that I can reach internally via a different FQDN than external and the Let's Encrypt certificate is only valid for the external FQDN/domain. (The above case is the one that fails, but it will end up in "RuntimError: Sending request failed", while actually the connection to the server was established - but the SSL/TLS certificate was found to be invalid. :)
(192.168.11.148 is my featherS2)

I also validated that the same issue happens even when the featherS2 has the correct time (well, at least something like "year/month/day correct").

A TLS certificate that works is the one for https://api.github.com 
The TLS certificate for https://api.thingspeak.com or even https://io.adafruit.com is not accepted.

This page has details and I'm not yet in the position to offer any solutions for this: 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_tls.html

I then stumbled upon #3424 to find that you already knew all of the above :D

I believe the function(s) should be enhanced to return SSL/TLS error messages, but again I don't know how to do this. If this would be solely Python level, I can help but when it comes to C - I'm in trouble.

I see in issue #3424 that the process appears to include 135 certificates, but those should be 138 and the archive has been updated a couple of times since 2019.

BennyE, 2020-11-17:
@jepler @tannewt I believe this one is fixed by #3691 and the initial issue of this post was fixed by adafruit_requests.py >= 1.7.5

jerryneedell, 2020-11-17:
I concur that this is fixed by #3961 and updated adafruit_requests - closing

closed 2020-11-17 by jerryneedell